Koenkk/zigbee2mqtt #22097: [Wrong device]:

rodrigobutta, 2024-04-05:
### Link

https://cnwstele.en.made-in-china.com/product/lEDRoijJXVcw/China-Us-EU-Tuya-WiFi-Smart-Wall-Fan-Light-Switch-Ceiling-Lamp-Voice-Remote-Control-Speed-Control-Electrical-Fan-Light-Touch-Switch.html

### Model

TS0601_fan_and_light_switch

### Description

TS0601_fan_and_light_switch recognized as TS0601_fan_switch

### Vendor

Tuya

### Picture (link)

https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcTMvugy5Tetuk6vzFSHCO-zCnp_5vgIXiRc-VAjl_zHKokrEjWD1598dw5qUVBipCfN5k8&usqp=CAU

### Database entry

cant find where to look for this

### Notes

Hi!

I'm trying for Z2M to recognize the right device (it´s listed) but with no luck. May be the device itself is saying the wrong model?

The device is
https://www.zigbee2mqtt.io/devices/TS0601_fan_and_light_switch.html

And it get recognized as
https://www.zigbee2mqtt.io/devices/TS0601_fan_and_light_switch.html

Thanks and let me know what can I do, since there is no need of a new device I think

Koenkk, 2024-04-06:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-06 by Koenkk

rodrigobutta, 2024-04-06:
Wow man, you´re quick. I´ll check the commit so I can help with some PR in the future if needed. Thanks!