Koenkk/zigbee2mqtt #22101: Inovelli Fan Switch - Cannot set breeze mode

SourceQuality, 2024-04-05:
### What happened?

An error message is displayed when attempting to set the breezemode on Inovelli Blue Fan Switch (VZM35-SN)

Publish 'set' 'breezeMode' to 'Bedroom Switch' failed: 'Error: ZCL command 0x048727fffe194435/2 manuSpecificInovelli.write({"breezeMode":"30948737.8"}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4655,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

### What did you expect to happen?

Breeze mode parameters are set correctly

### How to reproduce it (minimal and precise)

Set parameters for breeze mode and press submit

### Zigbee2MQTT version

 1.36.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Setup

Home Assistant x86-64

Koenkk, 2024-04-07:
@InovelliUSA could you check this?

InovelliUSA, 2024-04-09:
> @InovelliUSA could you check this?

I created a PR that should fix this issue: https://github.com/Koenkk/zigbee-herdsman-converters/pull/7373

Koenkk, 2024-04-09:
Thanks @InovelliUSA !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-09 by Koenkk