Koenkk/zigbee2mqtt #22111: TuYa ZS06 stop working

KiraPC, 2024-04-06:
### What happened?

I'm using the ZS06 with SmartIR, when I change channel with Alexa to a channel with 2 digits (eg 31) the bluster stop working until I restart it (plug and unplug)

I think it could be related to #18705

### What did you expect to happen?

I expect to work setting sequentially the digits. 

### How to reproduce it (minimal and precise)

Send message to the ir_code_to_send topic in parallel

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12

### Setup

HASS OS on Proxmox

diogo-madeira, 2024-05-13:
I have the same problem with the same device.....

KiraPC, 2024-11-10:
No stale

closed 2025-01-19 by github-actions[bot]