Koenkk/zigbee2mqtt #22112: tuya ZWT198/ZWT100-BH - AVATTO thermostat

kirtapix, 2024-04-06:
### What happened?

In z2m expose device menu : 
- system mode settings "manual" and "auto" are inverted, when you set "auto" in z2m frontend, it sets "manual" mode on the device.
- workday settings "5-2" and "6-1" are inverted too, when you set "5-2" on z2m frontend it sets "6-1" setting on device.

### What did you expect to happen?

Obviously I expect that manual mode sets manual mode, and auto mode sets auto mode.
Same for workday : when I choose "5-2" on frontend, it might set "5-2" on device.

### How to reproduce it (minimal and precise)

Use z2m frontend  to notice that system mode settings and workday settings are messed up.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Setup

z2m installed on HA OS, on a rpi4

johnfromul, 2024-04-07:
1) have same issues with ZWT198/ZWT100-BH AVATTO thermostat - i used SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102(N) coordinator
2) can you explain to me  - exposes - sensor - internal/external/both ....how is this feature usable? This thermostat have only one NTC temperature sensor. If i check all exposed entites I don't see any "remote temp sensor" or simialry named entity.
=3) how to use external temperature sensor?

B3WiN, 2024-10-01:
Same issues here.

@johnfromul the zwt100 has the option of connecting an external sensor, the zwt198 not.

studymonkey999, 2024-10-02:
any solutions for this one. i think i have the same problem with this model. also i feel like i can never understand when it powers up and when it powers down. i am standing next to radiator for the test. which is not nice. happy to know if there is solution.

B3WiN, 2024-10-02:
The latest 1.40.2 release solved the preset problem for me (Avatto ZWT198 / _TZE204_xnbkhhdr).  https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.2

The workday problem still exist.

kirtapix, 2024-10-02:
I did the update but still having issues with presets ans working days.
Is there any action that i have to do to apply the fixe ?

B3WiN, 2024-10-02:
> I did the update but still having issues with presets ans working days. Is there any action that i have to do to apply the fixe ?

Which model and zigbee manufacturer?

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7966

According to this for _TZE200_viy9ihs7 the code is:  return {auto: tuya.enum(1), manual: tuya.enum(0), temporary_manual: tuya.enum(2)};

For the others (_TZE204_lzriup1j and _TZE204_xnbkhhdr):     return {auto: tuya.enum(0), manual: tuya.enum(1), temporary_manual: tuya.enum(2)};

That change solved me the preset problem.

kirtapix, 2024-10-02:
I have Avatto model TS0601 manufacturer _TZE200_viy9ihs7.
I did the update, restarted everything but presets auto and manual are still inverted

B3WiN, 2024-10-03:
> I have Avatto model TS0601 manufacturer _TZE200_viy9ihs7. I did the update, restarted everything but presets auto and manual are still inverted

Hmm... theoretically, the TZE200 should work as an inverse compared to the TZE204s in this two points, but if you still have them working in reverse with the updated code, in fact, then you don't need this exchange in the code.

If possibble, I recommend that you rewrite the code manually and try it with the TZE204 line of code and if it works, report it in this issues or even create a PR.

kirtapix, 2024-10-03:
As soon as I have time I'll do this. Thanks for your reply !

Liionboy, 2024-10-20:
Same problem for me Mode are Inverted

Liionboy, 2024-11-10:
tuya.valueConverterBasic.lookup((_, device) => {
                        // https://github.com/Koenkk/zigbee2mqtt/issues/21353#issuecomment-1938328429
                        if (device.manufacturerName === '_TZE200_viy9ihs7') {
                            return {auto: tuya.enum(1), manual: tuya.enum(0), temporary_manual: tuya.enum(2)};
                        } else {
                            return {auto: tuya.enum(0), manual: tuya.enum(1), temporary_manual: tuya.enum(2)};
                        }
                    }),

This line need to be edited and be like this:

                        if (device.manufacturerName === '_TZE200_viy9ihs7') {
                            return {auto: tuya.enum(0), manual: tuya.enum(1), temporary_manual: tuya.enum(2)};

domo86, 2024-12-01:
on ZWT100-BH with the 1.41.0 release, the following problem is still present :

"system mode settings "manual" and "auto" are inverted, when you set "auto" in z2m frontend, it sets "manual" mode on the device."

Do you know if this will be solved in the next realse ?

Thanks

domome, 2024-12-04:
Hello, with 1.42.0 release the problem has not been solved.

> on ZWT100-BH with the 1.41.0 release, the following problem is still present :
> 
> "system mode settings "manual" and "auto" are inverted, when you set "auto" in z2m frontend, it sets "manual" mode on the device."
> 
> Do you know if this will be solved in the next realse ?
> 
> Thanks

Liionboy, 2024-12-04:
For me in this update was solved, but there are another problem with week day, 6+1 is 5+2 and viceversa

Thanks

Danieldiazi, 2024-12-05:
> Hello, with 1.42.0 release the problem has not been solved.

I've created a Pull Request solving this issue: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8446

domome, 2024-12-06:
Now it works correctly for my device '_TZE204_lzriup1, thank you.

closed 2025-02-12 by github-actions[bot]