Koenkk/zigbee2mqtt #22115: IKEA RODRET (W2342) - No Converter available

loswochos, 2024-04-06:
### What happened?

After pairing a new Ikea Rodret Dimmer (W2342), I have got "no converter available" messages in the log. Device is not working.

### What did you expect to happen?

A working device like all other paired Ikea switches/dimmers. :-)

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Z2M in Docker Container on x86 Device

mrmaximas, 2024-04-07:
Please post z2m database entry for this device.

loswochos, 2024-04-08:
@mrmaximas 

`{"id":33,"type":"EndDevice","ieeeAddr":"0x4c5bb3fffed81524","nwkAddr":64097,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"RODRET Dimmer","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,32,4096,64636],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genPollCtrl":{"attributes":{"checkinInterval":172800}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0001571b54","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0001571b54","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0001571b54","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20221114","swBuildId":"1.0.47","zclVersion":8,"interviewCompleted":true,"meta":{"configured":449731383},"lastSeen":1712608062034,"checkinInterval":43200}`

mrmaximas, 2024-04-09:
@Koenkk could you add another option E2201 IKEA RODRET (W2342) for https://www.zigbee2mqtt.io/devices/E2201.html

mrskycriper, 2024-04-09:
I looks like there are some Unicode symbols in its modelId. The current one is exactly the same: `RODRET Dimmer`.

Can you please dev console to get the actual modelId from the `modelId` attribute of the `genBasic` cluster?

Koenkk, 2024-04-09:
@loswochos can you provide the debug log from starting z2m until these errors?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

loswochos, 2024-04-09:
@mrskycriper 

Unfortunately I get no response when I request the modelId. This does not work with any of my Ikea switches. Maybe because it is not an active zigbee device?

@Koenkk 

Here are the relevant parts of the log:

`info  2024-04-09 21:56:53: 0x4c5bb3fffed81524 (0x4c5bb3fffed81524): E2201 - IKEA RODRET wireless dimmer/power switch (EndDevice)
debug 2024-04-09 22:00:21: Received Zigbee message from '0x4c5bb3fffed81524', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2024-04-09 22:00:21: MQTT publish: topic 'zigbee2mqtt/0x4c5bb3fffed81524', payload '{"action":"on","battery":100,"linkquality":40,"update":{"installed_version":16777287,"latest_version":16777287,"state":"idle"},"update_available":null}'
info  2024-04-09 22:00:21: MQTT publish: topic 'zigbee2mqtt/0x4c5bb3fffed81524', payload '{"action":"","battery":100,"linkquality":40,"update":{"installed_version":16777287,"latest_version":16777287,"state":"idle"},"update_available":null}'
info  2024-04-09 22:00:21: MQTT publish: topic 'zigbee2mqtt/0x4c5bb3fffed81524/action', payload 'on'
error 2024-04-09 22:00:21: No converter available for 'action' ("on")
error 2024-04-09 22:00:21: No converter available for 'battery' (100)
error 2024-04-09 22:00:21: No converter available for 'linkquality' (40)
error 2024-04-09 22:00:21: No converter available for 'update' ({"installed_version":16777287,"latest_version":16777287,"state":"idle"})
error 2024-04-09 22:00:21: No converter available for 'update_available' (null)
error 2024-04-09 22:00:21: No converter available for 'action' ("")
error 2024-04-09 22:00:21: No converter available for 'battery' (100)
error 2024-04-09 22:00:21: No converter available for 'linkquality' (40)
error 2024-04-09 22:00:21: No converter available for 'update' ({"installed_version":16777287,"latest_version":16777287,"state":"idle"})
error 2024-04-09 22:00:21: No converter available for 'update_available' (null)`

mrskycriper, 2024-04-09:
@loswochos you have to wake it up, otherwise it won't respond to requests

loswochos, 2024-04-09:
@mrskycriper

Ah, make sense. Here is the response:

`2024-04-09 22:57:38 Read result of 'genBasic': {"modelId":"RODRET Dimmer"}`

mrskycriper, 2024-04-09:
> `2024-04-09 22:57:38 Read result of 'genBasic': {"modelId":"RODRET Dimmer"}`

Now that doesn't make any sense. It should be recognized and working. The model is exactly the same symbol by symbol.

loswochos, 2024-04-10:
> > `2024-04-09 22:57:38 Read result of 'genBasic': {"modelId":"RODRET Dimmer"}`
> 
> Now that doesn't make any sense. It should be recognized and working. The model is exactly the same symbol by symbol.

Maybe a faulty device? Is there anything I can try?

loswochos, 2024-05-14:
I was at Ikea today. There are different numbers on the packaging of the Rodret remote controls. I was able to discover at least the following:

- 2321
- 2342
- 2332
- 2349

Maybe different (firmware) versions?

Koenkk, 2024-05-15:
Besides the `No converter available..` I think the device is working fine, I see action being published:

```
MQTT publish: topic 'zigbee2mqtt/0x4c5bb3fffed81524/action', payload 'on'
```

boojew, 2024-09-06:
I have the same device, with model w2352, and I dont get the error, but in the debug console I dont see any events whatsoever for the device; however it lists all the exposed entities - they just never update

closed 2025-01-06 by github-actions[bot]