Koenkk/zigbee2mqtt #22117: Gledopto GL-C-006P

PskNorz, 2024-04-06:
### What happened?

Today update get realeased and it broke the controller. Colour temperature doesnt work. You choose cold white and it turn off the light.

### Zigbee2MQTT version

 1.36.1-1

### Adapter firmware version

 20230507

### Adapter

Sonoff dongle P

### Setup

Haos - RPI4

mrmaximas, 2024-04-07:
have you tried to re-pair this device?

PskNorz, 2024-04-07:
> have you tried to re-pair this device?

Yes I did, found also [this](https://github.com/Koenkk/zigbee2mqtt/issues/21988#issue-2214097135)

monoimoi, 2024-04-12:
Also experiencing the same issue. QC sticker notes 02/2022 date for my devices. Firmware version 10476801 (build date 20210719) was originally installed and worked fine. Latest firmware update cause control of color temperature to be lost and instead control the brightness of cool white only.

nyok92, 2024-04-12:
```
{
    "color_mode": "color_temp",
    "linkquality": 109,
    "state": "ON",
    "update": {
        "installed_version": 10,
        "latest_version": 12,
        "state": "available"
    },
    "update_available": null,
    "brightness": 125,
    "color_temp": 370
}
```
QC  passed 21/22
FW ; 10276802
FW date : 20210719

Can't FW update the device ; Z2MQTT logs :

```
Info 2024-04-12 11:09:43Updating '0xcc86ecfffe187e3e' to latest firmware
Info 2024-04-12 11:09:47Update of '0xcc86ecfffe187e3e' at 0.00%
Info 2024-04-12 11:09:47MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe187e3e', payload '{"color_mode":"color_temp","color_temp":370,"linkquality":109,"state":"ON","update":{"installed_version":10,"latest_version":12,"progress":0,"state":"updating"},"update_available":null}'
Info 2024-04-12 11:12:17MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe187e3e', payload '{"color_mode":"color_temp","color_temp":370,"linkquality":109,"state":"ON","update":{"installed_version":10,"latest_version":12,"state":"available"},"update_available":null}'
Info 2024-04-12 11:12:17MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0xcc86ecfffe187e3e"},"error":"Update of '0xcc86ecfffe187e3e' failed (OTA: Timeout, device did not request any image blocks)","status":"error","transaction":"2mfdd-1"}'
Error 2024-04-12 11:12:17Update of '0xcc86ecfffe187e3e' failed (OTA: Timeout, device did not request any image blocks)Info 2024-04-12 11:09:43Updating '0xcc86ecfffe187e3e' to latest firmware
Info 2024-04-12 11:09:47Update of '0xcc86ecfffe187e3e' at 0.00%
Info 2024-04-12 11:09:47MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe187e3e', payload '{"color_mode":"color_temp","color_temp":370,"linkquality":109,"state":"ON","update":{"installed_version":10,"latest_version":12,"progress":0,"state":"updating"},"update_available":null}'
Info 2024-04-12 11:12:17MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe187e3e', payload '{"color_mode":"color_temp","color_temp":370,"linkquality":109,"state":"ON","update":{"installed_version":10,"latest_version":12,"state":"available"},"update_available":null}'
Info 2024-04-12 11:12:17MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0xcc86ecfffe187e3e"},"error":"Update of '0xcc86ecfffe187e3e' failed (OTA: Timeout, device did not request any image blocks)","status":"error","transaction":"2mfdd-1"}'
Error 2024-04-12 11:12:17Update of '0xcc86ecfffe187e3e' failed (OTA: Timeout, device did not request any image blocks)

```

Then I can't control the device nymore:

`2024-04-12 11:14:54Failed to read state of '0xcc86ecfffe187e3e' after reconnect (ZCL command 0xcc86ecfffe187e3e/11 genLevelCtrl.read(["currentLevel"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51571 - 11 - 30 - 8 - 1 after 10000ms))`

Solved with Powercycle or with  re-pairing the device, but OTA update didn't worked.

Klikini, 2024-06-21:
I clicked update my mini version not knowing any of this had happened, and now it doesn't respond to any light controls.

I tried installing [this firmware](https://github.com/catduckgnaf/zigbee-OTA/blob/3cdcc512f4cc589ab112ecc1ed0adf585fbea005/images/Gledopto/GL-C-006P(MINI)_20451203_20240227.ota) using both deCONZ and Zigbee2MQTT but neither one works. deCONZ just sits in the "Idle" state after clicking Query/Update, and Zigbee2MQTT reports an image type mismatch.

Is there a way to recover it?

I also have no idea how to interpret this QC sticker, but apparently it could be relevant: 
[image]

closed 2025-01-06 by github-actions[bot]