Koenkk/zigbee2mqtt #22119: "SZ-T04 - Nous Temperature and humidity sensor with clock" fails to bind

ptr727, 2024-04-07:
### What happened?

Got a couple "GIRIER Tuya Smart ZigBee Temperature and Humidity Sensor Digital Thermometer Indoor Hygrometer" from AliExpress. Device sticker says "Temperature and Humidity Sensor" "Model: SZT04".

Device is detected as a [SZ-T04 - Nous Temperature and humidity sensor with clock](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/nous.ts#L26), but fails to bind.

### What did you expect to happen?

Binding succeeds

### How to reproduce it (minimal and precise)

Run latest-dev docker tag, enable herdsman debug logging, set z2m log to debug, permit join, hold button for 5s

### Zigbee2MQTT version

1.36.1 commit: ffc2ff1

### Adapter firmware version

 20221226

### Adapter

 zStack3x0

### Setup

Docker latest-dev

### Debug log

[z2mlog.log](https://github.com/Koenkk/zigbee2mqtt/files/14895440/z2mlog.log)
[_zigbee2mqtt_logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/14895441/_zigbee2mqtt_logs.txt)

mrmaximas, 2024-04-07:
please post z2m database entry for this device.

ptr727, 2024-04-07:
Here you go:  
`{"id":43,"type":"EndDevice","ieeeAddr":"0xa4c138ed012db8f3","nwkAddr":34331,"manufId":4417,"manufName":"_TZE200_qrztc3ev","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�L�-f�L�-\u0012","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_qrztc3ev","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b00258f2865","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1712523649114}`

Now strangely, after trying to bind the device was not in my devices list, but the display on the LCD showed the connected, it was only now when you asked me that I saw it was actually in the devices. Not sure if I missed it the the first time, or if it was added after I re-bound when capturing more logs for this issue.

closed 2024-11-04 by github-actions[bot]