Koenkk/zigbee2mqtt #22120: Lixee Data feedback Value "Null"

Coldness00, 2024-04-07:
### What happened?

My Lixee device used to work fine as I was with Energy Supplier Engie with Base offer, then I decided to switch to EDF with Tempo
Since then, I get only feedback value: "Null"

I of rcourse econfigured first, then Remove from ZigBee network, rebooted Container, and reconfigure no changes
I also tried to power off, power on the Lixee device, but it brought no results

Linky is set up as Historique

See screen shot:
[image]
[image]

Then I tried by filtering the value I desire to be repported, so I have now the Entity in Home Assistant showing up with no values:
[image]
[image]

What is even more strange, is if I go to consol, select Cluster "seMetering", then "currentTier1SummDelivered" and all the other endpoints, I do get the correct feedback but it then never update. 

I can see in the log the payload of my Lixee not changing except for Link Quality, current tarif, available power that are read and pushed.

### What did you expect to happen?

I expect to get the Data from the Linky as it should

### How to reproduce it (minimal and precise)

Change contract...

### Zigbee2MQTT version

 1.36.1 commit: ffc2ff1

### Adapter firmware version

20230507

### Adapter

SLZB-06, zStack3x0

### Setup

Docker via Synology NAS

### Debug log

No error on Logs

kevincaradant, 2024-05-23:
Hi

I experienced the same issue for one or two days, but then everything was resolved. Don't forget to update your Linxee firmware, as it fixed many issues for me.

This is my configuration if you want to use it to achieve the same setup. I don't use 'standard mode' because I don't have a solar panel."

Regards

[image]
[image]

Coldness00, 2024-05-26:
@kevincaradant merci :)
J'avais les mêmes infos mais ca ne marchait pas...
J'ai trouve une solution que j'ai documenté ici:
https://github.com/fairecasoimeme/Zlinky_TIC/issues/256

kevincaradant, 2024-05-28:
Ah nice, well play :)

I have a new issue and it's seems not te be related to the report section as you said. 

I don't have anymore the "BBRHCJB" report. It's frozen since 3 days 
However, on the Linky I have the real value which continue to increase. 

I don't know what to do ^^

closed 2024-12-29 by github-actions[bot]