Koenkk/zigbee2mqtt #22123: [Wrong device]: AVATTO ME167

Blastmun, 2024-04-07:
### Link

https://www.cafago.com/fr/p-c-e0289-2.html

### Model

AVATTO	ME167

### Description

TRV

### Vendor

AVATTO

### Picture (link)

https://img.tttcdn.com/product/xy/2000/2000/p/gu1/C/2/C-E0289-2/C-E0289-2-1-8031-jjP3.jpg

### Database entry

{"id":25,"type":"EndDevice","ieeeAddr":"0xa4c1386b785167a1","nwkAddr":47523,"manufId":4417,"manufName":"_TZE200_6rdj8dzm","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"h֔-i","65506":55,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZE200_6rdj8dzm","zclVersion":3,"appVersion":67,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1712479872318}

### Notes

Hello, I am having a problem with my thermostatic valves. The "battery_low" field never changes to "True". So I never see when the valve has no more battery

mrmaximas, 2024-04-07:
you described the problem as the "wrong device". but that's not true, is it?

Blastmun, 2024-04-08:
> you described the problem as the "wrong device". but that's not true, is it?

My bad,I recreated it with "problem" labels

closed 2024-11-05 by github-actions[bot]