Koenkk/zigbee2mqtt #22124: [Feature request]: Sonoff SNBZ-06P occupancy staye

PresenceCasa, 2024-04-07:
### Is your feature request related to a problem? Please describe

Hello
The sensor only reports movements by z2m. It should be nice to get occupancy state too.

### Describe the solution you'd like

Get a binary sensor state for occupancy

### Describe alternatives you've considered

ZHA reports separately both: occupancy and movement  states. for the same Sonoff SNZB-06P. 

### Additional context

Z2M already perfectly reports both states with esphome LD2410 and LD2450 sensors. I guess that it should not to be difficult to implement this possibility for the SNZB-06P.

jamesonuk, 2024-04-09:
It already has this in Z2M
[image]
and in HA I think the sensor is just badly named as "motion" when it is in fact occupancy.

jamesonuk, 2024-04-09:
@Koenkk I did some digging as I was intrigued as to why this sensor's name didn't match Z2M when all the others did.    First thing was that the discovery message did not seem to have been retained so wasn't in MQTT.   Luckily HA had a copy  

<details>
  <summaryDiscovery Message</summary>
```
availability:
  - topic: zigbee2mqtt/bridge/state
device:
  identifiers:
    - zigbee2mqtt_0x0ceff6fffe6b1ae9
  manufacturer: SONOFF
  model: Zigbee occupancy sensor (SNZB-06P)
  name: '0x0ceff6fffe6b1ae9'
  sw_version: 1.0.6
  via_device: zigbee2mqtt_bridge_0x00212effff070ec3
device_class: motion
object_id: 0x0ceff6fffe6b1ae9_occupancy
origin:
  name: Zigbee2MQTT
  sw_version: 1.36.1
  support_url: https://www.zigbee2mqtt.io
payload_off: false
payload_on: true
state_topic: zigbee2mqtt/0x0ceff6fffe6b1ae9
unique_id: 0x0ceff6fffe6b1ae9_occupancy_zigbee2mqtt
value_template: '{{ value_json.occupancy }}'
platform: mqtt
```
</details>
but this doesn't include a name attribute so HA has picked one.

I have a Tuya presence detector too, which I realise can't be compared directly but looking at that, the discovery message was retained and the device_class is presence rather than motion.

The Sonoff device iconverter has
```
        fromZigbee: [fz.occupancy],
        exposes: [e.occupancy()],
```
compared to the Tuya one which has
```
        exposes: [
            e.illuminance_lux(), e.presence(),
```
but obviously comes from the annoying Tuya DPs.

Would I be right in thinking that Sonoff has used the ZCL cluster / attributes normally used for motion sensors hence e.occupancy is reallty motion and there is no way of getting this device to report presence rather than motion?

Koenkk, 2024-04-09:
I'm not sure what is expected here, from the OP:

> Get a binary sensor state for occupancy

But this exactly what this sensor [provides](https://www.zigbee2mqtt.io/devices/SNZB-06P.html#occupancy-binary)?

psilantropy, 2024-04-23:
ZHA provides motion and occupancy as separate entities in home assistant. Whereas Z2M provides only motion (which seems to actually be occupancy). It’s just confusing for anybody using these devices. 

The fact the devices are also a bit crap and get stuck on detected makes setting these up even more annoying.

closed 2024-11-20 by github-actions[bot]