Koenkk/zigbee2mqtt #22126: the device status does not change in dashboard

serlinGi, 2024-04-07:
### What happened?

hi all,
different devices (for the moment I have checked with lights of various brands and models) do not change the status in the z2m dashboard;  
if I act as devices->esposes or as homeassistant the lights turn on and off regularly and their status is also published regularly in the device's dashboard;  
if instead I act directly from the dashboard it doesn't happen either physically or in the values ​​displayed;  
the same thing happens if I turn it on or off from the physical button: the light turns on but the status is not updated either on z2m and consequently not even on HA

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

zStack3x0 sonoff dongle usb P

### Setup

docker container in Linux Debian

mrmaximas, 2024-04-07:
how about some logs?

serlinGi, 2024-04-07:
yes, of course, it would be appropriate but... 
I forgot to write it before: no message appears in the logs (neither when setting the "info" level nor "debug" ) when I turn it on from the physical switch or when I try to turn it on from the dashboard, without getting any light to come on
however, everything is fine when I switch on from devices->exposes->state: the is published correctly the 'received MQTT message...' and 'Publishing set state...'

I don't know what else to do😢

serlinGi, 2024-04-07:
> yes, of course, it would be appropriate but... I forgot to write it before: no message appears in the logs (neither when setting the "info" level nor "debug" ) when I turn it on from the physical [...]

[image]

TRSEHOME, 2024-04-08:
Hi, i have the same issue: when I change the device status via the dashboard, nothing happens. There are also no entries in the log. However, when I switch the device via the "Devices" tab, the status in the dashboard updates.

AiramsRT, 2024-04-09:
Hello.  Same problem here from HomeAasistsnt

hansf71, 2024-04-10:
Hello, I noticed the same problem: plugs and lights will not turn on or off from the dashboard. From the 'Exposes' tab of the device it works fine. Also mqtt commands are executed as they should. Besides on-off also brightness won't work from the dashboard.
I am using the latest release. This problem seems to be introduced since upgrading to a newer Z2M version recently, I only noticed this now. (Debian 12 with SMLight SLZB06)

erdnamulb, 2024-04-11:
Hi, same problem for me: no direct interaction via the Z2M Dashboard to ALL devices is possible anymore. If I go to a device-specific sub-page (click on device name) and this is e.g. a lamp, the interaction works (i.e. turning it directly on or off from there). But on the main Dashboard NO operation possible at all.
MQTT messages, all Zigbee connections work flawlessly like before Update to 1.36.1. The Dashboard seems to be somewhat buggy now...

serlinGi, 2024-04-11:
for me the main problem is that there is no correspondence when I act with the physical button.
that's where I have problems: I don't know that a light has been turned on;  the automations connected to it do not start;  etc. etc

for the rest, the not working dashboard does not cause me any harm;  when I use HA or a voice assistant, everything works fine.

and it's really depressing not to find any confirmation in the logs: no messages, no queue... as if nothing existed!! 😢

alexsaas, 2024-04-12:
Same here. No controls in dashboard working, since 1.36.1-1.

Chakkaaa, 2024-04-12:
Same here. Dashboard light toggles having no effect. The details page of the device works for switching the lights.

sammko, 2024-04-12:
The javascript console in the browser reports this when performing an action in the dashboard.

```
Uncaught TypeError: a is not a function
    onChange Dashboard-cc086107.js:255
    onChange DashboardFeatureWrapper-ef543cb2.js:1
    onChange DashboardFeatureWrapper-ef543cb2.js:1
    onChange DashboardFeatureWrapper-ef543cb2.js:1
    i DashboardFeatureWrapper-ef543cb2.js:1
    xye index-fb4a4614.js:40
    Hye index-fb4a4614.js:40
    jye index-fb4a4614.js:40
    qC index-fb4a4614.js:40
    EH index-fb4a4614.js:40
    Hg index-fb4a4614.js:40
    Lw index-fb4a4614.js:43
    JR index-fb4a4614.js:40
    Hg index-fb4a4614.js:40
    ow index-fb4a4614.js:40
    eke index-fb4a4614.js:40
    DH index-fb4a4614.js:40
    Rg index-fb4a4614.js:40
    l index-fb4a4614.js:40
    $l index-fb4a4614.js:40
    createRoot index-fb4a4614.js:43
    <anonymous> index-fb4a4614.js:217
```

Strixx76, 2024-04-16:
Same here on version 1.36.1. Nothing in the logs but browser console gives me this when trying to change the state from the dashboard:
```
Uncaught TypeError: a is not a function
    at onChange (Dashboard-cc086107.js:1:3112)
    at Yt.onChange (DashboardFeatureWrapper-ef543cb2.js:1:27283)
    at Yt.onChange (DashboardFeatureWrapper-ef543cb2.js:1:27283)
    at onChange (DashboardFeatureWrapper-ef543cb2.js:1:2373)
    at i (DashboardFeatureWrapper-ef543cb2.js:1:1405)
    at Object.xye (index-fb4a4614.js:40:9864)
    at Hye (index-fb4a4614.js:40:10021)
    at jye (index-fb4a4614.js:40:10081)
    at qC (index-fb4a4614.js:40:31567)
    at EH (index-fb4a4614.js:40:31985)
```

SkyByte, 2024-04-17:
I can confirm this issue, in the dashboard the control items have no effect any more, nor status changes. Only with the 'details' page of each device switching/controlling is possible.

echaritos, 2024-04-19:
same issue here, since 1.36.1

Tomek5111, 2024-04-27:
Hello,
same Problem..  I can't switch devices (on/off) via the dashboard

adamhf, 2024-04-28:
Same Issue here.  I also get errors in the javascript console, but nothing in the application logs

```
index-fb4a4614.js:40 Uncaught TypeError: a is not a function
    at onChange (Dashboard-cc086107.js:1:3112)
    at Yt.onChange (DashboardFeatureWrapper-ef543cb2.js:1:27283)
    at Yt.onChange (DashboardFeatureWrapper-ef543cb2.js:1:27283)
    at onChange (DashboardFeatureWrapper-ef543cb2.js:1:2373)
    at i (DashboardFeatureWrapper-ef543cb2.js:1:1405)
    at Object.xye (index-fb4a4614.js:40:9864)
    at Hye (index-fb4a4614.js:40:10021)
    at jye (index-fb4a4614.js:40:10081)
    at qC (index-fb4a4614.js:40:31567)
    at EH (index-fb4a4614.js:40:31985)
```

echaritos, 2024-05-02:
is this fixed in 1.37.0 ? can someone verify?

Chakkaaa, 2024-05-02:
I just updated and I can verify that the dashboard toggles are now functioning as they should.

serlinGi, 2024-05-02:
> is this fixed in 1.37.0 ? can someone verify?

I can only check this evening (Italian time)

thx

SixFive7, 2024-05-02:
Yes, I can verify this issue has been resolved since the last update. My dashboard is working again.

serlinGi, 2024-05-02:
> is this fixed in 1.37.0 ? can someone verify?

HI,

 I just finished some tests;
 I can confirm that it's much better than before but everything isn't all right yet;

the lights controlled by the Vimar buttons (On_Off_Switch_v1.0 - model 14592.0) respond by operating them from the dashboard but not by the physical button: by acting on this I do not get any status updates anywhere;

some lights controlled with relay TS0001 (Tuya WHD002) sometimes yes, sometimes no: the mqtt status always responds but not always on the dashboard, they remain unchanged without a precise rule (random);

the other devices (Lidl lights various models, the various plugs TS0121, TS011f, etc.) seem to me to work well on both fronts and always seem to respond correctly.

can I do anything else?
👋
(forgive my English obtained with help of translator)

Tomek5111, 2024-05-03:
It works for me now after the update :)

serlinGi, 2024-05-11:
Hi,
in the last few hours, with the latest dev version (6b60c4e), after having clicked 'reconfigure' again for every single device that was giving problems, everything seems to be working as best as possible.  
I believe that if this continues, this issue and similar ones #22302 to it can be closed.  
Thanks @Koenkk

Syxxxx, 2024-09-07:
Hello,
I have the same probleme on 1.40.0/1
If i use a remote, the status of light is not updated.
But if i use z2m to change the state,  i have the update of status
```
info 2024-09-07 22:13:29z2m: Successfully changed options
info 2024-09-07 22:13:29z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"0mqqh-1"}'
info 2024-09-07 22:13:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe7f4334', payload '{"action":"off","action_group":57242,"battery":35,"device":{"applicationVersion":1,"dateCode":"20200708","friendlyName":"0xcc86ecfffe7f4334","hardwareVersion":1,"ieeeAddr":"0xcc86ecfffe7f4334","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2001/E2002","networkAddress":33204,"powerSource":"Battery","softwareBuildID":"1.0.024","stackVersion":103,"type":"EndDevice","zclVersion":3},"last_seen":"2024-09-07T20:13:43.098Z","linkquality":164,"update":{"installed_version":65572,"latest_version":33816598,"state":"available"}}'
info 2024-09-07 22:13:43z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe7f4334', payload '{"battery":35,"device":{"applicationVersion":1,"dateCode":"20200708","friendlyName":"0xcc86ecfffe7f4334","hardwareVersion":1,"ieeeAddr":"0xcc86ecfffe7f4334","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2001/E2002","networkAddress":33204,"powerSource":"Battery","softwareBuildID":"1.0.024","stackVersion":103,"type":"EndDevice","zclVersion":3},"last_seen":"2024-09-07T20:13:43.200Z","linkquality":140,"update":{"installed_version":65572,"latest_version":33816598,"state":"available"}}'
info 2024-09-07 22:13:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe7f4334', payload '{"action":"on","battery":35,"device":{"applicationVersion":1,"dateCode":"20200708","friendlyName":"0xcc86ecfffe7f4334","hardwareVersion":1,"ieeeAddr":"0xcc86ecfffe7f4334","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2001/E2002","networkAddress":33204,"powerSource":"Battery","softwareBuildID":"1.0.024","stackVersion":103,"type":"EndDevice","zclVersion":3},"last_seen":"2024-09-07T20:13:45.821Z","linkquality":164,"update":{"installed_version":65572,"latest_version":33816598,"state":"available"}}'
info 2024-09-07 22:13:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe7f4334', payload '{"battery":35,"device":{"applicationVersion":1,"dateCode":"20200708","friendlyName":"0xcc86ecfffe7f4334","hardwareVersion":1,"ieeeAddr":"0xcc86ecfffe7f4334","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2001/E2002","networkAddress":33204,"powerSource":"Battery","softwareBuildID":"1.0.024","stackVersion":103,"type":"EndDevice","zclVersion":3},"last_seen":"2024-09-07T20:13:45.920Z","linkquality":144,"update":{"installed_version":65572,"latest_version":33816598,"state":"available"}}'
info 2024-09-07 22:14:14z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffeb31fbc', payload '{"brightness":168,"color_mode":"color_temp","color_temp":454,"color_temp_startup":250,"device":{"applicationVersion":33,"dateCode":"20211102","friendlyName":"0xb4e3f9fffeb31fbc","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffeb31fbc","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"L1529","networkAddress":40108,"powerSource":"Mains (single phase)","softwareBuildID":"2.3.087","stackVersion":98,"type":"Router","zclVersion":3},"last_seen":"2024-09-07T20:14:14.198Z","linkquality":164,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":587757105,"latest_version":587757105,"state":"idle"}}'
info 2024-09-07 22:14:17z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffeb31fbc', payload '{"brightness":168,"color_mode":"color_temp","color_temp":454,"color_temp_startup":250,"device":{"applicationVersion":33,"dateCode":"20211102","friendlyName":"0xb4e3f9fffeb31fbc","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffeb31fbc","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"L1529","networkAddress":40108,"powerSource":"Mains (single phase)","softwareBuildID":"2.3.087","stackVersion":98,"type":"Router","zclVersion":3},"last_seen":"2024-09-07T20:14:17.217Z","linkquality":168,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":587757105,"latest_version":587757105,"state":"idle"}}'
info 2024-09-07 22:14:17z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xb4e3f9fffeb31fbc', payload '{"brightness":168,"color_mode":"color_temp","color_temp":454,"color_temp_startup":250,"device":{"applicationVersion":33,"dateCode":"20211102","friendlyName":"0xb4e3f9fffeb31fbc","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffeb31fbc","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"L1529","networkAddress":40108,"powerSource":"Mains (single phase)","softwareBuildID":"2.3.087","stackVersion":98,"type":"Router","zclVersion":3},"last_seen":"2024-09-07T20:14:17.217Z","linkquality":168,"power_on_behavior":"previous","state":"ON","update":{"installed_version":587757105,"latest_version":587757105,"state":"idle"}}'
info 2024-09-07 22:14:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x847127fffe082b6e', payload '{"current":0.03,"device":{"applicationVersion":65,"dateCode":"","friendlyName":"0x847127fffe082b6e","hardwareVersion":1,"ieeeAddr":"0x847127fffe082b6e","manufacturerID":4098,"manufacturerName":"_TZ3000_rdtixbnu","model":"TS0121_plug","networkAddress":45927,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"energy" [... cut ...]
info 2024-09-07 22:14:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xcc86ecfffe7f4334', payload '{"battery":35,"device":{"applicationVersion":1,"dateCode":"20200708","friendlyName":"0xcc86ecfffe7f4334","hardwareVersion":1,"ieeeAddr":"0xcc86ecfffe7f4334","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E2001/E2002","networkAddress":33204,"powerSource":"Battery","softwareBuildID":"1.0.024","stackVersion":103,"type":"EndDevice","zclVersion":3},"last_seen":"2024-09-07T20:14:22.441Z","linkquality":140,"update":{"installed_version":65572,"latest_version":33816598,"state":"available"}}'
```
thanks

robygi, 2024-10-18:
Hi,
I have the same problem with z2m version 1.40.2 in a docker container on Debian 12.
I have a water leak sensor that just after z2m installation was reporting in real time its status in the z2m dashboard.
Now it doesn't update the dashboard but everything is working fine, included receiving mqtt message from the sensor.
I don't see anything meaningful to me in the logs after triggering the sensor:
```
[2024-10-18 15:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-10-18 15:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=5 frmTx=5)
[2024-10-18 15:49:50] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=1](frmRx=1) Added to rxQueue
[2024-10-18 15:49:50] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=5)
[2024-10-18 15:49:50] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=180 Len=33]
[2024-10-18 15:49:50] debug: 	zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":1280,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":201}], [packetInfo:{"senderShortId":7582,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":232,"lastHopRssi":-42,"lastHopTimestamp":0}], [messageContents=192300010000170000]
[2024-10-18 15:49:50] debug: 	zh:controller: Received payload: clusterID=1280, address=7582, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=232, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":35,"commandIdentifier":0},"payload":{"zonestatus":1,"extendedstatus":0},"command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}}
[2024-10-18 15:49:50] debug: 	z2m: Received Zigbee message from 'ZBRainSensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/voltage', payload '3000'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/battery', payload '100'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/occupancy', payload 'false'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/tamper', payload 'false'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/battery_low', payload 'false'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/last_seen', payload '2024-10-18T15:49:50+02:00'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/linkquality', payload '232'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/water_leak', payload 'true'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-friendlyName', payload 'ZBRainSensor'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-model', payload 'SQ510A'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-ieeeAddr', payload '0xa4c138f219c054ff'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-networkAddress', payload '7582'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-type', payload 'EndDevice'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-manufacturerID', payload '4742'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-powerSource', payload 'Battery'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-applicationVersion', payload '113'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-stackVersion', payload '2'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-zclVersion', payload '3'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-hardwareVersion', payload '0'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-dateCode', payload ''
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-softwareBuildID', payload '0122052017'
[2024-10-18 15:49:50] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ZBRainSensor/device-manufacturerName', payload 'eWeLink'

```

closed 2024-12-29 by serlinGi