Koenkk/zigbee2mqtt #22128: TS0502B (MiBoxer FUT035Z+) not working after update Zigbee2MQTT to version 1.36.1

NuttShell, 2024-04-07:
### What happened?

After several on/off cycles, the controller (TS0502B ) stops responding to commands and an error appears in the logs. The maximum brightness level in the settings becomes 1000 instead of 254.
This error and similar behavior is repeated on two HomeAssistant installations with different adapters - Sonoff ZBDongle-P and Egony Stick V4 with Zigbee2MQTT version 1.36.1-1.
Downgrade Zigbee2MQTT version 1.36.1-1 to 1.36.0-1 solve this problem

### What did you expect to happen?

in Zigbee2MQTT version 1.36.0-1 all working with no errors and problems.

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20221226

### Adapter

Sonoff ZBDongle-P and Egony Stick V4

### Setup

Add-on on Home Assistant Supervised 2024.4.1

### Debug log

Publish 'set' 'state' to '0xa4c138f7dbfa8df2' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0xa4c138f7dbfa8df2/1 genLevelCtrl.moveToLevelWithOnOff({"level":724,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 724)'

**and some log with wrong "brightness" value:**
 2024-04-08 01:01:44MQTT publish: topic 'zigbee2mqtt/0xa4c138f7dbfa8df2', payload '{"brightness":1000,"color_mode":"xy","color_temp":0,"do_not_disturb":true,"last_seen":"2024-04-07T18:01:45.718Z","linkquality":36,"state":"OFF"}'
Info 2024-04-08 01:01:44MQTT publish: topic 'zigbee2mqtt/0xa4c138f7dbfa8df2', payload '{"brightness":1000,"color_mode":"xy","color_temp":0,"do_not_disturb":true,"last_seen":"2024-04-07T18:01:46.125Z","linkquality":40,"state":"OFF"}'
Info 2024-04-08 01:01:45MQTT publish: topic 'zigbee2mqtt/0xa4c138f7dbfa8df2', payload '{"brightness":1000,"color_mode":"xy","color_temp":0,"do_not_disturb":true,"last_seen":"2024-04-07T18:01:46.613Z","linkquality":36,"state":"OFF"}'
Info 2024-04-08 01:01:45MQTT publish: topic 'zigbee2mqtt/0xa4c138f7dbfa8df2', payload '{"brightness":1000,"color_mode":"xy","color_temp":0,"do_not_disturb":true,"last_seen":"2024-04-07T18:01:46.815Z","linkquality":40,"state":"OFF"}'

NuttShell, 2024-04-07:
looks similar / related to
https://github.com/Koenkk/zigbee2mqtt/issues/22078

Koenkk, 2024-04-07:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/22078

closed 2024-04-07 by Koenkk