Koenkk/zigbee2mqtt #22130: WZ5 incorrectly recognized as TS0503B

slush0, 2024-04-07:
### What happened?

Hello,

I have physically this device (two units), and I bought them specifically after checking that it is supported by z2m: https://www.zigbee2mqtt.io/devices/WZ5_cct.html. Unfortunately, no matter how hard I try, it gets recognized as TS0503B: https://www.zigbee2mqtt.io/devices/TS0503B.html#tuya-ts0503b

Thanks to this, most of the device functionality is unavailable in zigbee2mqtt. I can turn it on/off, change color (in RGB mode), but brightness does not work and I cannot control LED in CCT/RGBW mode.

I tried to debug this for several hours, but I'm at dead end now. The device has direct connection to SkyConnect hub (I have two of them, tried combination of both dimmers with both hubs). I don't have any similar problems with dozen of other zigbee gadgets running at my home in the same setup.

### What did you expect to happen?

I expected the device will be correctly assigned to WZ5_cct dimmer driver as it should be supported by zigbee2mqtt, according to its own docs.

### How to reproduce it (minimal and precise)

1. New WZ5 device, configured in CCT or RGBW.
2. Permit joining new devices in zigbee2mqtt frontend.
3. Configure network on WZ5 and connect it to the hub.
4. The device successfully connects to the hub, but zigbee2mqtt see the device as TS0503B.

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

7.1.1.0 build 273

### Adapter

SkyConnect

### Setup

Docker container on linux based host.

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14898817/log.txt)

AnderVentura, 2024-04-07:
I have the same problem. It was working before but recently it started to behave differently. After checking, I see Z2M recognizes the device as TS0505B

m-landowski, 2024-04-19:
I have the same problem,

jtherin, 2024-06-24:
I have two WZ5 in RGB+CCT (blue led on the controller) connected to 3 leds strips (rgb + cw + ww) each.

One is recognized as TS0505B_1, the other as WZ5_dim_2.
They should be WZ5_rgbcct :/

sakcaj, 2024-09-04:
Same issue. I was installing strip with the driver and it was all fine then suddenly, once I've finished the installation I found that the device type changed to TS0503B (Manufacturer _TZB210_nfzrlz29).

sakcaj, 2024-09-05:
Below is data/database.db entry of my device as well as a converter I wrote based on https://github.com/Koenkk/zigbee2mqtt/issues/11701
I've uploaded the converter, updated my config.yml, restarted Z2M but the device is still the same. I've removed the device via z2m GUI, added it again but it still appeared as TS0503B, the 
```
{
  "id": 31,
  "type": "Router",
  "ieeeAddr": "0xa4c138af678fce92",
  "nwkAddr": 3089,
  "manufId": 4417,
  "manufName": "_TZB210_nfzrlz29",
  "powerSource": "Mains (single phase)",
  "modelId": "TS0503B",
  "epList": [
    1,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 268,
      "inClusterList": [
        3,
        4,
        5,
        6,
        4096,
        8,
        768,
        61184,
        0
      ],
      "outClusterList": [
        25,
        10
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "65503": "\u0016\u0000\u0000\u0000\u0012",
            "65506": 54,
            "65508": 0,
            "stackVersion": 0,
            "dateCode": "",
            "appVersion": 64
          }
        },
        "lightingColorCtrl": {
          "attributes": {
            "57600": [
              354,
              0
            ],
            "colorCapabilities": 16,
            "colorTempPhysicalMin": 153,
            "colorTempPhysicalMax": 500,
            "tuyaRgbMode": 0
          }
        },
        "genOnOff": {
          "attributes": {
            "onOff": 0
          }
        },
        "genLevelCtrl": {
          "attributes": {
            "61440": 1000,
            "currentLevel": 254
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 64,
  "stackVersion": 0,
  "hwVersion": 1,
  "dateCode": "",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {
    "configured": 332242049
  },
  "lastSeen": 1725522767860
}
```
Converter:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0503B', manufacturerName: '_TZB210_nfzrlz29'}],
    model: 'WZ5_cct',
    vendor: 'Skydance',
    description: 'Skydance WZ5_cct',
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;

closed 2025-01-06 by github-actions[bot]