Koenkk/zigbee2mqtt #22132: V1.36.1 + ZBDongle-P  // Can't pair devices // Can't talk to all my devices anymore

happen-studio, 2024-04-07:
### What happened?

I installed Z2M 1 year ago, everything worked. 
I upgraded to 1.36.0 a month ago — a few days ago, everything stopped working. 
Upgrading to 1.36.1 does not help. 

No error according the USB Dongle.
Is there a option to check, if the dongle still works with the firmware? 

### Zigbee2MQTT version

1.36.1 + 1.36.0

### Adapter firmware version

20230507

### Adapter

ZBDongle-P

### Setup

plain, Proxmox, Ubuntu

### Debug log

If I want to pair a new device, there is no output.

If I want to update va value on a configured device:

Zigbee2MQTT:info  2024-04-07 23:28:40: Zigbee2MQTT started!
Zigbee2MQTT:debug 2024-04-07 23:31:15: Received MQTT message on 'zigbee2mqtt/HK-Esszimmer-Wand/set' with data '{"current_heating_setpoint":15.5}'
Zigbee2MQTT:debug 2024-04-07 23:31:15: Publishing 'set' 'current_heating_setpoint' to 'HK-Esszimmer-Wand'
Zigbee2MQTT:error 2024-04-07 23:31:40: Publish 'set' 'current_heating_setpoint' to 'HK-Esszimmer-Wand' failed: 'Error: ZCL command 0x94deb8fffec1f2d3/1 manuSpecificTuya.dataRequest({"seq":0,"dpValues":[{"dp":2,"datatype":2,"data":[0,0,0,155]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:debug 2024-04-07 23:31:40: Error: ZCL command 0x94deb8fffec1f2d3/1 manuSpecificTuya.dataRequest({"seq":0,"dpValues":[{"dp":2,"datatype":2,"data":[0,0,0,155]}]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/home/home/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.execute (/home/home/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/home/home/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.zclCommand (/home/home/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28)
    at Endpoint.command (/home/home/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:591:24)
    at sendDataPoints (/home/home/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/legacy.ts:129:5)
    at sendDataPointValue (/home/home/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/legacy.ts:179:12)
    at Object.convertSet (/home/home/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/legacy.ts:7112:13)
    at Publish.onMQTTMessage (/home/home/zigbee2mqtt/lib/extension/publish.ts:259:36)
    at EventEmitter.wrappedCallback (/home/home/zigbee2mqtt/lib/eventBus.ts:174:17)

klio-klio, 2024-04-08:
I also have the problem but only when pairing.
[image]

peterskajr, 2024-04-08:
I have a similar problem.  All  working great a few days ago, now devices become unavailable.  Restarting Zigbee2MQTT allows the devices to be online again, but only for a short time and then they will go offline again.  Not always the same devices go offline after a restart.

mrmaximas, 2024-04-08:
try this fw 20240315 https://github.com/Koenkk/Z-Stack-firmware/discussions/496
it's works good

robertomontinaro80, 2024-04-12:
Same issue for me for that getonoff command after the update of Zigbee2Mqtt on Home assistant.
I can't turn on any lamp in my home.
I add also this error stack trace if it can help

`Publish 'set' 'state' to 'osram_kitchen_lamp' failed: 'Error: ZCL command 0x7cb03eaa0a038452/3 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))`

closed 2024-11-09 by github-actions[bot]