Koenkk/zigbee2mqtt #22133: AVATTO ME167 battery_low not working

Blastmun, 2024-04-08:
### What happened?

Hello, I am having a problem with my thermostatic valves. The "battery_low" field never changes to "True". So I never see when the valve has no more battery.
Thanks

### How to reproduce it (minimal and precise)

Produc Picture: 
[https://img.tttcdn.com/product/xy/2000/2000/p/gu1/C/2/C-E0289-2/C-E0289-2-1-8031-jjP3.jpg
](url)

Product Link:
[https://www.cafago.com/fr/p-c-e0289-2.html
](url)

### Zigbee2MQTT version

1.36.1-1 

### Adapter firmware version

20211217

### Adapter

sonoff zigbee 3.0 usb dongle plus

### Setup

Proxmox VM

Blastmun, 2024-04-10:
Help?

closed 2024-11-07 by github-actions[bot]

DoWenzl94, 2025-01-11:
i have the same issue, did you have found a solution? @Blastmun