Koenkk/zigbee2mqtt #22136: Sengled E11-N1EA Color Has Been Removed

russellmoran99, 2024-04-08:
### What happened?

Hello @Koenkk , this issue has returned.  I did not know if you check on issues that are closed, so I thought I would open a new one and link to the original issue that is now closed.  I repaired all my devices yesterday and color is missing.  I have followed your instructions, and it did not work.  Another user has reported this as well this morning.  Thank you for your help.

https://github.com/Koenkk/zigbee2mqtt/issues/13365

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonos Zigbee v3 EZSP

### Setup

PI4 Zigbee2MQTT addon from Repository

Koenkk, 2024-04-08:
Should already be fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) , could you check this?

russellmoran99, 2024-04-08:
How do I go about checking it?  I am good with directions.  :)

mrskycriper, 2024-04-09:
> How do I go about checking it? I am good with directions. :)

Use the Edge version of the Home Assistant addon.

russellmoran99, 2024-04-11:
> Should already be fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) , could you check this?

@Koenkk I am using HA via Pi4.  On the link you provided I believe I am supposed to switch to Edge Version.  If I do that, will I have to repair all my devices?  I am truly not wanting to do that as I just did that less than 6 days ago.

Koenkk, 2024-04-11:
No repairing is needed.

loafbutts, 2024-04-11:
I ran into this problem today, found the issue here, and tried the Zigbee2MQTT Edge add-on in Home Assistant. It worked for me; all color controls for my Sengled bulbs are functioning normally again. Thanks!

russellmoran99, 2024-04-14:
> No repairing is needed.

I was able to install the developers edge version.  It worked great for an hour now my Sengled lights are unavailable in the home assistant dashboard.  They work in z2m edge but says “unavailable” in HA.  This seems to be another known issue I have seen when googling.  How do I fix this?  I reverted back to main branch version for now and they are available again.

russellmoran99, 2024-04-14:
> I ran into this problem today, found the issue here, and tried the Zigbee2MQTT Edge add-on in Home Assistant. It worked for me; all color controls for my Sengled bulbs are functioning normally again. Thanks!

Do the lights still function?  Mine worked for about an hour, then all Sengled lights became unavailable in Home Assistant.  Reverted back to main branch version and they’re available again.

loafbutts, 2024-04-14:
Yeah, mine have been working as expected since my last post. Sorry to hear you're having trouble again!

russellmoran99, 2024-04-14:
@Koenkk I have done further investigation and confirm my light or switch (not sure which it uses) entity does not get passed thru from Z2M Edge to HA. I can see the Sengled energy usage, but not the switch/light.

russellmoran99, 2024-04-14:
UPDATE: RESOLVED:  Fixed the issue by updating HA.  

Sent from AT&T Yahoo Mail for iPhone

closed 2024-11-11 by github-actions[bot]