Koenkk/zigbee2mqtt #22137: Sengled E21-N1EA RGBW bulbs no longer expose RGB

bat1939, 2024-04-08:
### What happened?

The Sengled E21-N1EA are no longer showing the ability to change RGB colors. Only white color temps are showing

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20210120

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Proxmox Home Assistant VM, Add-on in Home Assistant.

jrob, 2024-04-08:
I am having the same issue. 
Same version
Same Adapter
Not sure about dongle firmware version
HA running on an Intel NUC

jrob, 2024-04-08:
Reverting to 1.36.0 restored the rgb color settings.

bat1939, 2024-04-08:
> Reverting to 1.36.0 restored the rgb color settings.

I have also reverted back to 1.36.0 and rgb color is back.

Koenkk, 2024-04-08:
Already fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2024-04-08 by Koenkk

askmahesh1, 2024-05-02:
Is this merged into the main branch.

Koenkk, 2024-05-02:
Fix is available in 1.37.0