Koenkk/zigbee2mqtt #22138: Home Assistant: Sometimes Zigbee2Mqtt fails to start (Error while starting zigbee-herdsman )

MariusArmand, 2024-04-08:
### What happened?

Since a month, I noticed after a reboot of Home Assistant, the Zigbee2Mqtt addon won't boot.
In the log I see this:
[17:47:31] INFO: Preparing to start...
[17:47:31] INFO: Socat not enabled
[17:47:31] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-04-08 17:47:45: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-04-08 17:47:45: Failed to start zigbee
Zigbee2MQTT:error 2024-04-08 17:47:45: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-04-08 17:47:45: Exiting...
Zigbee2MQTT:error 2024-04-08 17:47:45: Error: Failure send getNetworkParameters:{"type":"Buffer","data":[24,0,1,40,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:664:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Driver.needsToBeInitialised (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:288:27)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:224:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

### What did you expect to happen?

I expected the addon to start as usual.
Strangely, at random, after a few tries to start the addon, the addon will start without issues.

### How to reproduce it (minimal and precise)

-Reboot Home Assistant
-Check if Zigbee2Mqtt addon was started
-If not, try to start manually
-If still not started, try again until it eventually works

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Add-on on Home Assistant on x86-64 NUC

### Debug log

[17:47:31] INFO: Preparing to start...
[17:47:31] INFO: Socat not enabled
[17:47:31] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-04-08 17:47:45: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-04-08 17:47:45: Failed to start zigbee
Zigbee2MQTT:error 2024-04-08 17:47:45: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-04-08 17:47:45: Exiting...
Zigbee2MQTT:error 2024-04-08 17:47:45: Error: Failure send getNetworkParameters:{"type":"Buffer","data":[24,0,1,40,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:664:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Driver.needsToBeInitialised (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:288:27)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:224:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

Psytoshgen, 2024-04-13:
got similar trouble on Sonoff Zigbee 3.0 USB Dongle Plus V2 

11:57:44] INFO: Preparing to start...
[11:57:44] INFO: Socat not enabled
[11:57:44] INFO: Starting Zigbee2MQTT...
[2024-04-13 11:57:46] info: 	z2m: Logging to console, file (filename: log.log)
[2024-04-13 11:57:48] info: 	z2m: Starting Zigbee2MQTT version 1.36.1-dev (commit #e10ab5c)
[2024-04-13 11:57:48] info: 	z2m: Starting zigbee-herdsman (0.42.1)
[2024-04-13 11:57:58] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
[2024-04-13 11:57:58] error: 	zh:ezsp:ezsp: Connection attempt 1 error: Error: Failure to connect
    at SerialDriver.resetForReconnect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:347:19)
    at SerialDriver.emit (node:events:517:28)
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:348:22
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at SerialDriver.openSerialPort (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:108:13)
    at SerialDriver.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:63:13)
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:353:17)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:163:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)

andyubird, 2024-04-19:
> ### What happened?
> Since a month, I noticed after a reboot of Home Assistant, the Zigbee2Mqtt addon won't boot. In the log I see this: [17:47:31] INFO: Preparing to start... [17:47:31] INFO: Socat [...]

I have the exact same setup and the same issue, and can't start even after multiple retries.

anivaros, 2024-05-02:
Same here. 

> [15:24:22] INFO: Preparing to start...
[15:24:23] INFO: Socat not enabled
[15:24:25] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-05-02 15:24:33: Logging to console and directory: '/config/zigbee2mqtt/log/2024-05-02.15-24-33' filename: log.txt
Zigbee2MQTT:info  2024-05-02 15:24:33: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
Zigbee2MQTT:info  2024-05-02 15:24:33: Starting zigbee-herdsman (0.40.3)
Zigbee2MQTT:error 2024-05-02 15:24:46: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-05-02 15:24:46: Failed to start zigbee
Zigbee2MQTT:error 2024-05-02 15:24:46: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-05-02 15:24:46: Exiting...
Zigbee2MQTT:error 2024-05-02 15:24:46: Error: Failure send getNetworkParameters:{"type":"Buffer","data":[24,0,1,40,0]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:664:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Driver.needsToBeInitialised (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:288:27)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:224:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)

Started about a month ago. Previously all worked fine.

closed 2024-11-29 by github-actions[bot]