Koenkk/zigbee2mqtt #22139: Hue wall switch module behaving incorrectly after update to 1.36.1

Joern-W, 2024-04-08:
### What happened?

After updating to 1.36.1 all my hue wall switch modules (Philips 929003017102) show strange behavior:

- short press of button results in four actions: left_press, left_press_release, left_press and left_press_release
- long press of button results in actions:  
-- left_press and left_press_release on button press 
-- left_press and left_press_release on button release
- second button stopped working completely
- light groups bind to a switch modules won't turn on or off anymore (however, the switch module is still sending action messages to the coordinator)

### What did you expect to happen?

In previous versions of Z2M the behavior was:

- short press of button results in two actions: left_press and left_press_release
- long press of button results in actions:  
-- left_press and left_hold on button press 
-- left_hold_release on button release
- second button created actions right_press and rightt_press_release; right_hold and right_hold_release
- light groups bind to a switch modules turn on or off at button press

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

zStack3x0 20230507

### Adapter

zigstar stick v4

### Setup

Add-On on HomeAssistant on Raspberry Pi 4 , HAOS

Joern-W, 2024-04-08:
Debug Log:

debug 2024-04-08 19:57:45: Loaded state from file /config/zigbee2mqtt/state.json
info  2024-04-08 19:57:45: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-08.19-57-45' filename: log.txt
debug 2024-04-08 19:57:45: Removing old log directory '/config/zigbee2mqtt/log/2024-03-20.19-52-11'
info  2024-04-08 19:57:45: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
info  2024-04-08 19:57:45: Starting zigbee-herdsman (0.40.3)
debug 2024-04-08 19:57:45: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyUSB0"}}'
info  2024-04-08 19:57:57: zigbee-herdsman started (resumed)
info  2024-04-08 19:57:57: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20230507,"transportrev":2},"type":"zStack3x0"}'
debug 2024-04-08 19:57:57: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6755}
info  2024-04-08 19:57:57: Currently 77 devices are joined:

...

info  2024-04-08 19:57:57: Bad Schalter (0x001788010cc2f228): 929003017102 - Philips Hue wall switch module (EndDevice)

...

debug 2024-04-08 19:57:58: Received MQTT message on 'zigbee2mqtt/Bad Schalter' with data '{"action":"","battery":100,"device_mode":"dual_push_button","linkquality":47}'

...

info  2024-04-08 19:58:02: MQTT publish: topic 'homeassistant/sensor/0x001788010cc2f228/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x001788010cc2f228"],"manufacturer":"Philips","model":"Hue wall switch module (929003017102)","name":"Bad Schalter","sw_version":"1.0.5","via_device":"zigbee2mqtt_bridge_0x00124b0021cc4194"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Bad Schalter","object_id":"bad_schalter_battery","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/Bad Schalter","unique_id":"0x001788010cc2f228_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2024-04-08 19:58:02: MQTT publish: topic 'homeassistant/sensor/0x001788010cc2f228/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x001788010cc2f228"],"manufacturer":"Philips","model":"Hue wall switch module (929003017102)","name":"Bad Schalter","sw_version":"1.0.5","via_device":"zigbee2mqtt_bridge_0x00124b0021cc4194"},"enabled_by_default":true,"entity_category":"diagnostic","icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/Bad Schalter","name":"Action","object_id":"bad_schalter_action","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Bad Schalter","unique_id":"0x001788010cc2f228_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
info  2024-04-08 19:58:02: MQTT publish: topic 'homeassistant/sensor/0x001788010cc2f228/device_mode/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x001788010cc2f228"],"manufacturer":"Philips","model":"Hue wall switch module (929003017102)","name":"Bad Schalter","sw_version":"1.0.5","via_device":"zigbee2mqtt_bridge_0x00124b0021cc4194"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:tune","json_attributes_topic":"zigbee2mqtt/Bad Schalter","name":"Device mode","object_id":"bad_schalter_device_mode","origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Bad Schalter","unique_id":"0x001788010cc2f228_device_mode_zigbee2mqtt","value_template":"{{ value_json.device_mode }}"}'
info  2024-04-08 19:58:02: MQTT publish: topic 'homeassistant/select/0x001788010cc2f228/device_mode/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/Bad Schalter/set/device_mode","device":{"identifiers":["zigbee2mqtt_0x001788010cc2f228"],"manufacturer":"Philips","model":"Hue wall switch module (929003017102)","name":"Bad Schalter","sw_version":"1.0.5","via_device":"zigbee2mqtt_bridge_0x00124b0021cc4194"},"enabled_by_default":true,"entity_category":"config","icon":"mdi:tune","json_attributes_topic":"zigbee2mqtt/Bad Schalter","name":"Device mode","object_id":"bad_schalter_device_mode","options":["single_rocker","single_push_button","dual_rocker","dual_push_button"],"origin":{"name":"Zigbee2MQTT","sw":"1.36.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/Bad Schalter","unique_id":"0x001788010cc2f228_device_mode_zigbee2mqtt","value_template":"{{ value_json.device_mode }}"}'
info  2024-04-08 19:58:02: MQTT publish: topic 'homeassistant/sensor/0x001788010cc2f228/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x001788010cc2f228"],"manufacturer":"Philips","model":"Hue wall switch module (929003017102)","name":"Bad Schalter","sw_version":"1.0.5","via_device":"zigbee2mqtt_bridge_0x00124b0021cc4194"},"enabled_by_default":false,"entity_categor [... cut ...]
debug 2024-04-08 20:01:14: Received Zigbee message from 'Bad Schalter', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":1,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info  2024-04-08 20:01:14: MQTT publish: topic 'zigbee2mqtt/Bad Schalter', payload '{"action":"left_press_release","battery":100,"device_mode":"dual_push_button","linkquality":65}'
info  2024-04-08 20:01:14: MQTT publish: topic 'zigbee2mqtt/Bad Schalter', payload '{"action":"","battery":100,"device_mode":"dual_push_button","linkquality":65}'
info  2024-04-08 20:01:14: MQTT publish: topic 'zigbee2mqtt/Bad Schalter/action', payload 'left_press_release'

sanghoon, 2024-04-09:
I have a similar issue.
My hue wall switch stopped working (only for the second button)

I found that an error occurs when I try to change device_mode (e.g. single_rocker, dual_rocker, ...)
```
2024-04-09 09:02:39 Publish 'set' 'device_mode' to 'sw_wall_dressroom' failed: 'Error: ZCL command 0x001788010d9f665d/1 genBasic.write({"52":{"value":0,"type":48}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4107,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 2667 - 1 - 122 - 0 - 4 after 10000ms)'
```

I also found that an error occurs when I try to re-configure the device.
```
2024-04-09 09:03:40 Failed to configure 'sw_wall_dressroom', attempt 1 (Error: Bind 0x001788010d9f665d/1 genPowerCfg from '0x00124b0024c08750/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:692:28 at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:43:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/philips.ts:2168:13) at Configure.configure (/app/lib/extension/configure.ts:122:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Info 2024-04-09 09:03:40MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"sw_wall_dressroom"},"error":"Failed to configure (Bind 0x001788010d9f665d/1 genPowerCfg from '0x00124b0024c08750/1' failed (AREQ - ZDO - bindRsp after 10000ms))","status":"error","transaction":"s7p13-1"}'
```

I hope these logs help to debug the issue

hugodejong, 2024-04-10:
Same issue for me, similar messages in the log and reported events.

Tra2x, 2024-04-11:
sadly I have the same Issue. Use only one button, still not working anymore and binding to a group isn't possible :(

krhermans, 2024-04-14:
same issue for me, restoring a full backup didn't resolve the issue either.
i'm using the Philips wall switch in dual_push_button mode
Hopefully, we can expect a quick fix or at least identify the cause of this problem.

hugodejong, 2024-04-18:
@krhermans Force-removing and repairing has not resolved this issue for me. I guess we'll have to wait until the next release?

timdonovanuk, 2024-04-21:
Same issue for me, I think they suddenly reverted back to single_rocker mode but not sure. 

These devices are the most expensive and most unreliable part of my zigbee mesh, I regret getting them. Constantly disconnect:

```
Error 2024-04-21 10:03:51Publish 'set' 'device_mode' to 'bedroom_wall_module' failed: 'Error: ZCL command 0x001788010cc4be6f/1 genBasic.write({"52":{"value":2,"type":48}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4107,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43831 - 1 - 46 - 0 - 4 after 10000ms)'
Error 2024-04-21 10:04:14Publish 'set' 'device_mode' to 'bedroom_wall_module' failed: 'Error: ZCL command 0x001788010cc4be6f/1 genBasic.write({"52":{"value":3,"type":48}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4107,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43831 - 1 - 47 - 0 - 4 after 10000ms)'
Error 2024-04-21 10:04:37Publish 'get' 'device_mode' to 'bedroom_wall_module' failed: 'Error: ZCL command 0x001788010cc4be6f/1 genBasic.read([52], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4107,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43831 - 1 - 48 - 0 - 1 after 10000ms)'
```

nversluis, 2024-04-21:
I have this exact same issue, but have also identified a cause:
If I press both buttons at the same time, I can compare their outputs:

```
info 2024-04-21 17:59:54MQTT publish: topic 'zigbee2mqtt/Schakelaars Badkamer', payload '{"action":"left_press_release","battery":80,"device_mode":"dual_rocker","linkquality":174}'
info 2024-04-21 17:59:54MQTT publish: topic 'zigbee2mqtt/Schakelaars Badkamer', payload '{"action":"","battery":80,"device_mode":"dual_rocker","linkquality":174}'
```

As you can see, the second message has an empty action payload, while the left button still correctly generates a "right_press_release" payload.

Joern-W, 2024-04-21:
@timdonovanuk that was a good point. Setting device mode to something different and then back to dual_push_button solved most of my issues.
Bindings still don't work.

I agree with you, those switches are expensive crap. However, I only know of one alternative device, that is even more expensive.

timdonovanuk, 2024-04-22:
The very odd thing is only 2 out of 4 of these devices reverted back to single_rocker modes.

Pferdebockwurst, 2024-04-27:
> @timdonovanuk that was a good point. Setting device mode to something different and then back to dual_push_button solved most of my issues. Bindings still don't work.
> 
> I agree with you, those switches are expensive crap. However, I only know of one alternative device, that is even more expensive.

I also had problems with some of these. They suddenly simply didn't respond and I had to re-pair them. I never was able to identify a reason for this behavior. But since a while, when I re-pair them, I chose the coordinator or a dedicated router for the pairing. Never had connection issues again since then. I think that a certain light bulb, of which I have a couple in my house, dont act properly as a router. So I try to avoid that new devices pair to them. The hue bulbs, for example, work great as routers.

nversluis, 2024-04-28:
I have managed to fix the issue!
I went to Exposes tab of the device menu and spamming a different mode (single_rocker in my case) and the refresh button next to it, until it is actually applied. 
You can check if the setting applied successfully by going to another tab (such as Bind) and then back to Exposes. If the setting is still the one you clicked, it has actually applied.
Then I used the same method to put i back to dual_rocker mode. Both my switches now work perfectly once more.

Joern-W, 2024-04-29:
@nversluis There is no need of "spamming" a different mode. The reason why the switch doesn't react is simple: To save battery, the switch is in sleep mode most of the time. Only occasionally, it wakes up and answers to incoming messages. Btw, this is true for most battery operated zigbee devices.
So, if you want to modify the settings of the switch, you can simply wake it up by pressing the button connected to the switch. This will make it respond to your mode change message.

nversluis, 2024-04-29:
@Joern-W I did wake up the device immediately before changing the setting and waited patiently at first. I would get error messages saying the command failed. It took quite a few retries before the command actually executed successfully. And it's not a connection issue because the left_press button still worked perfectly and quickly.

timdonovanuk, 2024-04-29:
> the switch is in sleep mode most of the time.

Yeah unfortunately these switches have narcolepsy, and sometimes go into "sleep mode" for an hour or two and refuse to respond to presses.

firstchair, 2024-09-19:
To save the setting just quickly press enter after selecting it

Bastian007, 2024-09-21:
I had execactly the same issue:
Long press of button results NOT in actions:
-- left_press and left_hold on button press
-- left_hold_release on button release

I solved the issue by removing the device in zigbee2mqtt 
and re-pairing it again. After that long press is fired again. 

I am using latest version: 1.40.1-1

chrismaki, 2024-10-29:
I was having this same problem, I ended up force-removing the device from Z2M and then re-adding and the second input started working again.

closed 2025-01-06 by github-actions[bot]