Koenkk/zigbee2mqtt #22140: ERROR : zh:controller:greenpower: Received undefined command from '0'

mrmaximas, 2024-04-08:
### What happened?

I don't have any greenpower devices, but over the last week or more once or twice per day it shows up in the log:
[2024-04-08 20:09:28] error: 	zh:controller:greenpower: Received undefined command from '0'

### What did you expect to happen?

-

### How to reproduce it (minimal and precise)

install latest devs

### Zigbee2MQTT version

1.36.1-dev commit: cabc9f5e

### Adapter firmware version

20240315

### Adapter

CC2652P7

### Setup

Mac OS Sonoma latest on M2

### Debug log

-

ghitab77, 2024-05-01:
Same issue with 1.37.1

akhosha2, 2024-05-02:
> Same issue with 1.37.1

Just updated to 1.37.0. getting this warning when allowing new items to pair. Tried to pair two GiEX GXO2s and could only get one to pair for a few minutes. I then got a message saying that the one I did get to join left my network. Now neither can join.

sharpcodex, 2024-05-02:
Same issue with 1.37.0

abastosjr, 2024-05-02:
I have the same problem, V 1.37.0

akhosha2, 2024-05-02:
So I managed to 'trick' my way into adding a new device by deleting and existing device (including the MQTT device in the integration). Then I went to pair it again with the new device and they both paired. Interestingly the existing device I deleted and repaired went back to it's original device number and confirmation.

GigiPompieru, 2024-05-02:
> I have the same problem, V 1.37.0

same issue since upgrading.

illiteratealliterator, 2024-05-03:
I'm only just installing zigbee2mqtt for the first time, and ran into this error when trying to add my first device. I'm running in docker, and I see the error in the UI (and the log) when I click the Disable/Permit Join button in the frontend.

Here's the debug logs surrounding the error, although they may not be related:

> [2024-05-03 14:18:13] debug:    zh:ezsp:ezsp: ==> {"_cls_":"sendMulticast","_id_":56,"_isRequest_":true,"apsFrame":{"profileId":41440,"sequence":30,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"groupId":65533,"options"
:256},"hops":0,"nonmemberRadius":3,"messageTag":31,"message":{"type":"Buffer","data":[25,10,2,11,254,0]}}
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: --> DATA (4,3,0): 8400013800e0a12100f2f20001fdff1e00031f06190a020bfe00
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: --> [43c621a96c2af57d337894b8d7aa546fb6824e24b4ebd76d89f6386362227e]
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: -?- waiting (5)
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: <-- [34c1b1a94d2a89b90b7e]
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: <-- DATA (3,4,0): 34c1b1a94d2a89b90b7e
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: --> ACK  (4)
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: --> [8430fc7e]
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: <-- ACK (4): 34c1b1a94d2a89b90b7e
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: Unexpected packet sequence 4 | 5
[2024-05-03 14:18:13] debug:    zh:ezsp:ezsp: <== Frame: 83900119009c
[2024-05-03 14:18:13] debug:    zh:ezsp:ezsp: <== 0x19: {"_cls_":"stackStatusHandler","_id_":25,"_isRequest_":false,"status":156}
[2024-05-03 14:18:13] debug:    zh:ezsp:driv: stackStatusHandler:
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: <-- [45c6a1a96c2a1543ad367e]
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: <-- DATA (4,5,0): 45c6a1a96c2a1543ad367e
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: --> ACK  (5)
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: --> [8520dd7e]
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: <-- ACK (5): 45c6a1a96c2a1543ad367e
[2024-05-03 14:18:13] debug:    zh:ezsp:ezsp: <== Frame: 848001380000f1
[2024-05-03 14:18:13] debug:    zh:ezsp:ezsp: <== 0x38: {"_cls_":"sendMulticast","_id_":56,"_isRequest_":false,"status":0,"sequence":241}
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: -+- waiting (5) success
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: <-- [55c6b1a97d312a1652f8b54ad7555593b463bfd8abedce9874fbdf698bf7803fa9ed7e]
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: <-- DATA (5,5,0): 55c6b1a9112a1652f8b54ad7555593b463bfd8abedce9874fbdf698bf7803fa9ed7e
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: --> ACK  (6)
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: --> [8610be7e]
[2024-05-03 14:18:13] debug:    zh:ezsp:uart: <-- ACK (5): 55c6b1a9112a1652f8b54ad7555593b463bfd8abedce9874fbdf698bf7803fa9ed7e
[2024-05-03 14:18:13] debug:    zh:ezsp:ezsp: <== Frame: 849001450003e0a12100f2ff0001fdfff1ff000000ffff06190a020bfe00
[2024-05-03 14:18:13] debug:    zh:ezsp:ezsp: <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":3,"apsFrame":{"profileId":41440,"sequence":241,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255
,"groupId":65533,"options":256},"lastHopLqi":255,"lastHopRssi":0,"sender":0,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[25,10,2,11,254,0]}}
[2024-05-03 14:18:13] debug:    zh:ezsp: processMessage: {"messageType":3,"apsFrame":{"profileId":41440,"sequence":241,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":255,"groupId":65533,"options":256},"lqi":255,"rssi":0,"sen
der":0,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[25,10,2,11,254,0]}}
[2024-05-03 14:18:13] **error:    zh:controller:greenpower: Received undefined command from '0'**
[2024-05-03 14:18:13] debug:    zh:controller: Received payload: clusterID=33, address=0, groupID=null, endpoint=242, destinationEndpoint=242, wasBroadcast=true, linkQuality=255, frame={"header":{"frameControl":{"frameType":1,"manufact
urerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":10,"commandIdentifier":2},"payload":{"options":11,"commisioningWindow":254},"command":{"ID":2,"parame
ters":[{"name":"options","type":32},{"name":"commisioningWindow","type":33}],"name":"commisioningMode"}}
[2024-05-03 14:18:13] debug:    zh:controller: Skipping command 'commisioningMode' because it is missing from the lookup
[2024-05-03 14:18:14] debug:    z2m: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"014pc-7"}'
[2024-05-03 14:18:14] debug:    zh:ezsp:uart: <-- [65c6b1a96b2a164fa674eb04aaa76d499db3d85bf3ce6701b77e]
[2024-05-03 14:18:14] debug:    zh:ezsp:uart: <-- DATA (6,5,0): 65c6b1a96b2a164fa674eb04aaa76d499db3d85bf3ce6701b77e
[2024-05-03 14:18:14] debug:    zh:ezsp:uart: --> ACK  (7)
[2024-05-03 14:18:14] debug:    zh:ezsp:uart: --> [87009f7e]
[2024-05-03 14:18:14] debug:    zh:ezsp:uart: <-- ACK (5): 65c6b1a96b2a164fa674eb04aaa76d499db3d85bf3ce6701b77e
[2024-05-03 14:18:14] debug:    zh:ezsp:ezsp: <== Frame: 8490013f0003fdffe0a12100f2ff0001fdfff01e0000
[2024-05-03 14:18:14] debug:    zh:ezsp:ezsp: <== 0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":3,"indexOrDestination":65533,"apsFrame":{"profileId":41440,"sequence":240,"clusterId":33,"sourceEndpoint":242,"d
estinationEndpoint":255,"groupId":65533,"options":256},"messageTag":30,"status":0,"message":{"type":"Buffer","data":[]}}

BitterHA, 2024-05-03:
> Can someone provide the debug logs of this?
> 
> See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

same error, here my log : 
[2024-05-03 09:19:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:22:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:25:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:29:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:32:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:35:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:39:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:42:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:45:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:49:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:52:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:55:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 09:59:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:02:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:05:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:09:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:12:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:15:34] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2024-05-03 10:15:44] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":5} after 4000ms
[2024-05-03 10:15:44] error: 	zh:ezsp:uart: -!- break waiting (5)
[2024-05-03 10:15:44] error: 	zh:ezsp:uart: Can't send DATA frame (4,2,0): 7c0001340000725804010000010100010000aeaf0518400b0a00
[2024-05-03 10:15:48] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":5} after 4000ms
[2024-05-03 10:15:48] error: 	zh:ezsp:uart: -!- break rewaiting (5)
[2024-05-03 10:15:48] error: 	zh:ezsp:uart: Can't resend DATA frame (4,2,1): 7c0001340000725804010000010100010000aeaf0518400b0a00
[2024-05-03 10:15:49] error: 	zh:controller:device: Default response to 0xa4c1385956b90aec failed
[2024-05-03 10:22:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:25:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:29:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:32:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:35:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:39:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:42:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:45:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:49:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:52:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:55:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 10:59:00] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 11:02:20] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 11:05:40] error: 	zh:controller:greenpower: Received undefined command from '0'
[2024-05-03 11:06:24] info: 	z2m: Successfully changed options
[2024-05-03 11:06:24] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"7yx2t-1"}'
[2024-05-03 11:06:24] debug: 	zh:ezsp:ezsp: Time to watchdog ... 0
[2024-05-03 11:06:24] debug: 	zh:ezsp:ezsp: ==> nop: null
[2024-05-03 11:06:24] debug: 	zh:ezsp:ezsp: ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
[2024-05-03 11:06:24] debug: 	zh:ezsp:uart: --> DATA (4,2,0): ac00010500
[2024-05-03 11:06:24] debug: 	zh:ezsp:uart: --> [42ee21a9512ac1b57e]
[2024-05-03 11:06:24] debug: 	zh:ezsp:uart: -?- waiting (5)
[2024-05-03 11:06:24] debug: 	zh:ezsp:uart: <-- [25eea1a9512a8bd47e]
[2024-05-03 11:06:24] debug: 	zh:ezsp:uart: <-- DATA (2,5,0): 25eea1a9512a8bd47e
[2024-05-03 11:06:24] debug: 	zh:ezsp:uart: --> ACK  (3)
[2024-05-03 11:06:24] debug: 	zh:ezsp:uart: --> [83401b7e]
[2024-05-03 11:06:24] debug: 	zh:ezsp:uart: <-- ACK (5): 25eea1a9512a8bd47e
[2024-05-03 11:06:24] debug: 	zh:ezsp:ezsp: <== Frame: ac80010500
[2024-05-03 11:06:24] debug: 	zh:ezsp:ezsp: <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
[2024-05-03 11:06:24] debug: 	zh:ezsp:uart: -+- waiting (5) success
[2024-05-03 11:06:32] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/PLUG DARTS', payload '{"child_lock":"UNLOCK","current":0,"energy":0.94,"indicator_mode":"off/on","linkquality":100,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available [... cut ...]
[2024-05-03 11:08:22] debug: 	zh:ezsp:uart: <-- ACK (6): 368fb1a96b2a155eaa904b27ad5493499d4e277fdace67f4e77e
[2024-05-03 11:08:22] debug: 	zh:ezsp:ezsp: <== Frame: cd90013f0000ecf304010207010100010000d4370000
[2024-05-03 11:08:22] debug: 	zh:ezsp:ezsp: <== 0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":0,"indexOrDestination":62444,"apsFrame":{"profileId":260,"sequence":212,"clusterId":1794,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":55,"status":0,"message":{"type":"Buffer","data":[]}}
[2024-05-03 11:08:22] debug: 	z

lettore, 2024-05-03:
Same for me when I start/stop pairing, pairing works well anyway.
Versione Zigbee2MQTT
[1.37.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.0) commit: [46f34c8](https://github.com/Koenkk/zigbee2mqtt/commit/46f34c8)
Tipo Coordinator
EZSP v13
Revisione Coordinator
7.4.2.0 build 0
Indirizzo IEEE Coordinatore
0xe0798dfffeb31b38
Versione frontend
0.6.165
19.32.0
0.45.0

litecross91, 2024-05-03:
same here
1.37.0-1 
usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2
Coordinator firmware version: '{"meta":{"maintrel":"7 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.7.0 build 435"},"type":"EZSP v8"}'
Core
2024.4.4
Supervisor
2024.04.4
Operating System
12.2
Frontend
20240404.2

tibur79, 2024-05-04:
Same issue since upgrading. Received undefined command from '0'

litecross91, 2024-05-04:
Check this: https://github.com/Koenkk/zigbee2mqtt/issues/22226#issuecomment-2092769288

Koenkk, 2024-05-04:
Does this error disappear after using the ember drivers instead? ([docs](https://github.com/Koenkk/zigbee2mqtt/discussions/21462))

tibur79, 2024-05-04:
Hi, I updated everything as per the guide and managed to add the new devices but the green wording remains, thanks

VladislavVesely, 2024-05-04:
Same issue here

rboutan, 2024-05-05:
Bonjour,
Voici le message depuis la dernière mise à jour.
Merci.

[image]

tibur79, 2024-05-05:
oui merci l'erreur est celle de la photo mais depuis que j'ai mis à jour j'ai réussi à connecter le nouvel appareil dans la semaine je recevrai les nouveaux appareils et je vous tiendrai au courant de leur appairage

rboutan, 2024-05-05:
J'ai restauré l'ancienne version et l'erreur a disparu.
Je vais attendre un peu avant de refaire une MAJ.
Merci pour l'info.

rafpigna, 2024-05-05:
Any pairing works fine for me, no issues at all when adding new devices to the network, but the error is always there. I also have other bunch of errors in the log about CRC or other data related issues, but I will open a new issue for that.

CoderJoeyMoment, 2024-05-05:
+1

Zigbee2MQTT version
1.37.0-dev commit: beedfae / 1.37.0 stable (both)
Coordinator type
EZSP v12
Coordinator revision
7.3.1.0 build 176
Coordinator IEEE Address
0xe0798dfffe999c5a

Koenkk, 2024-05-05:
Does this happen when permitting join or when you have `permit_join: true` in your `configuration.yaml`?

akhosha2, 2024-05-05:
> Does this happen when permitting join or when you have `permit_join: true` in your `configuration.yaml`?

When I'm permitting join.

lordforall, 2024-05-06:
Same issue for me.

Latest updates for everything including firmware controller and also trying ember driver.
Can no longer add any devices and just get the error when clicking permit to join

zhongqhang, 2024-05-06:
i use xiaomi gataway 3 
same problem
[image]

Koenkk, 2024-05-06:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-06 by Koenkk

litecross91, 2024-05-06:
> Fixed!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Sorry but how can we install it? In HACS I can "Redownload" it and choose it but in Addons doesn't. Where is the dev branch?

GigiPompieru, 2024-05-06:
> > Fixed!
> > Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.
> 
> Sorry but how can we install it? In HACS I can "Redownload" it and choose it but in Addons doesn't. Where is the dev branch?

you should wait for it to make it to the main one