Koenkk/zigbee2mqtt #22144: The brightness is changing randomly on turn on/off for FUT035Z+

bzumik1, 2024-04-08:
### What happened?

Hi, I am having [FUT035Z+](https://www.zigbee2mqtt.io/devices/FUT035Z%2B.html#miboxer-fut035z%252B) which should be fully supported device. I am facing issue that when the light is turned on the brightness start to change randomly which not only looks bad on the dashboard but also causes problems to my automations because the state of the light is always updated. This behaviour stops after 10-20s, but the final brightness is also random and does not correspond to real state of the controller. After few tries I am also
[image]
 getting this error and the light stop working. I am not sure if that can have something to do with that. 
[image]

[image]

[image]

### What did you expect to happen?

I expect this light to work normally.

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Add-on on HAOS running on Intel NUC

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14911142/log.txt)

bzumik1, 2024-04-10:
Hi, I am new on GitHub, should I somehow chase this issue or should I keep it like this and someone will pick it up?

centeljon, 2024-04-13:
I can confirm this issue - i see the same behavior. i downgraded to 1.36.0 which runs errorfree for this device.

bzumik1, 2024-04-13:
Thanks for response, let me try that.

Koenkk, 2024-04-15:
Fixed in https://github.com/Koenkk/zigbee2mqtt/issues/22078 !

closed 2024-04-15 by Koenkk