Koenkk/zigbee2mqtt #22148: Aqara Double Relay T2 (LLKZMK12LM) - incorrect Energy(Sum of consumed energy) measuring unit.

george-oniceag, 2024-04-09:
### What happened?

Device reports extremly high energy consuption . From what i noticed the Energy(Sum of consumed energy) reported in kWh increments very fast while Power(Instantaneous measured power) has a constant of 0W with 4W in peak consumption.(a 4W Bulb) I think that the Energy value should be reported in **Wh (not kWh)**!

[image]

Device state:
`{
    "**consumption**": 717.3026123046875,
    "current": 0.05,
    "device_temperature": 35,
    "**energy**": 717.3,
    "interlock": "OFF",
    "last_seen": 1712661486883,
    "led_disabled_night": false,
    "linkquality": 208,
    "mode": "power",
    "operation_mode_l1": "control_relay",
    "operation_mode_l2": "control_relay",
    "power": 0.58,
    "power_on_behavior": "previous",
    "power_outage_count": 21,
    "pulse_length": 200,
    "state": "OFF",
    "state_l1": "OFF",
    "state_l2": "ON",
    "switch_type": "momentary",
    "switch_type_l1": "momentary",
    "update": {
        "installed_version": 27,
        "latest_version": 30,
        "state": "available"
    },
    "voltage": 230.75,
    "action": null,
    "device": {
        "applicationVersion": 27,
        "dateCode": "20230606",
        "friendlyName": "LightSwitch_BaiePDepo",
        "hardwareVersion": 1,
        "ieeeAddr": "0x54ef44100096b25d",
        "manufacturerID": 4447,
        "manufacturerName": "Aqara",
        "model": "LLKZMK12LM",
        "networkAddress": 28787,
        "powerSource": "DC Source",
        "stackVersion": 27,
        "type": "Router",
        "zclVersion": 3
    },
    "update_available": null
}`

All 3 devices have that i have added **2 weeks ago** have the same issue...

### What did you expect to happen?

Expected a 714W usage over the last 2 week. 

### How to reproduce it (minimal and precise)

Add an aqara double relay T2 with no additional configuration.

Zigbee2mqtt(Sonoff Zigbee adapter) + mqtt + HA  ( in docker swarm ) 

### Zigbee2MQTT version

 1.36.0 commit: 86ed71c

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

### Setup

Docker Swarm on CentOS 9 Stream Virtual Machine with USB passtrough

george-oniceag, 2024-04-09:
related with: https://github.com/Koenkk/zigbee-herdsman-converters/issues/6889

Koenkk, 2024-04-09:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-09 by Koenkk