Koenkk/zigbee2mqtt #22149: No Herdsman logging under Windows with 1.36.1-dev commit: c98e5a74

okastl, 2024-04-09:
### What happened?

Herdsman logging no longer works under Windows with 1.36.1-dev commit: c98e5a74

On abort (ctrl-c) this is thrown:

D:\NB\z2m\zigbee2mqtt\node_modules\winston\node_modules\readable-stream\lib\_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (D:\NB\z2m\zigbee2mqtt\node_modules\winston\node_modules\readable-stream\lib\_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (D:\NB\z2m\zigbee2mqtt\node_modules\winston\node_modules\readable-stream\lib\_stream_writable.js:300:21)
    at DerivedLogger.log (D:\NB\z2m\zigbee2mqtt\node_modules\winston\lib\winston\logger.js:252:14)
    at DerivedLogger.<computed> [as info] (D:\NB\z2m\zigbee2mqtt\node_modules\winston\lib\winston\create-logger.js:95:19)
    at Logger.info (D:\NB\z2m\zigbee2mqtt\lib\util\logger.ts:158:21)
    at Znp.onPortClose (D:\NB\z2m\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\znp\znp.ts:113:16)
    at Object.onceWrapper (node:events:629:26)
    at Socket.emit (node:events:526:35)
    at TCP.<anonymous> (node:net:337:12)

### What did you expect to happen?

Should work as before

### How to reproduce it (minimal and precise)

Start z2m with herdsman logging:

set DEBUG=zigbee-herdsman*
node index.js
(or npm start)

### Zigbee2MQTT version

1.36.1-dev commit: c98e5a74

### Adapter firmware version

n/a

### Adapter

n/a

### Setup

Debugging on Windows plain

Koenkk, 2024-04-09:
`set DEBUG=zigbee-herdsman*` is not needed anymore, just setting `log_level: debug` is enough.

okastl, 2024-04-09:
> `set DEBUG=zigbee-herdsman*` is not needed anymore, just setting `log_level: debug` is enough.

Oh, and it logs to the normal logfile? Sweet! I didn't notice, because I didn't look there.

closed 2024-04-09 by okastl