Koenkk/zigbee2mqtt #22150: Problem with external device definitions

han-alink, 2024-04-09:
Can you please create a new issue and share all external definitions there?

_Originally posted by @mrskycriper in https://github.com/Koenkk/zigbee2mqtt/discussions/22023#discussioncomment-9060436_

han-alink, 2024-04-09:
Here is the file content (renamed to .txt)

[ha_devices.txt](https://github.com/Koenkk/zigbee2mqtt/files/14920530/ha_devices.txt)

schauveau, 2024-04-11:
You can run `node` on any external converter file to verify that is compiles properly. That only works if the file is within the `/opt/zigbee2mqtt` directory structure.   

```
#  node  /opt/zigbee2mqtt/ha_devices.txt
/opt/zigbee2mqtt/ha_devices.txt:439
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), temperature({"endpointNames":["1"]}), batteryPercentage(), humidity({"endpointNames":["2"], reporting: {min: 300, max: 60000, change: 10}})]
                                                                                                 ^
TypeError: batteryPercentage is not a function
    at Object.<anonymous> (/opt/zigbee2mqtt/ha_devices.txt:439:98)
    at Module._compile (node:internal/modules/cjs/loader:1376:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1435:10)
    at Module.load (node:internal/modules/cjs/loader:1207:32)
    at Module._load (node:internal/modules/cjs/loader:1023:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:135:12)
    at node:internal/main/run_main_module:28:49

Node.js v20.11.1
```

Here the problem is obviously with `batteryPercentage`.  The symbol is supposed to be exported by  `zigbee-herdsman-converters/lib/modernExtend.js` but I see no trace of `batteryPercentage in the file. 

However, there is a `battery` function in the exports and one of its argument is `percentage`.

han-alink, 2024-04-14:
> Can you please create a new issue and share all external definitions there?
> 
> _Originally posted by @mrskycriper in [#22023 (reply in thread)](https://github.com/Koenkk/zigbee2mqtt/discussions/22023#discussioncomment-9060436)_

Hi Sergei,
Why did you ask me to create a new link and publish my device configuarion file? I actually hoped to receive some help with the update of the device descriptions.

closed 2024-11-11 by github-actions[bot]