Koenkk/zigbee2mqtt #22151: New device support for MEG5779 from Schneider Electric

SupperCraft5000, 2024-04-09:
### Link

https://www.se.com/de/de/product/MEG5779-0319/connected-raumtemperaturreglermodul-polarweiss-gl%C3%A4nzend-system-m-zb/

### Database entry

{"id":14,"type":"Router","ieeeAddr":"0x040d84fffe1a721c","nwkAddr":4914,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"MEG5779","epList":[1,2,4,5,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,513,516,2821,65027,65302,65315],"outClusterList":[3,10,25,1026,1030],"clusters":{"genBasic":{"attributes":{"modelId":"MEG5779","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":8}},"msOccupancySensing":{"attributes":{}},"seMetering":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":770,"inClusterList":[0,3,1026],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2332}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b72318","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null}],"meta":{}},"4":{"profId":260,"epId":4,"devId":263,"inClusterList":[0,3,1030],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[{"cluster":1030,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b72318","endpointID":1}],"configuredReportings":[{"cluster":1030,"attrId":0,"minRepIntval":10,"maxRepIntval":60,"repChange":0,"manufacturerCode":null}],"meta":{}},"5":{"profId":260,"epId":5,"devId":13,"inClusterList":[0,3,4,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"seMetering":{"attributes":{"divisor":1000,"multiplier":1,"instantaneousDemand":0,"currentSummDelivered":[0,0]}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b72318","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":1024,"minRepIntval":10,"maxRepIntval":65000,"repChange":5000,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":[0,100],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1712678689979}

### Comments

I'm not skilled enough to add a new Device to zigbee2mqtt so i need a bit of help.

### External definition

```shell
const {deviceEndpoints, temperature, occupancy, electricityMeter, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['MEG5779'],
    model: 'MEG5779',
    vendor: 'Schneider Electric',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"4":4,"5":5,"242":242}}), temperature({"endpointNames":["2"]}), occupancy(), electricityMeter({"cluster":"metering"}), identify()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

chrissileinus, 2024-05-24:
Hey together,

I wrote a working external converter for the  MEG5779. But it`s is far away from perfect.

    const {} = require('zigbee-herdsman-converters/lib/modernExtend');
    const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
    const tz = require('zigbee-herdsman-converters/converters/toZigbee');
    const exposes = require('zigbee-herdsman-converters/lib/exposes');
    const reporting = require('zigbee-herdsman-converters/lib/reporting');
    const e = exposes.presets;
    const ea = exposes.access;
    
    const definition = {
        zigbeeModel: ['MEG5779'],
        model: 'MEG5779',
        vendor: 'Schneider Electric',
        description: 'Smart thermostat',
        fromZigbee: [
            fz.stelpro_thermostat, 
            fz.metering, 
            fz.schneider_pilot_mode, 
            fz.wiser_device_info, 
            fz.hvac_user_interface, 
            fz.temperature
        ],
        toZigbee: [
            tz.thermostat_occupied_heating_setpoint, 
            tz.thermostat_system_mode, 
            tz.thermostat_running_state,
            tz.thermostat_local_temperature, 
            tz.thermostat_control_sequence_of_operation, 
            tz.schneider_pilot_mode,
            tz.schneider_thermostat_keypad_lockout, 
            tz.thermostat_temperature_display_mode
        ],
        exposes: [
            e.binary('keypad_lockout', ea.STATE_SET, 'lock', 'unlock')
                .withDescription('Enables/disables physical input on the device'),
            e.enum('schneider_pilot_mode', ea.ALL, ['contactor', 'pilot'])
                .withDescription('Controls piloting mode'),
            e.enum('temperature_display_mode', ea.ALL, ['celsius', 'fahrenheit'])
                .withDescription('The temperature format displayed on the thermostat screen'),
            e.climate()
                .withSetpoint('occupied_heating_setpoint', 4, 30, 0.5)
                .withLocalTemperature()
                .withSystemMode(['off', 'heat'])
                .withRunningState(['idle', 'heat'])
                .withPiHeatingDemand(),
            e.temperature()
        ],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            const endpoint2 = device.getEndpoint(2);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['hvacThermostat']);
            await reporting.thermostatPIHeatingDemand(endpoint1);
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint1);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['msTemperatureMeasurement']);
            await reporting.temperature(endpoint2);
            await endpoint1.read('hvacUserInterfaceCfg', ['keypadLockout', 'tempDisplayMode']);
        },
    };
    
    module.exports = definition;

huco95, 2024-09-20:
> Hey together,
>
> I wrote a working external converter for the MEG5779. But it`s is far away from perfect.
>
> ```
> const {} = require('zigbee-herdsman-converters/lib/modernExtend');
> const fz = [...]

I will try this, thanks!! Did you make any improvements on it?

chrissileinus, 2024-09-23:
> I will try this, thanks!! Did you make any improvements on it?

Hey,

I didn't have time to improve it. But I also haven't had problems with it. 😉

huco95, 2024-09-23:
> 
> Hey,
> 
> I didn't have time to improve it. But I also haven't had problems with it. 😉

For me everything is working except the running state, I can set the temperature but I do not know if it has start heating or not. Does it work for you?

chrissileinus, 2024-09-24:
> > Hey,
> > I didn't have time to improve it. But I also haven't had problems with it. 😉
> 
> For me everything is working except the running state, I can set the temperature but I do not know if it has start heating or not. Does it work for you?

This is strange. I see the state changing.

huco95, 2024-09-25:
@chrissileinus I finally managed to make it work. Did you know the difference between contactor and pilot mode? I don't know what this setting if for 😅

chrissileinus, 2024-09-26:
@huco95 I also was unable to figure it out. I think it is a kind of operation mode. My plan is to play with it again at this upcoming heating period. I hope that I get it working properly. I'll let you know.

mad-scientist, 2024-12-01:
Hi folks, I am super happy I found this post. I just installed the definition and will see how it performs. I am in principle experienced in programming, but so far clueless about the Z2M concept. 
I have been using ZHA before, and using the ZHA Tools was able to change the display settings. I like to have the display permanently showing the temperature at a low brightness setting. If I can find the time and some guidance, I will try work on this over Christmas ...

chrissileinus, 2024-12-02:
Hey Mad, I looking forward to see some improvements. Happy thinkering!

mad-scientist, 2024-12-09:
My brain is overheating trying to understand how this works! I start to get a feel how the code is structured, but I am still light years away from making any change ...

Can you say why there are two temperature values - one as the local temperature in the climate device in endpoint 1 and then another as separate measurement in endpoint 2? The second looks fine, while the first one in the climate seems never to update?
[image]

mad-scientist, 2025-01-09:
Hey guys, this is hard stuff. There is very little documentation, and I had to understand a lot of basics first.

But: I got the display brightness settings working! 🎉 You can now adjust the display brightness during both active and inactive states.

I also found that a number of other features not really working, eg. the keypad lockout, running state. Will keep you posted!

```
const { deviceAddCustomCluster, numeric } = require('zigbee-herdsman-converters/lib/modernExtend');
const Zcl = require('zigbee-herdsman').Zcl;
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const megExtend = {
    addDisplayConfigurationCluster: () =>
        deviceAddCustomCluster('meghvacUserInterfaceCfg', {
            ID: 0x0204,
            manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
            attributes: {
                displayBrightnessActive: {ID: 0xe000, type: Zcl.DataType.UINT8},
                displayBrightnessInactive: {ID: 0xe001, type: Zcl.DataType.UINT8},
                displayActiveTimeout: {ID: 0xe002, type: Zcl.DataType.UINT16},
            },
            commands: {},
            commandsResponse: {},
        }),
    displayBrightnessActive: () =>
        numeric({
            name: 'display_brightness_active',
            cluster: 'meghvacUserInterfaceCfg',
            attribute: 'displayBrightnessActive',
            description: 'Sets brightness of the temperature display during active state',
            entityCategory: 'config',
            unit: '%',
            valueMin: 0,
            valueMax: 100,
            valueStep: 1,
        }),
    displayBrightnessInactive: () =>
        numeric({
            name: 'display_brightness_inactive',
            cluster: 'meghvacUserInterfaceCfg',
            attribute: 'displayBrightnessInactive',
            description: 'Sets brightness of the temperature display during inactive state',
            entityCategory: 'config',
            unit: '%',
            valueMin: 0,
            valueMax: 100,
            valueStep: 1,
        }),
    displayActiveTimeout: () =>
        numeric({
            name: 'display_active_timeout',
            cluster: 'meghvacUserInterfaceCfg',
            attribute: 'displayActiveTimeout',
            description: 'Sets timeout of the temperature display active state',
            entityCategory: 'config',
            unit: 'seconds',
            valueMin: 5,
            valueMax: 600,
            valueStep: 5,
        }),
};

const definition = {
    zigbeeModel: ['MEG5779'],
    model: 'MEG5779',
    vendor: 'Schneider Electric',
    description: 'Merten Connected Room Temperature Controller',
    fromZigbee: [
        fz.stelpro_thermostat, 
        fz.metering, 
        fz.schneider_pilot_mode, 
        fz.wiser_device_info, 
        fz.hvac_user_interface, 
        fz.temperature,
    ],
    toZigbee: [
        tz.thermostat_occupied_heating_setpoint, 
        tz.thermostat_system_mode, 
        tz.thermostat_running_state,
        tz.thermostat_local_temperature, 
        tz.thermostat_control_sequence_of_operation, 
        tz.schneider_pilot_mode,
        tz.schneider_thermostat_keypad_lockout, 
        tz.thermostat_temperature_display_mode,
    ],
    exposes: [
        e.binary('keypad_lockout', ea.STATE_SET, 'lock', 'unlock')
            .withDescription('Enables/disables physical input on the device'),
        e.enum('schneider_pilot_mode', ea.ALL, ['contactor', 'pilot'])
            .withDescription('Controls piloting mode'),
        e.enum('temperature_display_mode', ea.ALL, ['celsius', 'fahrenheit'])
            .withDescription('The temperature format displayed on the thermostat screen'),
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 4, 30, 0.5)
            .withLocalTemperature()
            .withSystemMode(['off', 'heat'])
            .withRunningState(['idle', 'heat'])
            .withPiHeatingDemand(),
        e.temperature(),
    ],
    extend: [ 
        megExtend.addDisplayConfigurationCluster(), 
        megExtend.displayBrightnessActive(),
        megExtend.displayBrightnessInactive(),
        megExtend.displayActiveTimeout(),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['hvacThermostat', 'meghvacUserInterfaceCfg']);
        await reporting.thermostatPIHeatingDemand(endpoint1);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint1);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['msTemperatureMeasurement']);
        await reporting.temperature(endpoint2);
        await endpoint1.read('hvacUserInterfaceCfg', ['keypadLockout', 'tempDisplayMode']);
        await endpoint1.read('meghvacUserInterfaceCfg', ['displayBrightnessActive']);
        await endpoint1.read('meghvacUserInterfaceCfg', ['displayBrightnessInactive']);
        await endpoint1.read('meghvacUserInterfaceCfg', ['displayActiveTimeout']);
    },
};

module.exports = definition;
```

chrissileinus, 2025-03-11:
Hey mad-scientist,

I used your code now over a month. And i`m really happy with it. Your definition runs stable for me.

big thanks
Christian

MarkusElsing, 2025-04-21:
Hi, I tried to add support for cooling, works on my test device. The modified converter is:

`
const { deviceAddCustomCluster, numeric } = require('zigbee-herdsman-converters/lib/modernExtend');
const Zcl = require('zigbee-herdsman').Zcl;
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

console.log("MEG5779 converter loaded");  // Log during startup

const megExtend = {
   addDisplayConfigurationCluster: () =>
       deviceAddCustomCluster('meghvacUserInterfaceCfg', {
           ID: 0x0204,
           manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
           attributes: {
               displayBrightnessActive: {ID: 0xe000, type: Zcl.DataType.UINT8},
               displayBrightnessInactive: {ID: 0xe001, type: Zcl.DataType.UINT8},
               displayActiveTimeout: {ID: 0xe002, type: Zcl.DataType.UINT16},
           },
           commands: {},
           commandsResponse: {},
       }),
   displayBrightnessActive: () =>
       numeric({
           name: 'display_brightness_active',
           cluster: 'meghvacUserInterfaceCfg',
           attribute: 'displayBrightnessActive',
           description: 'Sets brightness of the temperature display during active state',
           entityCategory: 'config',
           unit: '%',
           valueMin: 0,
           valueMax: 100,
           valueStep: 1,
       }),
   displayBrightnessInactive: () =>
       numeric({
           name: 'display_brightness_inactive',
           cluster: 'meghvacUserInterfaceCfg',
           attribute: 'displayBrightnessInactive',
           description: 'Sets brightness of the temperature display during inactive state',
           entityCategory: 'config',
           unit: '%',
           valueMin: 0,
           valueMax: 100,
           valueStep: 1,
       }),
   displayActiveTimeout: () =>
       numeric({
           name: 'display_active_timeout',
           cluster: 'meghvacUserInterfaceCfg',
           attribute: 'displayActiveTimeout',
           description: 'Sets timeout of the temperature display active state',
           entityCategory: 'config',
           unit: 'seconds',
           valueMin: 5,
           valueMax: 600,
           valueStep: 5,
       }),
};

const definition = {
   zigbeeModel: ['MEG5779'],
   model: 'MEG5779',
   vendor: 'Schneider Electric',
[... 45 lines cut ...]
       megExtend.displayBrightnessInactive(),
       megExtend.displayActiveTimeout(),
   ],
   configure: async (device, coordinatorEndpoint, logger) => {
       const endpoint1 = device.getEndpoint(1);
       const endpoint2 = device.getEndpoint(2);
       await reporting.bind(endpoint1, coordinatorEndpoint, ['hvacThermostat', 'meghvacUserInterfaceCfg']);
       await reporting.thermostatPIHeatingDemand(endpoint1);
       await reporting.thermostatPICoolingDemand(endpoint1); // add cooling support
       await reporting.thermostatOccupiedHeatingSetpoint(endpoint1);
       await reporting.thermostatOccupiedCoolingSetpoint(endpoint1); // add cooling sopport
       await reporting.bind(endpoint2, coordinatorEndpoint, ['msTemperatureMeasurement']);
       await reporting.temperature(endpoint2);
       await endpoint1.read('hvacUserInterfaceCfg', ['keypadLockout', 'tempDisplayMode']);
       await endpoint1.read('meghvacUserInterfaceCfg', ['displayBrightnessActive']);
       await endpoint1.read('meghvacUserInterfaceCfg', ['displayBrightnessInactive']);
       await endpoint1.read('meghvacUserInterfaceCfg', ['displayActiveTimeout']);
   },
};

module.exports = definition;

`

Still a few issues with Home Assistant discovery method in Openhab, working on this next.

mad-scientist, 2025-04-21:
I had not posted this before -- I have removed all the copy&paste stuff from other devices, where I could after several evenings of analysis not identify a specific purpose for the MEG5779. For example, the weird additional temperature sensor is now gone. I am using this code for the past three months now without any issue.

@MarkusElsing if would be neat if you could please merge this with your additions?

```
const { deviceAddCustomCluster, numeric } = require('zigbee-herdsman-converters/lib/modernExtend');
const Zcl = require('zigbee-herdsman').Zcl;
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = { manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC };

const megExtend = {
    hvacUserInterfaceCfgCluster: () =>
        deviceAddCustomCluster('hvacUserInterfaceCfg', {
            ID: Zcl.Clusters.hvacUserInterfaceCfg.ID,
            attributes: {
                displayBrightnessActive: {
                    ID: 0xe000, 
                    type: Zcl.DataType.UINT8,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
                displayBrightnessInactive: {
                    ID: 0xe001, 
                    type: Zcl.DataType.UINT8,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
                displayActiveTimeout: {
                    ID: 0xe002, 
                    type: Zcl.DataType.UINT16,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
            },
            commands: {},
            commandsResponse: {},
        }),
    displayBrightnessActive: () =>
        numeric({
            name: 'display_brightness_active',
            cluster: 'hvacUserInterfaceCfg',
            attribute: 'displayBrightnessActive',
            description: 'Sets brightness of the temperature display during active state',
            entityCategory: 'config',
            unit: '%',
            valueMin: 0,
            valueMax: 100,
            valueStep: 1,
            zigbeeCommandOptions: manufacturerOptions,
        }),
    displayBrightnessInactive: () =>
        numeric({
            name: 'display_brightness_inactive',
            cluster: 'hvacUserInterfaceCfg',
            attribute: 'displayBrightnessInactive',
            description: 'Sets brightness of the temperature display during inactive state',
            entityCategory: 'config',
            unit: '%',
            valueMin: 0,
            valueMax: 100,
            valueStep: 1,
            zigbeeCommandOptions: manufacturerOptions,
        }),
    displayActiveTimeout: () =>
        numeric({
            name: 'display_active_timeout',
            cluster: 'hvacUserInterfaceCfg',
            attribute: 'displayActiveTimeout',
            description: 'Sets timeout of the temperature display active state',
            entityCategory: 'config',
            unit: 'seconds',
            valueMin: 5,
            valueMax: 600,
            valueStep: 5,
            zigbeeCommandOptions: manufacturerOptions,
        }),
};

const definition = {
    zigbeeModel: ['MEG5779'],
    model: 'MEG5779',
    vendor: 'Schneider Electric',
    description: 'Merten Connected Room Temperature Controller',
    fromZigbee: [
        fz.thermostat, 
        fz.hvac_user_interface
    ],
    toZigbee: [
        tz.thermostat_occupied_heating_setpoint, 
        tz.thermostat_system_mode, 
        tz.thermostat_running_state,
        tz.thermostat_local_temperature, 
    ],
    exposes: [
        e.climate()
            .withLocalTemperature()
            .withSetpoint('occupied_heating_setpoint', 4, 30, 0.5)
            .withSystemMode(['off', 'heat'])
            .withRunningState(['idle', 'heat'])
            .withPiHeatingDemand(),
    ],
    extend: [ 
        megExtend.hvacUserInterfaceCfgCluster(),
        megExtend.displayBrightnessActive(),
        megExtend.displayBrightnessInactive(),
        megExtend.displayActiveTimeout(),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['hvacThermostat', 'hvacUserInterfaceCfg']);
        await reporting.thermostatPIHeatingDemand(endpoint1);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint1);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['msTemperatureMeasurement']);
        await reporting.temperature(endpoint2);
        await endpoint1.read('hvacUserInterfaceCfg', ['displayBrightnessActive', 'displayBrightnessInactive', 'displayActiveTimeout'], manufacturerOptions);
    },
};

module.exports = definition;
```

MarkusElsing, 2025-04-22:
Hi, here is my version, works for me, not 100% happy yet. I am looking for things like "away mode" and calibration to complete the functionality.

`
const { deviceAddCustomCluster, numeric } = require('zigbee-herdsman-converters/lib/modernExtend');
const Zcl = require('zigbee-herdsman').Zcl;
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = { manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC };

const megExtend = {
    hvacUserInterfaceCfgCluster: () =>
        deviceAddCustomCluster('hvacUserInterfaceCfg', {
            ID: Zcl.Clusters.hvacUserInterfaceCfg.ID,
            attributes: {
                displayBrightnessActive: {
                    ID: 0xe000, 
                    type: Zcl.DataType.UINT8,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
                displayBrightnessInactive: {
                    ID: 0xe001, 
                    type: Zcl.DataType.UINT8,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
                displayActiveTimeout: {
                    ID: 0xe002, 
                    type: Zcl.DataType.UINT16,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
            },
            commands: {},
            commandsResponse: {},
       }),
   displayBrightnessActive: () =>
       numeric({
           name: 'display_brightness_active',
           cluster: 'hvacUserInterfaceCfg',
           attribute: 'displayBrightnessActive',
           description: 'Sets brightness of the temperature display during active state',
           entityCategory: 'config',
           unit: '%',
           valueMin: 0,
           valueMax: 100,
           valueStep: 1,
           zigbeeCommandOptions: manufacturerOptions,
       }),
   displayBrightnessInactive: () =>
       numeric({
           name: 'display_brightness_inactive',
           cluster: 'hvacUserInterfaceCfg',
           attribute: 'displayBrightnessInactive',
           description: 'Sets brightness of the temperature display during inactive state',
           entityCategory: 'config',
           unit: '%',
           valueMin: 0,
           valueMax: 100,
           valueStep: 1,
           zigbeeCommandOptions: manufacturerOptions,
       }),
   displayActiveTimeout: () =>
       numeric({
           name: 'display_active_timeout',
           cluster: 'hvacUserInterfaceCfg',
           attribute: 'displayActiveTimeout',
[... 49 lines cut ...]
            .withDescription('Position of the valve (= demanded cooling) where 0% is fully closed and 100% is fully open'),
   ],
   extend: [
       megExtend.hvacUserInterfaceCfgCluster(),
       megExtend.displayBrightnessActive(),
       megExtend.displayBrightnessInactive(),
       megExtend.displayActiveTimeout(),
   ],
   configure: async (device, coordinatorEndpoint, logger) => {
       const endpoint1 = device.getEndpoint(1);
       await reporting.bind(endpoint1, coordinatorEndpoint, ['hvacThermostat', 'hvacUserInterfaceCfg']);
       await reporting.thermostatPIHeatingDemand(endpoint1);
       await reporting.thermostatPICoolingDemand(endpoint1); // add cooling support
       await reporting.thermostatOccupiedHeatingSetpoint(endpoint1);
       await reporting.thermostatOccupiedCoolingSetpoint(endpoint1); // add cooling sopport
       await reporting.temperature(endpoint2);
       await endpoint1.read('hvacUserInterfaceCfg',
			    ['keypadLockout','tempDisplayMode','displayBrightnessActive','displayBrightnessInactive','displayActiveTimeout'],
			    manufacturerOptions);
   },
};

module.exports = definition;

`

MarkusElsing, 2025-04-22:
How do I do code blocks here ? Seems not to work...

closed 2025-07-03 by github-actions[bot]

monosoul, 2025-08-20:
Hey @MarkusElsing and @mad-scientist , thanks for the converter you provided. I'm trying to use it rn, but get this error:
```
[2025-08-20 21:16:44] error: 	z2m: Publish 'get' 'running_state' to 'Living room floor thermostat' failed: 'Error: ZCL command 0xdc8e95fffeb6e7e5/1 hvacThermostat.read(["runningState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

Seems like `MEG5779` doesn't support getting `runningState`? Does it work for you?

MarkusElsing, 2025-08-20:
HI

I don’t see this, for me it works. But let me post in the thread the lastest version I have with a few fixes I did recently. It is still far from complete, no support yet for “occupancy” with setpoints, as well not for calibrations. I will add this later when I need it for my new installation. Currently I would note have a way to test this reasonably.

Best, Markus

Dr. Markus Elsing
CERN
1211 Genève 23
Suisse
EP-Department
Mobile: +41 75 411 4456

MarkusElsing, 2025-08-20:
Here is the latest version:

`
const { deviceAddCustomCluster, numeric } = require('zigbee-herdsman-converters/lib/modernExtend');
const Zcl = require('zigbee-herdsman').Zcl;
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const utils = require('zigbee-herdsman-converters/lib/utils');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const manufacturerOptions = { manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC };

const megExtend = {
    hvacUserInterfaceCfgCluster: () =>
        deviceAddCustomCluster('hvacUserInterfaceCfg', {
            ID: Zcl.Clusters.hvacUserInterfaceCfg.ID,
            attributes: {
                displayBrightnessActive: {
                    ID: 0xe000, 
                    type: Zcl.DataType.UINT8,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
                displayBrightnessInactive: {
                    ID: 0xe001, 
                    type: Zcl.DataType.UINT8,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
                displayActiveTimeout: {
                    ID: 0xe002, 
                    type: Zcl.DataType.UINT16,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
            },
            commands: {},
            commandsResponse: {},
        }),
    displayBrightnessActive: () =>
        numeric({
            name: 'display_brightness_active',
            cluster: 'hvacUserInterfaceCfg',
            attribute: 'displayBrightnessActive',
            description: 'Sets brightness of the temperature display during active state',
            entityCategory: 'config',
            unit: '%',
            valueMin: 0,
            valueMax: 100,
            valueStep: 1,
            zigbeeCommandOptions: manufacturerOptions,
        }),
    displayBrightnessInactive: () =>
        numeric({
            name: 'display_brightness_inactive',
            cluster: 'hvacUserInterfaceCfg',
            attribute: 'displayBrightnessInactive',
            description: 'Sets brightness of the temperature display during inactive state',
            entityCategory: 'config',
            unit: '%',
            valueMin: 0,
            valueMax: 100,
            valueStep: 1,
            zigbeeCommandOptions: manufacturerOptions,
        }),
    displayActiveTimeout: () =>
        numeric({
            name: 'display_active_timeout',
            cluster: 'hvacUserInterfaceCfg',
            attribute: 'displayActiveTimeout',
[... 84 lines cut ...]
        await endpoint1.configureReporting('hvacThermostat', [{
            attribute: 'pIHeatingDemand',
            minimumReportInterval: 10,
            maximumReportInterval: 3600,
            reportableChange: 10,
        }]);
        await endpoint1.configureReporting('hvacThermostat', [{
            attribute: 'pICoolingDemand',
            minimumReportInterval: 10,
            maximumReportInterval: 3600,
            reportableChange: 10,
        }]);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint1);
        await reporting.thermostatOccupiedCoolingSetpoint(endpoint1); // add cooling support
        // Read-only attributes — no need for binding
        // await endpoint1.read('hvacThermostat',['runningState']);
        await endpoint1.read('hvacUserInterfaceCfg',['keypadLockout','tempDisplayMode']);
        await endpoint1.read('hvacUserInterfaceCfg',['displayBrightnessActive']);
        await endpoint1.read('hvacUserInterfaceCfg',['displayBrightnessInactive']);
        await endpoint1.read('hvacUserInterfaceCfg',['displayActiveTimeout']);
    },
};

module.exports = definition;
`

monosoul, 2025-08-21:
That thermostat also supports setting min and max heat setpoint (limits the possible value range for temperature selector) - `tz.thermostat_min_heat_setpoint_limit` and `tz.thermostat_max_heat_setpoint_limit`.

In my case I don't care about cooling, so I went with this converter:
https://gist.github.com/monosoul/b2e07598d96eee3eaf9c07fb00576cbd

Thanks to everyone in this thread for making this thing work 🙂

Nailik, 2025-10-24:
Hi i will start working to convert this to modernExtend (as much as possible) if someone is already working on it please comment. 
Planning to get this merged and supported without externalConverter because the once mentioned here are already partly outdated and don't work with the latest z2mqtt.

mad-scientist, 2025-10-24:
Thanks for your effort!

Nailik, 2025-10-25:
Thanks a lot for providing this much information about the device. Especially the brightness endpoint.

I created a PR, for now there is no running_state, pi_cooling_demand and pilot_mode.

running_state and pilot_mode did not work for me and pi_cooling_demand is not yet supported by modernExtend.

Nailik, 2026-01-13:
I also noticed that with the latest z2m version the option to set the actual temperature on the thermostat seems to be missing

monosoul, 2026-01-13:
@MarkusElsing check the PR I linked, explicitly adding custom attributes with `write: true` should work

MarkusElsing, 2026-01-13:
With the default converter for the MEG5779 in the release I get:

Jan 14 00:29:03 openhab-at-home pnpm[189992]: [2026-01-14 00:29:03] error:         z2m: Publish 'set' 'display_brightness_inactive' to 'Heating FF LivingRoom' failed: 'Error: ZCL command 0x7cc6b6fffe0ada45/1 hvacUserInterfaceCfg.write({"57345":{"value":72,"type":32}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":4190,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'

Same if I add write:true to my version. I also have the problem that Zigbee2MQTT UI does not show in the hvacUserInterfaceCfg the display_brightness_inactive, neither with the default converter, nor with my version. Should I file a bug report ? I fear this is beyond my knowledge of this. I may revert to your classical converter meanwhile...

MarkusElsing, 2026-01-13:
It seems it has to do with the m.occupied() I've added naively. Putting this at the end seems to work, bizarre, without the write:true thing.  not that I understand why, maybe there is a problem with occupied() without attributes, not clear to me.

mad-scientist, 2026-01-17:
The thermostat now exposes four individual setpoints for heating and cooling and for occupied and unoccupied. As a result, on a Home Assistant dashboard, the card will show two settings. 

@MarkusElsing you added the cooling options - did you have any specific reason to do so? While technically the device does expose the endpoints, the physical MEG5779 device only has one temperature setting, hasn't it?

I suggest to remove the cooling setpoints, limits, and system mode

[image]

.

Nailik, 2026-01-17:
@mad-scientist i think it would automatically switch to cooling if supported.

I would like to be able to have a setting in z2mqtt to disable it since my floor heating doesn't support cooling at all.

mad-scientist, 2026-04-13:
@MarkusElsing was it you who submitted the PR? I would like to add one more feature, but I've never done this and no idea how to test before submitting.

In `schneiderElectricExtend` add to `customThermostatCluster`

```
                openWindowDetectionStatus: {
                    name: "openWindowDetectionStatus",
                    ID: 0xe012,
                    type: Zcl.DataType.ENUM8,
                    manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC,
                },
```
and add to the device definition
```
        m.binary({
            name: "open_window_detection_status",
            cluster: "hvacThermostat",
            attribute: "openWindowDetectionStatus",
            description: "This indicates whether an open window has been detected",
            valueOn: ["Open", 1],
            valueOff: ["Closed", 0],
            access: "STATE_GET",
            zigbeeCommandOptions: { manufacturerCode: Zcl.ManufacturerCode.SCHNEIDER_ELECTRIC },
        }),
```