Koenkk/zigbee2mqtt #22155: Ember: Ikea Tradfri switch not publishing actions

stefan-schweiger, 2024-04-09:
### What happened?

I've recently tried to use ember for my Sonoff stick and it seems most devices work fine, but the [Ikea Tradfri On/Off Swichtes](https://www.zigbee2mqtt.io/devices/E1743.html) don't seem to receive Zigbee / publish MQTT messages.

### What did you expect to happen?

Ember should support the same Zigbee devices that ezsp does.

### How to reproduce it (minimal and precise)

Set loglevel to debug, use ember as the adapter, add a Ikea Tradfri Switch, press it and see that nothing is received / published.

Switch adapter to ezsp, do the same and see everything working.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

7.4.1.0 build 0

### Adapter

Sonoff Dongle E

### Setup

Docker on Unraid

### Debug log

With ezsp:

```
Received Zigbee message from 'Schalter B', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 901
MQTT publish: topic 'zigbee2mqtt/Schalter B', payload '{"action":"on","battery":60,"linkquality":196,"update":{"installed_version":570492465,"latest_version":604241926,"state":"available"}}'
```

With ember:

<nothing>

Nerivec, 2024-04-11:
Can you check your groups and your scenes, if you have any that you created with ID zero, you will have to delete them and recreate them with another ID. Z2M used to allow this, but to prevent issues with an obscure Zigbee specification it was prevented recently. Ikea has a tendency to use that specification to send messages.
I dealt with someone with a rather large Ikea-based network recently, and this problem was not reported, so hopefully, it is just that.
If not, you will have to [enable herdsman debug](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee-herdsman-debug-logging) and capture the button being triggered so we can see what is going on.

stefan-schweiger, 2024-04-15:
@Nerivec I have only started using Zigbee2MQTT very recently and did not configure and group or scenes which weren't set up by default.

[image]
[image]

When I try to set a specific scene id I only get this error:

`Publish 'set' 'scene_store' to 'Schalter B' failed: 'Error: ZCL command 0x000d6ffffe7b86b6/1 genScenes.store({"groupid":0,"sceneid":12}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":26310,"apsFrame":{"profileId":260,"clusterId":5,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":14},"zclSequence":21,"commandIdentifier":4} timed out after 10000ms)'`

Here is what I have captures with the DEBUG env var set and trying the different combinations (UP, DOWN, HOLD UP, HOLD DOWN -> not necessarily in that order)

```
2024-04-15T15:03:38.400Z zigbee-herdsman:adapter:ember:uart:ash:parser <<<< [FRAME raw=0607b1a96b2a164fa6904b2daa546d499dcb245b12ce671e4f7e]
2024-04-15T15:03:38.400Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA]
2024-04-15T15:03:38.401Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=6]
2024-04-15T15:03:38.401Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=6 frmNum=0] Added to rxQueue
2024-04-15T15:03:38.404Z zigbee-herdsman:adapter:ember:uart:ash ---> [FRAME type=ACK frmRx=1]
2024-04-15T15:03:38.405Z zigbee-herdsman:adapter:ember:uart:ash:writer >>>> [FRAME raw=8160597e]
2024-04-15T15:03:38.405Z zigbee-herdsman:adapter:ember:ezsp <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=69 Len=22]
2024-04-15T15:03:38.405Z zigbee-herdsman:adapter:ember:ezsp ezspMessageSentHandler(): callback called with: [type=MULTICAST], [indexOrDestination=65533], [apsFrame={"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":255,"options":256,"groupId":901,"sequence":240}], [messageTag=255], [status=SUCCESS], [messageContents=]
2024-04-15T15:03:39.594Z zigbee-herdsman:adapter:ember:uart:ash:parser <<<< [FRAME raw=1607b1a96b2a164fa6904b2daa546d499dcb245a12ce67588d7e]
2024-04-15T15:03:39.594Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA]
2024-04-15T15:03:39.594Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=6]
2024-04-15T15:03:39.594Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=6 frmNum=1] Added to rxQueue
2024-04-15T15:03:39.597Z zigbee-herdsman:adapter:ember:uart:ash ---> [FRAME type=ACK frmRx=2]
2024-04-15T15:03:39.597Z zigbee-herdsman:adapter:ember:uart:ash:writer >>>> [FRAME raw=82503a7e]
2024-04-15T15:03:39.598Z zigbee-herdsman:adapter:ember:ezsp <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=69 Len=22]
2024-04-15T15:03:39.598Z zigbee-herdsman:adapter:ember:ezsp ezspMessageSentHandler(): callback called with: [type=MULTICAST], [indexOrDestination=65533], [apsFrame={"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":255,"options":256,"groupId":901,"sequence":241}], [messageTag=255], [status=SUCCESS], [messageContents=]
2024-04-15T15:03:43.521Z zigbee-herdsman:adapter:ember:uart:ash:parser <<<< [FRAME raw=2607b1a96b2a164fa6904b2daa546d499dcb245912ce6793cb7e]
2024-04-15T15:03:43.521Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA]
2024-04-15T15:03:43.522Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=6]
2024-04-15T15:03:43.522Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=6 frmNum=2] Added to rxQueue
2024-04-15T15:03:43.526Z zigbee-herdsman:adapter:ember:uart:ash ---> [FRAME type=ACK frmRx=3]
2024-04-15T15:03:43.527Z zigbee-herdsman:adapter:ember:uart:ash:writer >>>> [FRAME raw=83401b7e]
2024-04-15T15:03:43.527Z zigbee-herdsman:adapter:ember:ezsp <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=69 Len=22]
2024-04-15T15:03:43.527Z zigbee-herdsman:adapter:ember:ezsp ezspMessageSentHandler(): callback called with: [type=MULTICAST], [indexOrDestination=65533], [apsFrame={"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":255,"options":256,"groupId":901,"sequence":242}], [messageTag=255], [status=SUCCESS], [messageContents=]
2024-04-15T15:03:44.482Z zigbee-herdsman:adapter:ember:uart:ash:parser <<<< [FRAME raw=3607b1a96b2a164fa6904b2daa546d499dcb245812ce67d5097e]
2024-04-15T15:03:44.482Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA]
2024-04-15T15:03:44.482Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=6]
2024-04-15T15:03:44.482Z zigbee-herdsman:adapter:ember:uart:ash <--- [FRAME type=DATA ackNum=6 frmNum=3] Added to rxQueue
2024-04-15T15:03:44.486Z zigbee-herdsman:adapter:ember:uart:ash ---> [FRAME type=ACK frmRx=4]
2024-04-15T15:03:44.486Z zigbee-herdsman:adapter:ember:uart:ash:writer >>>> [FRAME raw=8430fc7e]
2024-04-15T15:03:44.486Z zigbee-herdsman:adapter:ember:ezsp <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=69 Len=22]
2024-04-15T15:03:44.486Z zigbee-herdsman:adapter:ember:ezsp ezspMessageSentHandler(): callback called with: [type=MULTICAST], [indexOrDestination=65533], [apsFrame={"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":255,"options":256,"groupId":901,"sequence":243}], [messageTag=255], [status=SUCCESS], [messageContents=]
2024-04-15T15:03:46.546Z zigbee-herdsman:adapter:ember:uart:ash:parser <<<< [FRAME raw=4607b1a96b2a164fa6904b23aa546d499dcb245f12ce6705f07e]
[... cut ...]
2024-04-15T15:03:47.856Z zigbee-herdsman:adapter:ember:uart:ash:writer >>>> [FRAME raw=8610be7e]
2024-04-15T15:03:47.857Z zigbee-herdsman:adapter:ember:ezsp <=== [FRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=69 Len=22]
2024-04-15T15:03:47.857Z zigbee-herdsman:adapter:ember:ezsp ezspMessageSentHandler(): callback called with: [type=MULTICAST], [indexOrDestination=65533], [apsFrame={"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":255,"options":256,"groupId":901,"sequence":245}], [messageTag=255], [status=SUCCESS], [messageContents=]
```

stefan-schweiger, 2024-04-21:
@Nerivec do those logs help you in anyway or is there any further information that would help to track this down?

closed 2024-04-21 by Koenkk

stefan-schweiger, 2024-04-22:
Just tested with the latest dev docker image and can confirm that it now works, thanks!