Koenkk/zigbee2mqtt #22157: Ember crashing

Dinth, 2024-04-10:
### What happened?

Migrated to Ember today, after a few hours of working fine, Z2M has crashed

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20240408

### Adapter

SM06-M

### Setup

Addon to HASS working on a x86-64 mac mini

### Debug log

Zigbee2MQTT:info  2024-04-10 00:05:39: MQTT publish: topic 'zigbee2mqtt/Loft gas sensor', payload '{"carbon_monoxide":false,"co":0,"gas":false,"gas_value":0.001,"last_seen":"2024-04-09T23:05:39.266Z","linkquality":244}'
Zigbee2MQTT:debug 2024-04-10 00:05:39: Received Zigbee message from 'Master bedroom presence sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1}],"seq":33280}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-04-10 00:05:39: MQTT publish: topic 'zigbee2mqtt/Master bedroom presence sensor', payload '{"fading_time":60,"illuminance":0,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":6,"last_seen":"2024-04-09T23:05:39.271Z","linkquality":196,"medium_motion_detection_distance":4,"medium_motion_detection_sensitivity":8,"motion_state":"large","presence":true,"small_detection_distance":4,"small_detection_sensitivity":8}'
Zigbee2MQTT:debug 2024-04-10 00:05:39: Received Zigbee message from 'Loft gas sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":18}],"seq":8704}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-04-10 00:05:39: MQTT publish: topic 'zigbee2mqtt/Loft gas sensor', payload '{"carbon_monoxide":false,"co":0,"gas":false,"gas_value":0.001,"last_seen":"2024-04-09T23:05:39.514Z","linkquality":244}'
Zigbee2MQTT:debug 2024-04-10 00:05:39: Received Zigbee message from 'Loft gas sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":18}],"seq":8704}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-04-10 00:05:39: MQTT publish: topic 'zigbee2mqtt/Loft gas sensor', payload '{"carbon_monoxide":false,"co":0,"gas":false,"gas_value":0.001,"last_seen":"2024-04-09T23:05:39.530Z","linkquality":244}'
Zigbee2MQTT:debug 2024-04-10 00:05:39: Received Zigbee message from 'Loft gas sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,2],"type":"Buffer"},"datatype":2,"dp":19}],"seq":8960}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-04-10 00:05:39: MQTT publish: topic 'zigbee2mqtt/Loft gas sensor', payload '{"carbon_monoxide":false,"co":0,"gas":false,"gas_value":0.001,"last_seen":"2024-04-09T23:05:39.577Z","linkquality":244}'
Zigbee2MQTT:debug 2024-04-10 00:05:39: Received Zigbee message from 'Master bedroom presence sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":101}],"seq":33536}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-04-10 00:05:39: MQTT publish: topic 'zigbee2mqtt/Master bedroom presence sensor', payload '{"fading_time":60,"illuminance":0,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":6,"last_seen":"2024-04-09T23:05:39.593Z","linkquality":200,"medium_motion_detection_distance":4,"medium_motion_detection_sensitivity":8,"motion_state":"medium","presence":true,"small_detection_distance":4,"small_detection_sensitivity":8}'
Zigbee2MQTT:debug 2024-04-10 00:05:40: Received Zigbee message from 'Master bedroom presence sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-04-10 00:05:40: No converter available for 'ZG-205Z/A' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'
Zigbee2MQTT:info  2024-04-10 00:05:40: MQTT publish: topic 'zigbee2mqtt/Master bedroom presence sensor', payload '{"fading_time":60,"illuminance":0,"indicator":"ON","large_motion_detection_distance":6,"large_motion_detection_sensitivity":6,"last_seen":"2024-04-09T23:05:40.814Z","linkquality":200,"medium_motion_detection_distance":4,"medium_motion_detection_sensitivity":8,"motion_state":"medium","presence":true,"small_detection_distance":4,"small_detection_sensitivity":8}'
ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=107 Len=26].
!!! NCP FATAL ERROR reason=ERROR_WRONG_DIRECTION. ATTEMPTING RESET... !!!
┌──────────────┬──────────┬─────────────┐
│   (index)    │ Received │ Transmitted │
├──────────────┼──────────┼─────────────┤
│ Total frames │   352    │     350     │
│  Cancelled   │    0     │      0      │
│ DATA frames  │   291    │     108     │
│  DATA bytes  │   6596   │    2047     │
│ Retry frames │    56    │      2      │
│  ACK frames  │    3     │     239     │
│  NAK frames  │    1     │      0      │
│ nRdy frames  │    0     │      0      │
└──────────────┴──────────┴─────────────┘
┌──────────────────┬──────────┐
│     (index)      │ Received │
├──────────────────┼──────────┤
│    CRC errors    │    0     │
│   Comm errors    │    0     │
│ Length < minimum │    0     │
│ Length > maximum │    0     │
│   Bad controls   │    0     │
│   Bad lengths    │    1     │
│ Bad ACK numbers  │    0     │
│  Out of buffers  │    0     │
│   Retry dupes    │    53    │
│ Out of sequence  │    0     │
│   ACK timeouts   │    1     │
└──────────────────┴──────────┘
======== ASH stopped ========
ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last: [FRAME: ID=52:"SEND_UNICAST" Seq=107 Len=26].
======== EZSP stopped ========
======== Ember Adapter Stopped ========
======== Ember Adapter Starting ========
[... 27 lines cut ...]
│ nRdy frames  │    0     │      0      │
└──────────────┴──────────┴─────────────┘
┌──────────────────┬──────────┐
│     (index)      │ Received │
├──────────────────┼──────────┤
│    CRC errors    │    0     │
│   Comm errors    │    0     │
│ Length < minimum │    0     │
│ Length > maximum │    0     │
│   Bad controls   │    0     │
│   Bad lengths    │    0     │
│ Bad ACK numbers  │    0     │
│  Out of buffers  │    0     │
│   Retry dupes    │    0     │
│ Out of sequence  │    0     │
│   ACK timeouts   │    0     │
└──────────────────┴──────────┘
======== ASH stopped ========
======== EZSP stopped ========
======== Ember Adapter Stopped ========
Zigbee2MQTT:error 2024-04-10 00:05:46: Adapter disconnected, stopping
Zigbee2MQTT:debug 2024-04-10 00:05:46: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2024-04-10 00:05:46: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
Zigbee2MQTT:info  2024-04-10 00:05:46: Disconnecting from MQTT server
Zigbee2MQTT:info  2024-04-10 00:05:46: Stopping zigbee-herdsman...

Nerivec, 2024-04-11:
`ERROR_WRONG_DIRECTION` means Z2M received a _command_ from the adapter, instead of a _response_. Definitely not right. Can you exclude the usual suspects around USB and 2.4GHz interferences?

_The inability to properly restart after a fatal error should have been dealt with in `dev` already._

---

@darkxst

Any idea on why we're seeing this here? The SM06-M usually fails to set the message count, but others usually work fine.

```
[EzspConfigId] Failed to SET "ADDRESS_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY.
[EzspConfigId] Failed to SET "APS_UNICAST_MESSAGE_COUNT" TO "32" with status=ERROR_OUT_OF_MEMORY.
[EzspConfigId] Failed to SET "NEIGHBOR_TABLE_SIZE" TO "26" with status=ERROR_OUT_OF_MEMORY.
[EzspConfigId] Failed to SET "SOURCE_ROUTE_TABLE_SIZE" TO "200" with status=ERROR_INVALID_VALUE.
[EzspConfigId] Failed to SET "MULTICAST_TABLE_SIZE" TO "16" with status=ERROR_OUT_OF_MEMORY.
```

Nerivec, 2024-05-15:
Restart problem should be fixed in 1.37.1.
Config issues should be fixed with recent firmware release from SMLight (2.2.x core with latest emberznet available).
Let me know if you still experience issues.

closed 2024-05-15 by Nerivec

Dinth, 2024-05-16:
Hi. Just to wrap up, after upgrading Z2M and the firmware, i have not seen **this** particular crash anymore but the adapter was still crashing regularly in variety of different ways. Yesterday i have migrated to ZigStar poe adapter so i am back on zstack right now.

Nerivec, 2024-05-16:
@Dinth Thanks for the feedback. Any chance you have the logs associated with those crashes?

Dinth, 2024-05-17:
Hey. I dont have any debug logs but a few facts:
1. SLZB-06M was crashing both with ember and with the previous driver before, on average every 12-48h.
2. I have put a notification in place using the Z2M status in MQTT to know when it crashes and a few times i have managed to get to the logs immediately after the crash - with info log level i havent seen anything extraordinary and concerning there. 
3. At least with Ember (will get to that later on) Z2M watchdog was not kicking in even though MQTT status was changing. I have needed to manually restart Z2M every time the crash happened.
4. This must be something SMLight specific - over the last 6 months i have tested various PoE coordinators from them - two different 06, one 06M with both ezsp and ember, different firmware versions and connection methods (PoE, USB, Wifi) on each of them - and every single one of them was randomly crashing every 12-72h, although with different adapter models and different adapter drivers (ezsp/ember) the way they've been crashing was different. 
5. 2 days ago i have migrated to ZigStar PoE adapter and so far it has not crashed - fingers crossed that this will end my 5 month long Zigbee nightmare :)

darkxst, 2024-05-18:
Hi @Dinth,
  Do you happen to know if it was the ESP32 core firmware crashing (probably web ui and network interface stop working) or if it was the Zigbee chip crashing?
  
  Do you happen to know what core firmware version you were running, I believe there were a few stability fixes in the more recent 2.x.dev core firmware builds.
  
  In general to anyone seeing possible crashes or other issues on SLZB-06M please try the lastest core firmware, `2.2.x.dev`, and Zigbee OTA firmware `20240510`. Between them they do include a few fixes compared to the stable `2.0.18` firmware, that should improve stability.