Koenkk/zigbee2mqtt #22163: Device clash - Ikea Light Switch / Lumi Magnet Sensor

S33G, 2024-04-10:
### What happened?

I had a Ikea Light Switch which now has the image, model and appears to expose the same parameters as some newly added magnet sensors.

[image]

Devices in question:
- https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html#aqara-mccgq11lm
- https://www.zigbee2mqtt.io/devices/E2001_E2002.html#ikea-e2001%252Fe2002

Upon checking the `database.db` I see the following JSON:
`{
   "id":31,
   "type":"EndDevice",
   "ieeeAddr":"0x30fb10fffeeddc8d",
   "nwkAddr":61999,
   "manufId":4476,
   "manufName":"IKEA of Sweden",
   "powerSource":"Battery",
   "modelId":"lumi.sensor_magnet.aq2",
   "epList":[
      1
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":2080,
         "inClusterList":[
            0,
            1,
            3,
            32,
            4096,
            64599
         ],
         "outClusterList":[
            3,
            6,
            8,
            25,
            4096
         ],
         "clusters":{
            "genPollCtrl":{
               "attributes":{
                  "checkinInterval":172800
               }
            }
         },
         "binds":[
            {
               "cluster":32,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b00258d4379",
               "endpointID":1
            }
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":3,
   "stackVersion":103,
   "hwVersion":1,
   "dateCode":"20200708",
   "swBuildId":"1.0.024",
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      "configured":1324213189
   },
   "lastSeen":1712754702838,
   "checkinInterval":3600
}`

I've tried to "reconfigure" the device, with the yellow button in the GUI.
I've tried force removing device, pairing it again and it still appears in the same state.

The switch is completely unusable currently due to the exposed params being incorrect.

### What did you expect to happen?

My light switch to function normally and not to be replaced by the magnet sensor

### How to reproduce it (minimal and precise)

1. Add Ikea E2001/E2002
2. Add Aqara MCCGQ11LM
3. See existing Ikea E2001/E2002 become a "Aqara MCCGQ11LM"

### Zigbee2MQTT version

1.36.0

### Adapter firmware version

20240315

### Adapter

SMLIGHT SLZB-06

### Setup

Docker

mrskycriper, 2024-04-10:
Have you tried removing the `database.db` entry for this device manually? Looks like a weird device address clash

S33G, 2024-04-11:
> Have you tried removing the `database.db` entry for this device manually? Looks like a weird device address clash

Yep. Still comes through as a window sensor

closed 2024-11-08 by github-actions[bot]