Koenkk/zigbee2mqtt #22164: Zigbee2MQTT starts but service thinks it's timed out.

GForce2010, 2024-04-10:
### What happened?

I have just installed Zigbee2MQTT onto a virtual machine running Ubuntu 22.04.4 LTS.

I followed the instructions in the documentation, including the Ubuntu tip about installing Node.js through snap.

Zigbee2MQTT runs fine when run with "npm start".

I created the service file and changed the user listed but that kept failing to run, I had to change the line "ExecStart=/usr/bin/node index.js" to "ExecStart=/snap/bin/node index.js"

Now when I start the service Zigbee2MQTT starts up and works as expected, the log shows it's started and my devices become responsive, but the terminal comes back with the error "Job for zigbee2,qtt.service failed because a timeout was exceeded".

Getting the status of the service shows:
`zig@zigbee2mqtt:~$ sudo systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
     Active: activating (start) since Wed 2024-04-10 14:44:09 UTC; 3s ago
   Main PID: 3074 (node)
      Tasks: 0 (limit: 2220)
     Memory: 12.0K
        CPU: 87ms
     CGroup: /system.slice/zigbee2mqtt.service
             ‣ 3074 /snap/node/8481/bin/node index.js

Apr 10 14:44:09 zigbee2mqtt systemd[1]: Starting zigbee2mqtt...
`

Connecting to the frontend works but I can see the message "Zigbee2MQTT has started" pop up regularly in the corner, I assume because the service is being restarted because the OS doesn't know it's already running.

Contents of my service file

`
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment=NODE_ENV=production
Type=notify
ExecStart=/snap/bin/node index.js
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
# Or use StandardOutput=null if you don't want Zigbee2MQTT messages filling syslog, for more options see systemd.exec(5)
StandardError=inherit
WatchdogSec=10s
Restart=always
RestartSec=10s
User=zig

[Install]
WantedBy=multi-user.target
`

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

Texas Instruments LAUNCHXL-CC26X2R1

### Setup

Virtual machine running on Proxmox.

butlerfr, 2024-04-17:
same issue as : https://github.com/Koenkk/zigbee2mqtt/issues/21463
I think and I'm in the case too

closed 2024-11-19 by github-actions[bot]

Gitdyr, 2025-09-27:
My system is quite slow. I solved the problem with:

`WatchdogSec=10s`

and

`RestartSec=30s`

Gitdyr, 2025-09-28:
It works for me when I use:

`Type=exec`

and

`WatchdogSec=0`