Koenkk/zigbee2mqtt #22166: Sonoff Dongle-P with router firmware goes offline

szaman-89, 2024-04-10:
### What happened?

Just bought two Sonoff Dongle-P adapters and flashed both with router firmware but they can't stay online in Z2M even for 24 hours. My coordinator is also Sonoff Dongle-P with latest firmware. I've read that in some cases shortening reporting interval can help but it doesn't work for me. I did 4 days test for both adapters and last two days on shortened reporting interval and it looks like both of them goes offline in the same time. Reflashed sticks after each repairing need. I can't find any additional information about such problem with this adapter.

[image]

### What did you expect to happen?

Just to stay online and keep connection with coordinator.

### How to reproduce it (minimal and precise)

Just pair a Sonoff Dongle-P with router firmware and wait less than 24 hours.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20221226

### Adapter

Sonoff Dongle-P

### Setup

Add-on on Home Assistant on Docker container

harvindhillon, 2024-04-11:
I can confirm same issue

majkers, 2024-04-13:
@szaman-89 OFF_TOPIC to test my dongle, can You tell me how to display "W sieci/Poza siecią" column?

szaman-89, 2024-04-13:
You can find it here: https://www.zigbee2mqtt.io/guide/configuration/device-availability.html

supaeasy, 2024-04-18:
I have a Sonoff Dongle-E with router firmware that went offline and crashed half of my network last weekend. 
It is in use since about two years and never had any problems. I re-paired it on tuesday and it is working fine ever since.

I had recently reduced the reporting interval for all my lamps to 1-3 seconds and thought maybe that was too much for it. I have now changed that to 1-30s and so far nothing bad happened. But if other people have similar problems there might be more to that.

@szaman-89 "Noticed that routers can't be paired again without power replug or reset dongle." That is wrong. You can just press the boot button for >5s and it will re-pair.

harvindhillon, 2024-04-18:
Never had issues with Sonoff Dongle-E router

It's always the Sonoff Dongle-P that go down then refuse to re-add for some reason

Pferdebockwurst, 2024-04-27:
I have the same issue, but it never occured before the last update.

harvindhillon, 2024-04-30:
Very frustrating, two of the routers again went offline and now refusing to repair as always failing interview. Its a 20mins to 1h process of restarts, reflash, repairing to get them to work. Wondering if it is a bad router firmware or bad stick?

Pferdebockwurst, 2024-04-30:
> Very frustrating, two of the routers again went offline and now refusing to repair as always failing interview. Its a 20mins to 1h process of restarts, reflash, repairing to get them to work. Wondering if it is a bad router firmware or bad stick?

I can confirm. Very annoying.

harvindhillon, 2024-05-01:
Yea I have 4 of these in my network, and 3 went down one after another yesterday night and I spent all morning trying to recover my network as you can imagine most devices were routed through these dedicated routers.

At one point I thought I'd have to repair everything but alas 7hours later it is looking like I have managed to bring everything back. Time will tell how long this lasts.

I cant find any errors in logs, just starts timing out. I have my availability set at 20mins, I dont really want to bring it any lower than this as I have around 145 devices in my network,

Otherwise network is rock solid

szaman-89, 2024-05-06:
> I have a Sonoff Dongle-E with router firmware that went offline and crashed half of my network last weekend. It is in use since about two years and never had any problems. I re-paired it on tuesday [...]

Issue is about P version, not E and can't be paired again without repluging to power source.

After few weeks of observation I had a time when routers stays online for about a couple days and after server reboot situation came back to it's normal - not working longer than 24 hours. Yesterday I repluged one dongle and now it's 14 hours offline... Very frustrating...

harvindhillon, 2024-05-30:
I have a couple of SMLight 06 dongles that do not work via LAN but work fine with USB so I decided to switch them into routers.

Same exact problem, they drop offline after a couple of hours, so it seems to be linked to the chip itself rather than Sonoffs as these are using the same chipset.

Pferdebockwurst, 2024-08-09:
I'm still having this problem. Did someone find a solution?

supaeasy, 2024-08-09:
Maybe check if WiFi channels are interferring:
https://www.metageek.com/training/resources/zigbee-wifi-coexistence/

harvindhillon, 2024-08-09:
I havent had any issues for 2 months now. I have OTA turned off and availability at 35mins for routers

szaman-89, 2024-08-09:
In my case I put coordinator at extension cable and changed wifi channels on my routers and for couple months problem goes away.

harvindhillon, 2024-08-14:
Weirdly the problem is back for me :( Not sure what happened. 

I was away for 7 weeks and the routers stayed online but now I've been back for 3 weeks, 2 routers have gone offline and back to the pain of repairing. Which also takes forever cause interview always fails, it never pairs the first time. I wonder if it is just cause of the traffic in the system? I have a fairly large network

marcoraddatz, 2025-01-05:
> This issue is stale because it has been open 180 days with no activity. Remove stale label or comment or this will be closed in 30 days

Not stale. Having the same issue with v2.

closed 2025-03-15 by github-actions[bot]

SuperSandro2000, 2026-03-15:
This is still an issue