Koenkk/zigbee2mqtt #22174: Click Smart Plus Dimmers (TUYA) TS110F (_TZ3000_ha4ejozn (2Gang) & _TZ3000_estfrmup (1Gang)

gpshml, 2024-04-10:
### Link

https://click-smart.com/products/1-gang-smart-dimming-receiver

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0x60a423fffe801204","nwkAddr":34255,"manufId":4098,"manufName":"_TZ3000_ha4ejozn","powerSource":"Mains (single phase)","modelId":"TS110F","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,4,5,6,8],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS110F","manufacturerName":"_TZ3000_ha4ejozn","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"64512":255,"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8db9d","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8db9d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":257,"inClusterList":[4,5,6,8],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"64512":255,"currentLevel":127}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8db9d","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8db9d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1712783767932}

### Comments

I had this working by using a similar config and with a recent update, it stopped recognising the device for and associating with the custom config.  The file that was working was this one (see attached), is there a simple thing that is wrong or is it fundamentally broken.  I am happy to try and understand how to configure this as I can code but it's the context of Zigbee protocol that I don't have.
[TS110F_1Gang.txt](https://github.com/Koenkk/zigbee2mqtt/files/14938114/TS110F_1Gang.txt)
[TS110F_2Gang.txt](https://github.com/Koenkk/zigbee2mqtt/files/14938115/TS110F_2Gang.txt)

### External definition

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS110F'],
    model: 'TS110F',
    vendor: '_TZ3000_estfrmup',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;

const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS110F'],
    model: 'TS110F',
    vendor: '_TZ3000_ha4ejozn',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), onOff({"powerOnBehavior":false,"endpointNames":["1","2"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

gpshml, 2024-04-10:
{"id":9,"type":"Router","ieeeAddr":"0x842e14fffe19e467","nwkAddr":43933,"manufId":4098,"manufName":"_TZ3000_estfrmup","powerSource":"Mains (single phase)","modelId":"TS110F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,4,5,6,8],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS110F","manufacturerName":"_TZ3000_estfrmup","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"64512":255,"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8db9d","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c8db9d","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{"onLevelSupported":false}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1712783379786}

mrskycriper, 2024-04-11:
Are there any branding and/or model names on the boxes or devices themselves? Were the external definitions fully working before the update?

mrskycriper, 2024-04-11:
Nevermind, I found their names and models. Can you please test them with this external converter?

```js
const {deviceEndpoints} = require('zigbee-herdsman-converters/lib/modernExtend');
const {fingerprint, modernExtend as tuyaModernExtend} = require('zigbee-herdsman-converters/lib/tuya');
const {tuyaLight} = tuyaModernExtend;

const definitions = [
    {
        fingerprint: fingerprint('TS110F', ['_TZ3000_estfrmup']),
        model: 'CSP051',
        vendor: 'Click Smart+',
        description: '1 gang dimmer module',
        extend: [tuyaLight({minBrightness: true})],
    },
    {
        fingerprint: fingerprint('TS110F', ['_TZ3000_ha4ejozn']),
        model: 'CSP052',
        vendor: 'Click Smart+',
        description: '2 gang dimmer module',
        extend: [
            deviceEndpoints({endpoints: {'L1': 1, 'L2': 2}}),
            tuyaLight({minBrightness: true, endpointNames: ['L1', 'L2']}),
        ],
    },
];

module.exports = definitions;
```

gpshml, 2024-04-13:
Thanks, I will give it a try, they were working pretty well before, nothing
that I am aware of which didn’t work before the update and not sure if it
was MQTT or Home Assistant.

gpshml, 2024-04-13:
I have added the extension and reloaded HA and also re-started Zigbee2MQTT via the UI.  This is now showing the single converter (as added), the individual devices seem to still be referencing the descriptions.  What's the easiest way to get it to move to the new converter?

mrskycriper, 2024-04-13:
> I have added the extension and reloaded HA and also re-started Zigbee2MQTT via the UI. This is now showing the single converter (as added), the individual devices seem to still be referencing the descriptions. What's the easiest way to get it to move to the new converter?

Extensions are not the same thing as external converters. Are you sure that you didn't mix them up?

External converters are added via file system as `.js` files and then included via GUI into the configuration.

gpshml, 2024-04-13:
[image]
I meant External Converters, sorry.  I can see this is how showing but the revised descriptions don't show in the individual zigbee items.

How do I ensure the new converter is being used?

gpshml, 2024-04-13:
After re-upgrading to the latest version of Zigbee2MQTT and HA, I am getting errors which say it's 'unsupported'.

[image]

gpshml, 2024-04-13:
This is how it looks in the device:
[image]

[image]

External Definition Generated:
const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS110F'],
    model: 'TS110F',
    vendor: '_TZ3000_ha4ejozn',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), onOff({"powerOnBehavior":false,"endpointNames":["1","2"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;

mrskycriper, 2024-04-14:
Sorry, bug in a converter on my part, how about this one?

```js
const {deviceEndpoints} = require('zigbee-herdsman-converters/lib/modernExtend');
const {fingerprint, modernExtend: tuyaModernExtend} = require('zigbee-herdsman-converters/lib/tuya');
const {tuyaLight} = tuyaModernExtend;

const definitions = [
    {
        fingerprint: fingerprint('TS110F', ['_TZ3000_estfrmup']),
        model: 'CSP051',
        vendor: 'Click Smart+',
        description: '1 gang dimmer module',
        extend: [tuyaLight({minBrightness: true})],
    },
    {
        fingerprint: fingerprint('TS110F', ['_TZ3000_ha4ejozn']),
        model: 'CSP052',
        vendor: 'Click Smart+',
        description: '2 gang dimmer module',
        extend: [
            deviceEndpoints({endpoints: {'L1': 1, 'L2': 2}}),
            tuyaLight({minBrightness: true, endpointNames: ['L1', 'L2']}),
        ],
    },
];

module.exports = definitions;
```

mrskycriper, 2024-04-21:
Any luck tesing with the fixed external converter?

gpshml, 2024-04-21:
Thanks, apologies, I hadn’t got around to it.

It’s recognising them and showing the exposes but has the following error:

Error 2024-04-21 22:36:42Publish 'set' 'min_brightness' to 'Master Bedroom
- Sidelights' failed: 'Error: 'min_brightness' is not a number, got string
()'

Also the 52 is not showing up the 2nd endpoint in the ‘exposes’, weirdly in
‘bind’ it shows them both as destination 1

mrskycriper, 2024-04-22:
> Error 2024-04-21 22:36:42Publish 'set' 'min_brightness' to 'Master Bedroom - Sidelights' failed: 'Error: 'min_brightness' is not a number, got string
()'

Is the field for min_brightness empty by default? What happens if you set the value manually?

Does on/off and dimming work besides min_brightness?

> Also the 52 is not showing up the 2nd endpoint in the ‘exposes’, weirdly in
‘bind’ it shows them both as destination 1

About endpoints, can you test is on a dev branch? This was probably fixed already

gpshml, 2024-04-22:
Thanks, this seems to have now been resolved in 1.36.1.  I am seeing both end points.  Checking for logging again to see if that has been resolved.

mrskycriper, 2024-04-23:
> It seems to stick if you manually set the Min_Brightness, on/off and dimming seems to work (just L2 inaccessible).  I will see if I can update to a dev version, not done that before.
> 
> 

Can you test if it respects the minimum brightness setting? If it is, then it was just a problem of not reading the value on configuration.

closed 2024-11-20 by github-actions[bot]