Koenkk/zigbee2mqtt #22175: [New device support]: Avatto ZDMS16-2 2 gang dimmer

ddpsft, 2024-04-11:
### Link

https://aliexpress.com/item/1005006523496245.html

### Database entry

No idea how to get it

### Comments

I'm following the [page to add new Tuya based devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html), but after following it I'm no closer than before (using the regular method). Even though this isn't the regular method I'm posting it here wondering if someone that knows how to to it can make it run properly with given information.

After going to Tuya IoT platform I've got these datapoints:
`{"1":"Switch1","2":"Brightness1","3":"Minimum brightness1","4":"Type of light source1","5":"Maximum brightness1", "6":"Countdown1","7":"Switch2","8":"Brightness2","9":"Minimum brightness2","10":"Type of light source2","11":"Maximum brightness2","12":"Countdown2","13":"Mode","14":"Power-on Status Setting","47":"Switch Type"}`

But I still unable to make it run. According to log (after pressing its button):
```
Debug 2024-04-10 22:19:22Received Zigbee message from '0xa4c138a8a2fde3e7', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":12288}' from endpoint 1 with groupID 0
Debug 2024-04-10 22:19:22Skipping message, still interviewing
Debug 2024-04-10 22:19:22Received Zigbee message from '0xa4c138a8a2fde3e7', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":7}],"seq":12544}' from endpoint 1 with groupID 0
Debug 2024-04-10 22:19:22Skipping message, still interviewing
```
By pressing each gang switch individually it confirms that 1 refers to the first gang switch and 7 to the second gang switch. It's manual states that holding it would dim the light, but the reports from debug only differs in "seq" and do not appers to refer to % brightness

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_o9gyszw2',
        },
    ],
    
    model: 'ZDMS16',
    vendor: 'Avatto',
    description: 'Zigbee 2 channel Dimmer',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.Switch1(),
        /*e.Brightness1(),
        /*e.Minimum_brightness1(),
        e.Type_of_light_source1(),
        e.Maximum_brightness1(),
        e.Countdown1(),
        e.Switch2(),
        e.Brightness2(),
        e.Minimum_brightness2(),
        e.Type_of_light_source2(),
        e.Maximum_brightness2(),
        e.Countdown2(),
        e.Power-on_Status_Setting(),
        e.Switch_Type(),*/
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1,'Switch1',tuya.valueConverter.raw],
            /*[2,'Brightness 1',tuya.valueConverter.raw],
            /*[3,'Minimum brightness1',tuya.valueConverter.raw],
            [4,'Type of light source1',tuya.valueConverter.raw],
            [5,'Maximum brightness1',tuya.valueConverter.raw],
            [6,'Countdown1',tuya.valueConverter.raw],
            [7,'Switch 1',tuya.valueConverter.raw],
            [8,'Brightness 1',tuya.valueConverter.raw],
            [9,'Minimum brightness1',tuya.valueConverter.raw],
            [10,'Type of light source1',tuya.valueConverter.raw],
            [11,'Maximum brightness1',tuya.valueConverter.raw],
            [12,'Countdown1',tuya.valueConverter.raw],
            [14,'Power-on Status Setting',tuya.valueConverter.raw],
            [17,'Switch Type',tuya.valueConverter.raw],*/
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],

};

module.exports = definition;
```

mrmaximas, 2024-04-11:
database is in the z2m folder or subfolders

ddpsft, 2024-04-11:
> database is in the z2m folder or subfolders

Thank you!

{"id":23,"type":"Router","ieeeAddr":"0xa4c138a8a2fde3e7","nwkAddr":61105,"manufId":4417,"manufName":"_TZE204_o9gyszw2","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_o9gyszw2","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1712855728234}

Suxsem, 2024-04-13:
I have the same dimmer (only the 1gang version), I'm going to implement the custom converter. I will update you soon, hopefully today

Suxsem, 2024-04-13:
This is where I am so far.

Working:
- ON/OFF
- BRIGHTNESS CONTROL
- COUNTDOWN
- SWITCH TYPE

Not working:
- MIN BRIGHTNESS SETTING
- MAX BRIGHTNESS SETTING
- LIGHT TYPE SETTING

I'm not sure of these settings datapoints. Strangely enough switch type is on datapoint 4 (where usually there is light type setting) so I don't really known which datapoint corresponds to light type and I don't have a tuya gateway to test. Min and max brightness should be on datapoints 3 and 5 but I really don't see any difference in brightness when I change these values.
I hope that someone with tuya gateway can tell us the right datapoints for the missing settings.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_5cuocqty',
        },
    ],

    model: 'ZDMS16-1',
    vendor: 'Avatto',
    description: 'Zigbee 1 channel Dimmer',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
            tuya.exposes.lightBrightnessWithMinMax(),
            tuya.exposes.countdown(),
            tuya.exposes.lightType(),
            e.power_on_behavior().withAccess(ea.STATE_SET),
            tuya.exposes.switchType(),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
            [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
            [4, 'switch_type', tuya.valueConverterBasic.lookup({'toggle': tuya.enum(0), 'state': tuya.enum(1), 'momentary': tuya.enum(2)})],
            [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
            [6, 'countdown', tuya.valueConverter.countdown],
            [14, 'power_on_behavior', tuya.valueConverter.powerOnBehaviorEnum],
            [13, 'light_type', tuya.valueConverter.lightType],
        ],
    },
};

module.exports = definition;

```

ddpsft, 2024-04-13:
It works! Thank you!
About min and max brightness, using Tuya's gateway it just limits how far you can slide the brightness by cutting both ends of the slider. To make it work here it probably would be possible only if you can limit brightness max and min controls based those values instead of using standard 0-255/0-100

https://github.com/Koenkk/zigbee2mqtt/assets/86980245/a584e56c-3efb-41ca-be81-5ccd230b6430

Suxsem, 2024-04-13:
@ddpsft glat it worked!

Are you sure about the min/max? Does this mean that physical button (wall button) doesn't respect the min max values also when dimmer is used with tuya gateway?

Do you have the gateway? I would like to find the datapoint for bulb type

ddpsft, 2024-04-13:
I wasn't able to dim it using physical button. I don't know why, but I could try again. 

I do have the gateway, but those were all datapoints I could get. Maybe something in the image/code below can help you.

[image]
Code	Type	Values
switch_led_1	Boolean	
"{true,false}"
bright_value_1	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
brightness_min_1	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
brightness_max_1	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
countdown_1	Integer	
{
  "unit": "s",
  "min": 0,
  "max": 86400,
  "scale": 0,
  "step": 1
}
switch_led_2	Boolean	
"{true,false}"
bright_value_2	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
brightness_min_2	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
brightness_max_2	Integer	
{
  "min": 10,
  "max": 1000,
  "scale": 0,
  "step": 1
}
countdown_2	Integer	
{
  "unit": "s",
  "min": 0,
  "max": 86400,
  "scale": 0,
  "step": 1
}
relay_status	Enum	
{
  "range": [
    "0",
    "1",
    "2"
  ]
}
switch_type	Enum	
{
  "range": [
    "flip",
    "sync",
    "button"
  ]
}

Suxsem, 2024-04-13:
@ddpsft I tried with physical switch and min/max values are respected! So dp 3 is min brightness and 5 is max brightness.

Only thing missing is bulb type.
Do you know what does the "relay_status" mean? I think it could be the bulb type indeed. Can you try changing the bulb type from the app and see if it changes?

ddpsft, 2024-04-13:
There is no bulb type selection in Tuya's app. I'll post some screens to show will what is available there

Main screen. If you enter any dimmer you get only its individual slider and "settings"
[image]

Entering settings you have this
[image]

and this
[image]

Suxsem, 2024-04-13:
@ddpsft great many thanks! So we could say it's fully supported.

Can you confirm me that the 2-gang version uses dps:

7: switch 2
8: brightness 2
9: min brightness 2
10: switch type 2
11: max brightness 2
12: countdown 2

If so I will open a pull request to add support for both dimmers in Z2M

Thank you

ddpsft, 2024-04-13:
According to Tuya IoT thats correct. I'm trying to put it in your code but with no luck.

It should be something like this in my second channel?
[7, 'state2', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
[8, 'brightness2', tuya.valueConverter.scale0_254to0_1000],

In your case state and brightess is ok considering you only have 1 channel. Here if I use 2 after the parameter it doesnt appear in exposes tab.

Suxsem, 2024-04-13:
@ddpsft give that a try

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_o9gyszw2',
        },
    ],

    model: 'ZDMS16-2',
    vendor: 'Avatto',
    description: 'Zigbee 2 channels Dimmer',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
            tuya.exposes.lightBrightnessWithMinMax().withEndpoint('l1'),
            tuya.exposes.countdown().withEndpoint('l1'),
            tuya.exposes.switchType().withEndpoint('l1'),
            tuya.exposes.lightBrightnessWithMinMax().withEndpoint('l2'),
            tuya.exposes.countdown().withEndpoint('l2'),
            tuya.exposes.switchType().withEndpoint('l2'),
            e.power_on_behavior().withAccess(ea.STATE_SET),
    ],
    meta: {
		multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [2, 'brightness_l1', tuya.valueConverter.scale0_254to0_1000],
            [3, 'min_brightness_l1', tuya.valueConverter.scale0_254to0_1000],
            [4, 'switch_type_l1', tuya.valueConverterBasic.lookup({'toggle': tuya.enum(0), 'state': tuya.enum(1), 'momentary': tuya.enum(2)})],
            [5, 'max_brightness_l1', tuya.valueConverter.scale0_254to0_1000],
            [6, 'countdown_l1', tuya.valueConverter.countdown],
            [7, 'state_l2', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [8, 'brightness_l2', tuya.valueConverter.scale0_254to0_1000],
            [9, 'min_brightness_l2', tuya.valueConverter.scale0_254to0_1000],
            [10, 'switch_type_l2', tuya.valueConverterBasic.lookup({'toggle': tuya.enum(0), 'state': tuya.enum(1), 'momentary': tuya.enum(2)})],
            [11, 'max_brightness_l2', tuya.valueConverter.scale0_254to0_1000],
            [12, 'countdown_l2', tuya.valueConverter.countdown],
            [14, 'power_on_behavior', tuya.valueConverter.powerOnBehaviorEnum],
        ],
    },
	endpoint: (device) => {
		return {'l1': 1, 'l2': 1};
	},
};

module.exports = definition;
```

ddpsft, 2024-04-13:
Great! Now it's time to create that pull request. Thank you for your help

Suxsem, 2024-04-13:
so it works? :) I will open the PR tomorrow

nilshoell, 2024-05-23:
@ddpsft did you get the dimming using the physical switch to work?
If I do a long press on the switch the light just turns on/off, but I don't know if that just doesn't work as advertised or if it is an issue somewhere else.

mrmaximas, 2024-05-24:
> @ddpsft did you get the dimming using the physical switch to work? If I do a long press on the switch the light just turns on/off, but I don't know if that just doesn't work as advertised or if it is an issue somewhere else.

what type of physical switch do you use?

mrmaximas, 2024-05-24:
@nilshoell what type of physical switch do you use?

nilshoell, 2024-05-24:
A momentary switch, and I set the switch type accordingly in z2m.
Turning on/off with a short press works as expected, but the same happens on a long press of the switch.
When set to `toggle` the light would only stay on as long as the switch as pressed (which makes sense), while the behavior for `state` is the same as with `momentary` (which also makes sense, although I wouldn't expect dimming to work with a state switch).

mrmaximas, 2024-05-24:
> A momentary switch, and I set the switch type accordingly in z2m. Turning on/off with a short press works as expected, but the same happens on a long press of the switch. When set to `toggle` the [...]

try short press the switch and then hold it?

Suxsem, 2024-05-24:
Maybe the bulb is not dimmable?

nilshoell, 2024-05-24:
No, the bulb is dimmable, I can dim it via z2m, just not via the physical button/switch.
I will try @mrmaximas suggestion and report back, but then this module would have a different interaction pattern than the rest I have deployed.

If it works for the rest oft you maybe I just have a faulty unit or an early, buggy firmware.

Suxsem, 2024-05-24:
Min/max settings are correct?

ddpsft, 2024-05-28:
> @ddpsft did you get the dimming using the physical switch to work? If I do a long press on the switch the light just turns on/off, but I don't know if that just doesn't work as advertised or if it is an issue somewhere else.

Last month after Suxsem helped with the converter I just tested using Z2M and turn it off. As soon as i return to my country I'll try it to update you.

nilshoell, 2024-06-02:
Ok, I did some more testing using a different module, still no luck with the physical switch dimming. If I short press & then hold it just toggles the light regardless.
Seems like this module only supports dimming via Z2M, not via the physical switch.

Thanks for all your help :)

Suxsem, 2024-06-02:
@nilshoell mine works without any issue. Have you checked the min/max brightness settings?

nilshoell, 2024-06-02:
Those setting don't seem to do anything for the physical switch at all - it just toggles between off and whatever brightness I last set manually in z2m, regardless of where the min/max values are. Doesn't matter how long I press, or if I do a double-press or anything else.
I also tried the other endpoints for the second gang, same results, on multiple devices.

Maybe I just got a batch of modules with older firmware? Because clearly your experience is different.
Or could it be a difference between the 2-channel and 1-channel version? I only have the 2-channel (`ZDMS16-2`).

Suxsem, 2024-06-02:
Very strange. Let's wait for @ddpsft to see the behavior of his device.
Just to confirm, have you set the switch type to "momentary" right?

nilshoell, 2024-06-02:
Yes exactly, `switch_type` is set to `momentary` (and I have a momentary switch hooked up, other modules like the `TS110E_1gang_1` work fine including dimming on the physical switch).

ddpsft, 2024-06-03:
https://github.com/Koenkk/zigbee2mqtt/assets/86980245/5010d18c-1f89-42f4-8e7f-7c2b09668707

I really dont remember the behavior last month, but now It cant dim using momentary swtich, only thru Z2M.

Sloy, 2024-06-30:
Hello! Any updates on the physical switch dimming? I'm looking into getting a few of these dimmers, but if the physical switch can't dim it's a deal breaker for me 😕

ddpsft, 2024-06-30:
I'm not sure if I would recommend it. As in the video i've shared it's apparently unable to dim using physical switch and after that it stopped working (at least one channel). I was using it to test some automations and after a couple minutes it started flickering and now I'm unable even to turn it on/off.

Sloy, 2024-07-01:
Oh, I see. I'll pass then. Thanks for the heads up, and for all the testing you have done ❤️

Suxsem, 2024-07-01:
Mine is working perfectly since some month now. And dimming using physical switch works to. I'm on holidays now but when home I will send you some picture of my actual device

samumar82, 2024-07-03:
Hello, I've got this 1-gang model:ZDMS16-1.
Honestly I got more than one but only one is behaving differently: it has a really slow fade-in effect when turning on which is annoying. Also when turning off but it is slightly faster and less annoying.

This is the setup I have:

    "brightness": 229,
    "linkquality": 99,
    "max_brightness": 255,
    "min_brightness": 1,
    "power_on_behavior": "off",
    "state": "OFF",
    "countdown": null,
    "switch_type": null

Any suggestion to make it work and light up almost instantly?
All the others dimmers have the same downlights and are working just fine with a very fast fade-in effect which I love!
Thanks for helping!

Suxsem, 2024-07-03:
Can you query the firmware version using the developer tab? Maybe it has an older firmware

samumar82, 2024-07-03:
Using MqttExplorer looks like it's 1.39.0. Does it ring a bell?

Suxsem, 2024-07-03:
Can you compare it with the other dimmers you have?

samumar82, 2024-07-03:
Sure thing, I'll be back home in Saturday and will update, thank you.

samumar82, 2024-07-03:
What I remember I that after the first install it was working ok just like the others but then I accident clicked on "momentary" switch type and started fading. Tried all the switch types but nothing and ended up sending a new Json to remove the switch type as I was not able to remove it via the GUI but the fade effect is still there :-(

OlivierMarty, 2024-10-26:
For the next guy searching for these devices : I bought 2 for testing. They work OK but they don't support binding with a remote.

mogim0gi, 2024-10-26:
i wonder if they will survive 120w of 220v leds on one channel, other channel is 24w. will post in a month or so.

MichiMess, 2024-11-01:
> This is where I am so far.
> 
> Working:
> 
> * ON/OFF
> * BRIGHTNESS CONTROL
> * COUNTDOWN
> * SWITCH TYPE
> * MIN BRIGHTNESS SETTING
> * MAX BRIGHTNESS SETTING
> 
> Not working:
> 
> * LIGHT TYPE SETTING
> 
> I'm not sure of these settings datapoints. Strangely enough switch type is on datapoint 4 (where usually there is light type setting) so I don't really known which datapoint corresponds to light type and I don't have a tuya gateway to test. Min and max brightness should be on datapoints 3 and 5 but I really don't see any difference in brightness when I change these values. I hope that someone with tuya gateway can tell us the right datapoints for the missing settings.
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     fingerprint: [
>         {
>             modelID: 'TS0601',
>             manufacturerName: '_TZE204_5cuocqty',
>         },
>     ],
> 
>     model: 'ZDMS16-1',
>     vendor: 'Avatto',
>     description: 'Zigbee 1 channel Dimmer',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     configure: tuya.configureMagicPacket,
>     exposes: [
>             tuya.exposes.lightBrightnessWithMinMax(),
>             tuya.exposes.countdown(),
>             tuya.exposes.lightType(),
>             e.power_on_behavior().withAccess(ea.STATE_SET),
>             tuya.exposes.switchType(),
>     ],
>     meta: {
>         tuyaDatapoints: [
>             [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
>             [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
>             [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
>             [4, 'switch_type', tuya.valueConverterBasic.lookup({'toggle': tuya.enum(0), 'state': tuya.enum(1), 'momentary': tuya.enum(2)})],
>             [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
>             [6, 'countdown', tuya.valueConverter.countdown],
>             [14, 'power_on_behavior', tuya.valueConverter.powerOnBehaviorEnum],
>             [13, 'light_type', tuya.valueConverter.lightType],
>         ],
>     },
> };
> 
> module.exports = definition;
> ```

Thaaaanks!! This one is working with the _TZE204_nqqylykc also!!

Czeczi94, 2024-11-16:
https://pl.aliexpress.com/item/1005007097728653.html - i bought this dimmer but can't add it to zigbee2mqtt. Actually it connects but says it is not supported. Tried various ways to enable manual support but i'm still struggling with it. Could anyone point me to a guide how to configure it?

Giannarciso, 2024-11-16:
I can confirm this works for _TZE204_jtbgusdc too
Its ZDMS16-2 (2 gang dimmer from Avatto)

Thank you 

For Czeczi94:

1. Pair you unrecognized device
2. Search Zigbee Manufacturer, something like  _TZE204_jtbgusdc (in my case)
3. Create a file with namexxx.js in z2m directory or where you like it
4. In the Frontend go for Settings >> External converters >> in the box put the address of your file ej. /config/zigbee2mqtt/ZDMS16-2.js (in my case) >> Submit
5. Restart Z2M

Check devices, should be supported now

Hope it helps

Czeczi94, 2024-11-17:
> I can confirm this works for _TZE204_jtbgusdc too Its ZDMS16-2 (2 gang dimmer from Avatto)
>
> Thank you
>
> For Czeczi94:
>
> 1. Pair you unrecognized device
> 2. Search Zigbee Manufacturer, something like [...]

Can you expand on 2. - where do i search for it?

Giannarciso, 2024-11-17:
In the frontend click on the unrecognized device >> dev console >> generate_external_definition >> in the box see:     vendor: '_TZE204_jtbgusdc' in my case

There are other ways but thats one

Czeczi94, 2024-11-17:
> * In the Frontend go for Settings >> External converters >> in the box put the address of your file ej. /config/zigbee2mqtt/ZDMS16-2.js (in my case) >> Submit

Thanks for that. Seems that I was struggling because my .js file was not seen by zigbee2mqtt - i was trying to point to it via the configuration.yaml and probably i incorrectly configured it. But after putting the path into the Settings >> External converters >> in the box >> submit - everything worked fine. Now i have the device as supported.

pawel1313, 2024-12-03:
> Thanks for that. Seems that I was struggling because my .js file was not seen by zigbee2mqtt - i was trying to point to it via the configuration.yaml and probably i incorrectly configured it. But [...]

Hi.
I have the same module. The module is not recognized.
Did you manage to get it working?
How did you solve the problem?

marekmeli, 2024-12-07:
Hello, I have the same modules, both 1-channel and 2-channel versions. None of them are able to control dimming using physical switch, which is really annoying. I also tried to expose the event of holding the button (to set the dimming functionality manually via HA scripting), but it looks like there are no switch states reported by the units.

Any luck with yours?

pawel1313, 2024-12-12:
> Cześć, mam te same moduły, zarówno 1-kanałowe, jak i 2-kanałowe. Żaden z nich nie jest w stanie kontrolować ściemniania za pomocą fizycznego przełącznika, co jest naprawdę denerwujące. Próbowałem również ujawnić zdarzenie przytrzymania przycisku (aby ręcznie ustawić funkcjonalność ściemniania za pomocą skryptu HA), ale wygląda na to, że jednostki nie zgłaszają żadnych stanów przełączników.
> 
> Udało Ci się?

I couldn't even add it correctly to zigbee2MQTT. That's why I can't control the light from the app. These modules don't support dimming from the button.

leolstein, 2024-12-24:
Hi guys,

I'm a little newbie but I add that device to ST, using the Edge Personal Tuya Devices - Generic EF00 Device Driver.
as appear in this link: https://community.smartthings.com/t/st-edge-personal-tuya-devices-generic-ef00-device/255270

The problem is, I just can turn on or off the module, someone can help?

Thx.

closed 2025-03-04 by github-actions[bot]

FPSUsername, 2025-05-11:
Similarly to the TS110E models, is it possible that these units can support the [transition](https://github.com/Koenkk/zigbee2mqtt/issues/26757) feature?

The TS110E is quite unreliable (disconnects with 160-180 LQI, incorrect status reports, timeouts receiving messages, physical button sometimes doesn't turn on/off the light, mainly on L1), so I'm looking for an alternative.

FPSUsername, 2025-05-21:
Seems that this model is a lot more reliable than the TS110E 2 gang dimmer.

radzio, 2026-02-19:
Do you notice delays when using the physical switch? I have to wait about one second after pressing the switch to see the light turn on.