Koenkk/zigbee2mqtt #22176: 4 gang switch+2way scenes Zigbee switch(_TZ3000_qhyadm57) shows 4 gang switch only

hytepro, 2024-04-11:
### What happened?

**Database entry:**
{"id":2,"type":"Router","ieeeAddr":"0x847127fffe50af9d","nwkAddr":28261,"manufId":4098,"manufName":"_TZ3000_qhyadm57","powerSource":"Mains (single phase)","modelId":"TS0726","epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":4,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0726","manufacturerName":"_TZ3000_qhyadm57","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":69,"powerSource":1}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe3973ed","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":4,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe3973ed","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":4,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe3973ed","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":4,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe3973ed","endpointID":1}],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":4,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchMode":1}}},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":260,"epId":6,"devId":4,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchMode":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":971637444},"lastSeen":1712803546098}

### What did you expect to happen?

### **generated_external_definition:**

const {deviceEndpoints, identify, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0726'],
    model: 'TS0726',
    vendor: '_TZ3000_qhyadm57',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4,"5":5,"6":6}}), identify(), onOff({"powerOnBehavior":false,"endpointNames":["1","2","3","4","5","6"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;

### How to reproduce it (minimal and precise)

press all of the 4 gang switch are working normally, but the other two scene buttons have no response, and these two scene switches are not displayed on the Z2M device interface.

I checked all the issue articles about TS0726, but I didn't find any news about _TZ3000_qhyadm57

The same 4*switch+2*scenes device worked normally in the previous version of Z2M by adding External definition, but it does not work properly based on 1.36.1

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8

### Setup

containerized

### Debug log

Debug 2024-04-11 12:03:08Received MQTT message on 'zigbee2mqtt-1/bridge/request/options' with data '{"options":{"advanced":{"log_level":"info"}},"transaction":"be9q3-1"}'
Info 2024-04-11 12:03:08Successfully changed options
Info 2024-04-11 12:03:08MQTT publish: topic 'zigbee2mqtt-1/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"be9q3-1"}'
Info 2024-04-11 12:04:00MQTT publish: topic 'zigbee2mqtt-1/0x847127fffe50af9d', payload '{"action":null,"linkquality":124,"power_on_behavior":"previous","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state":"ON","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","switch_mode_l1":null,"switch_mode_l2":null,"switch_mode_l3":null,"switch_mode_l4":null}'
Info 2024-04-11 12:04:00MQTT publish: topic 'zigbee2mqtt-1/0x847127fffe50af9d', payload '{"action":null,"linkquality":124,"power_on_behavior":"previous","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state":"ON","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","switch_mode_l1":null,"switch_mode_l2":null,"switch_mode_l3":null,"switch_mode_l4":null}'
Info 2024-04-11 12:04:00MQTT publish: topic 'zigbee2mqtt-1/0x847127fffe50af9d', payload '{"action":null,"linkquality":124,"power_on_behavior":"previous","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","switch_mode_l1":null,"switch_mode_l2":null,"switch_mode_l3":null,"switch_mode_l4":null}'
Info 2024-04-11 12:04:00MQTT publish: topic 'zigbee2mqtt-1/0x847127fffe50af9d', payload '{"action":null,"linkquality":124,"power_on_behavior":"previous","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","switch_mode_l1":null,"switch_mode_l2":null,"switch_mode_l3":null,"switch_mode_l4":null}'
Info 2024-04-11 12:04:01MQTT publish: topic 'zigbee2mqtt-1/0x847127fffe50af9d', payload '{"action":null,"linkquality":128,"power_on_behavior":"previous","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state":"ON","state_l1":"OFF","state_l2":"ON","state_l3":"OFF","state_l4":"OFF","switch_mode_l1":null,"switch_mode_l2":null,"switch_mode_l3":null,"switch_mode_l4":null}'
Info 2024-04-11 12:04:01MQTT publish: topic 'zigbee2mqtt-1/0x847127fffe50af9d', payload '{"action":null,"linkquality":124,"power_on_behavior":"previous","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state":"ON","state_l1":"OFF","state_l2":"ON","state_l3":"OFF","state_l4":"OFF","switch_mode_l1":null,"switch_mode_l2":null,"switch_mode_l3":null,"switch_mode_l4":null}'
Info 2024-04-11 12:04:02MQTT publish: topic 'zigbee2mqtt-1/0x847127fffe50af9d', payload '{"action":null,"linkquality":120,"power_on_behavior":"previous","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state":"ON","state_l1":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"OFF","switch_mode_l1":null,"switch_mode_l2":null,"switch_mode_l3":null,"switch_mode_l4":null}'
Info 2024-04-11 12:04:02MQTT publish: topic 'zigbee2mqtt-1/0x847127fffe50af9d', payload '{"action":null,"linkquality":120,"power_on_behavior":"previous","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state":"ON","state_l1":"OFF","state_l2":"ON","state_l3":"ON","state_l4":"OFF","switch_mode_l1":null,"switch_mode_l2":null,"switch_mode_l3":null,"switch_mode_l4":null}'
Info 2024-04-11 12:04:02MQTT publish: topic 'zigbee2mqtt-1/0x847127fffe50af9d', payload '{"action":null,"linkquality":116,"power_on_behavior":"previous","power_on_behavior_l1":"previous","power_on_behavior_l2":"previous","power_on_behavior_l3":"previous","power_on_behavior_l4":"previous","state":"ON","state_l1":"OFF","state_l2":"ON","state_l3":"OFF","state_l4":"OFF" [... cut ...] tt-1/0x847127fffe50af9d","unique_id":"0x847127fffe50af9d_linkquality_zigbee2mqtt-1","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2024-04-11 12:05:12MQTT publish: topic 'zigbee2mqtt-1/bridge/response/device/options', payload '{"data":{"from":{"legacy":false,"state_action":true},"id":"0x847127fffe50af9d","restart_required":false,"to":{"homeassistant":{},"legacy":false,"optimistic":true,"retain":true,"state_action":true}},"status":"ok","transaction":"be9q3-3"}'

hytepro, 2024-04-13:
press all of the 4 gang switch are working normally, but the other two scene buttons have no response, and these two scene switches are not displayed on the Z2M device interface.

I checked all the issue articles about TS0726, but I didn't find any news about _TZ3000_qhyadm57

The same 4switch+2scenes device worked normally in the previous version of Z2M by adding External definition, but it does not work properly based on 1.36.1

hyteplus, 2024-04-26:
Closed! There was no any answer

Koenkk, 2025-03-11:
Will be added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8967