Koenkk/zigbee2mqtt #22183: Hive HALIGHTDIMWWE27 power_on_behavior unsupported attribute

mjf2708, 2024-04-11:
### What happened?

Zigbee2MQTT Frontend shows Power-on behavior is exposed by this bulb: Hive HALIGHTDIMWWE27. However, attempting to change brings up the error message UNSUPPORTED_ATTRIBUTE.

### What did you expect to happen?

Change to this attribute to one of the shown values would occur.

### How to reproduce it (minimal and precise)

Error message appears whenever an attempt is made to change this attribute. Also, periodic error messages appear in Frontend without any action.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Plain on Raspberry Pi 4 Bookworm

### Debug log

error 2024-04-11 14:20:00: Publish 'get' 'power_on_behavior' to 'Hive_bulb' failed: 'Error: ZCL command 0x00158d00013bbf77/1 genOnOff.read(["startUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Koenkk, 2024-04-13:
Device doesn't seem to support it, disabled!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-13 by Koenkk

mjf2708, 2024-04-15:
Unable to apply dev update, following instructions 'Switch to the dev branch':
```
pi@raspberrypi:/opt/zigbee2mqtt $ git fetch origin dev
From https://github.com/Koenkk/zigbee2mqtt
 * branch              dev        -> FETCH_HEAD
pi@raspberrypi:/opt/zigbee2mqtt $ git checkout dev
error: pathspec 'dev' did not match any file(s) known to git
pi@raspberrypi:/opt/zigbee2mqtt $ git fetch origin --unshallow
fatal: --unshallow on a complete repository does not make sense
```