Koenkk/zigbee2mqtt #22184: IKEA Badring E2202 not reporting water leak correctly

francisp2, 2024-04-11:
### What happened?

Bought 2 Badring water sensors, they pair, say supported, but stay like this
[image]

So it made one a little wet, it said 'Leaking', but 36 hours later, it still said 'Leaking'. I removed it, paired again, back to water leak : Null

Related to  IKEA Badring E2202 not reporting water leak correctly #21794 but that one is closed.

### What did you expect to happen?

I expected it to return to 'Not leaking' or something after it dried

### How to reproduce it (minimal and precise)

make it a little bit wet, and let it dry

### Zigbee2MQTT version

    1.36.1 commit: ffc2ff1d

### Adapter firmware version

20230507

### Adapter

2020 model of TubeZB non POE Ethernet coordinator

### Setup

bare metal (npm install) , happens both on proxmox vm or bare metal on pi 4, both running dietpi bookworm

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14947917/log.txt)

mrskycriper, 2024-04-11:
What happens in the logs while you test the sensor? The whole file is very long and not very informative. Can you make a controlled test of the sensor and copy the relevant results out of the log file?

aurelmarius, 2024-04-11:
I have this sensor lying arround, i will do some testing.

aurelmarius, 2024-04-11:
Can't reproduce the issue on 1.36.1-dev commit: [f774753] with SLZB-06 CC2652P fw 20230507. One difference is that mine doesn't expose Low Battery.
Some pictures and logs attached.

[image]
[image]
[image]
[zigbee2mqtt_logs_badring.txt](https://github.com/Koenkk/zigbee2mqtt/files/14950203/zigbee2mqtt_logs_badring.txt)

francisp2, 2024-04-13:
12 hours later, Zigbee network just consist of the Zigstar v4 and the Badring, still 'leaking'
[error.txt](https://github.com/Koenkk/zigbee2mqtt/files/14965155/error.txt)

aurelmarius, 2024-04-13:
@francisp2 upgrade to dev branch, mine is working ok on dev.

francisp2, 2024-04-14:
No. Installed dev, still 'Leaking', Exposes screen changed, but that is all.
[image]
[image]
[out.txt](https://github.com/Koenkk/zigbee2mqtt/files/14969432/out.txt)

Great-Chart, 2024-04-16:
I've found that if I short out (via screwdriver) the underside pins to trigger the alarm / wet (leaking) event(s); then once the short has been removed the device will then reset to display clear and (appears to) then stay persistent.
Prior to that the device had stuck in a flashing leaking state as per the other posts above (in my case I left it overnight).

Not on a dev branch, running Sonoff Dongle-P.
Uncertain how long this might now remain displaying correctly or if it'll revert to leaking over an extended period.

[image]

francisp2, 2024-05-13:
still the case on 1.37.1

francisp2, 2024-06-01:
still the case on 1.38

dg7pc, 2024-06-03:
I can confirm the behaviour with 2 devices. With one the haning leak state cleared after using the fingers to shortcut the sensor. The second (same Firmware 1.0.7) does not clear and still shows a leak.
Updated to: Zigbee2MQTT version 1.38.0-dev (commit #6c123c2)

Mark-64, 2024-06-04:
Same here with 2x BADRIG sensors and Z2M 1.37.1, both sensor stuck on wet.
If I shurtcut pins with the fingers mutliple times, then the sensors go back to dry, very unreliable.

alexruffell, 2024-06-24:
I have the same issue with 2 Badring sensors. I found that if you short the contacts for a very short period of time (it varies but <3 seems to reproduce the issue reliably) it often stays stuck in wet. Another short and it reverts to dry. I use ZHA so the issue may be related to the firmware of the device more than Zigbee2MQTT or ZHA.

where-is-s, 2024-07-29:
It's the same for me. IKEA Badring gets stuck on wet while working through Zigbee2MQTT 1.39.0. Did anyone manage to resolve this somehow?

vasileio, 2024-07-30:
> It's the same for me. IKEA Badring gets stuck on wet while working through Zigbee2MQTT 1.39.0. Did anyone manage to resolve this somehow?

Just shorting the contacts again resets it. It definitely looks like a firmware issue, rather than a Z2M or ZHA one, as the behaviour is the same in either, and also on-device (it does keep on beeping)

Elephantoplasty, 2024-08-09:
Another person who is finding the Badring gets stuck showing leaking after moisture activation.  And shorting the terminals also results in a leaking output, but it then resets back to normal within a few seconds.

where-is-s, 2024-08-12:
> Just shorting the contacts again resets it. It definitely looks like a firmware issue, rather than a Z2M or ZHA one, as the behaviour is the same in either, and also on-device (it does keep on beeping)

Yeah, that works of course. But in my case device stops beeping correctly, only Zigbee reporting is broken. I have to short contacts 1-3 times in order to reset device's state back to "dry". It is highly inconvenient because I have big anti-leakage scenarios connected to the sensor and it is important to me that the sensor reports that it is wet only while it is actually wet. I thought maybe there is some kind of workaround that could make it work without having to manually short contacts.

closed 2024-08-18 by Koenkk