Koenkk/zigbee2mqtt #22185: Bosch BMCT-SLZ Light/shutter control unit II: Failed to set Device mode (Status 'UNSUPPORTED_ATTRIBUTE')

tetemba, 2024-04-11:
### What happened?

After correctly pairing the device with Z2M (using the Home Assistant add-on gui), I'm trying to set the Device mode option in device properties > Expose, but I get an error:
`Publish 'set' 'device_mode' to 'Bosch' failed: 'Error: ZCL command Bosch/1 manuSpecificBosch10.write({"deviceMode":4}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4403,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

I also tried publishing {"device_mode": "light"} to zigbee2mqtt/FRIENDLY_NAME/set via HA development tools, as well as the MQTT Explorer app - with no result.

### How to reproduce it (minimal and precise)

Home Assistant > Zigbee2MQTT Add-on > click on the friendly name of the device > Expose tab > click on the "refresh" blue button next to Device mode.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

6.5.0.0 build 188

### Adapter

EZSP v7

### Setup

Add-on on Home Assistant, Virtual Machine

sopelt, 2024-04-11:
Hi @tetemba ... I seem to be facing the same issue... 7 instances of BMCT-SLZ that no longer accept any changes to the mode or calibration settings citing `UNSUPPORTED_ATTRIBUTE`.
I know that I set them up with `1.36.0` and calibrated them a few days later... then upgraded to `1.36.1` so it might be a regression between those two versions. Do you have an easy way to see if 1.36.0 works as expected? I probably won't be able to reproduce that right this moment ...

tetemba, 2024-04-11:
Hi @sopelt and thank you for the advice. I restored the Z2M version 1.36.0 from backup and the problem does not occur - I can change the device mode option in the gui - so the problem is with Z2H version 1.36.1.

sopelt, 2024-05-03:
@nice-tnetennba ... you don't happen to know if 1.37.0 works as expected? ;)

nice-tnetennba, 2024-08-02:
Very late reply, but I had a house renovation and forgot about the problem - it does not appear in newer versions.

closed 2024-08-02 by nice-tnetennba