Koenkk/zigbee2mqtt #22186: New device: Ikea LED2110R3 (Replacement for LED1923R5, IKEA Colored GU10)

janerikvw, 2024-04-11:
### Link

https://www.ikea.com/nl/nl/p/tradfri-led-lamp-gu10-345-lumen-draadloos-dimbaar-gekleurd-en-wit-spectrum-70547474/

### Database entry

{"id":2,"type":"Unknown","ieeeAddr":"0x3410f4fffe67ef5e","nwkAddr":41403,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1712849529297}

### Comments

This is the updated version of the old Colored Ikea GU10 lights. Stated as here: https://www.zigbee2mqtt.io/devices/LED1923R5_LED1925G6.html

### External definition

```shell
-
```

mrskycriper, 2024-04-11:
What zigbee model does it report? Database entry seems corrupted.

janerikvw, 2024-04-12:
It seems that it was an error on my side. I reinstalled everything and now they are detected as: https://www.zigbee2mqtt.io/devices/LED1923R5_LED1925G6.html

closed 2024-04-12 by janerikvw

DeliusPit, 2024-08-16:
Today I accidentally bought a LED2210R3. The box says LED2210R3, the manual also mentions LED1923R5.

So far I haven't been able to pair it with zigbee2mqtt, in the zigbee2mqtt I can find the 1923, but not the 2210. Does that mean that the 2210 is indeed not supported (yet)?

maxlojr, 2024-08-25:
Hello, same for me here

ahb-bs, 2025-02-23:
I've add LED2110R3 to with my Conbee II / Deconz and they appear and show, but after a few minutes stop responding.

ahb-bs, 2025-11-20:
I moved to a SLZB-06M in the end, as I got annoyed trying to make it work.