Koenkk/zigbee2mqtt #22189: Error: ZCL command genOnOff.on

rw-django-fan-2020, 2024-04-11:
### What happened?

My Home Assistant System crashed. Dont know why. Had to move to another Device (ThinClient). Restored my Home Assistant Backup. But zigbee dont work anymore. Help. 

### What did you expect to happen?

I expected it to work.

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

 0x26780700

### Adapter

ConBee2/RaspBee2

### Setup

Add-on on Home Assistant OS on Proxmox VM

### Debug log

error 2024-04-12 00:24:18Publish 'set' 'state' to 'Blitzwolf Steckdose 2' failed: 'Error: ZCL command 0x3425b4fffe4984d2/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (76))'
debug 2024-04-12 00:24:18Error: ZCL command 0x3425b4fffe4984d2/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (76)) at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:666:23) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:591:24) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

robertomontinaro80, 2024-04-12:
Similar error for me for that getonoff command after the update 
I can't turn on any lamp in my home.
I add also this error stack trace if it can help

`18Publish 'set' 'state' to 'osram_kitchen_lamp' failed: 'Error: ZCL command 0x7cb03eaa0a038452/3 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`

gentleman1979, 2024-04-12:
Similar problems for a few days 
`Zigbee2MQTT:error 2024-04-12 22:25:07: Publish 'set' 'state' to 'SFZ-Wand-rechts' failed: 'Error: ZCL command 0x00178801037e0089/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'`

albert-canfield, 2024-04-12:
Similar issue too with a SONOFF ZBMINIL2 switch. 

Publish 'set' 'state' to 'Entrance lights' failed: 'Error: ZCL command 0x0ceff6fffe16f9fb/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 33295 - 1 - 36 - 6 - 11 after 10000ms)'

majkers, 2024-04-13:
Same. Here. I updated to 1.36.1-1 from 1.35.2-1 because I wanted to test https://github.com/Koenkk/zigbee2mqtt/issues/21162

`Publish 'set' 'state' to 'gniazdko_xiaomi' failed: 'Error: Command 0x00158d0002347282/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (16))'`

robertomontinaro80, 2024-04-13:
> Same here. I updated to 1.36.1-1 from 1.35.2-1 because I wanted to test #21162
> 
> `Publish 'set' 'state' to 'gniazdko_xiaomi' failed: 'Error: Command 0x00158d0002347282/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (16))'`
> 
> `Publish 'set' 'state' to 'gniazdko_neo' failed: 'Error: Command 0xa4c1389a7623f41a/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (17))'`
> 
> They are Xiaomi ZNCZ02LM and Tuya TS011F_plug_1

For me ver 1.36.0 works fine. 1.36.1 does not.

hytepro, 2024-04-13:
Similar issue

Pepeuch, 2024-04-13:
Similaire issue with conbee 2 hassos supervised on rpi4 8go.

blakefrain, 2024-04-14:
I am having the same issue conbee 2 and multiple connected devices

vencakratky, 2024-04-16:
Similar Issue with Immax bulbs

galant-ho, 2024-04-17:
Having similar issue with my Aqara E1 Roller Shade Driver starting last week.  Thought it was my setting issue so repair it several times.  The E1 paired successfully, able to control the blind for certain times until the next day, lost control from HA.

`2024-04-17 08:03:54Publish 'set' 'state' to 'E1-blind' failed: 'Error: ZCL command 0x54ef4410009ff334/1 genAnalogOutput.write({"presentValue":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47747 - 1 - 208 - 13 - 4 after 10000ms)'`

shanec42, 2024-04-18:
I've got the same problem as others above.  Here's the important failure, but I'm including the full log just in case it helps.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/15019245/log.txt)

Zigbee2MQTT version [1.36.1] (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1) commit: [ffc2ff1](https://github.com/Koenkk/zigbee2mqtt/commit/ffc2ff1)
Coordinator type  zStack3x0
Coordinator revision  20230507
Coordinator IEEE Address  0x00124b002a2f28c2
Frontend version  0.6.161
zigbee_herdsman_converters_version  19.11.2
zigbee_herdsman_version  0.40.3

debug 2024-04-17 20:17:19: Received MQTT message on 'zigbee2mqtt/Office Ceiling Light 4/set' with data '{"state":"OFF"}'
debug 2024-04-17 20:17:19: Publishing 'set' 'state' to 'Office Ceiling Light 4'
error 2024-04-17 20:17:42: Publish 'set' 'state' to 'Office Ceiling Light 4' failed: 'Error: ZCL command 0xb0ce18140352fa2f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53320 - 1 - 64 - 6 - 11 after 10000ms)'
debug 2024-04-17 20:17:42: Error: ZCL command 0xb0ce18140352fa2f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53320 - 1 - 64 - 6 - 11 after 10000ms)
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47)
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:591:24)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:1129:32)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

siddharthghiya, 2024-04-18:
Hi, I was having the same issue where I restored my HA installation from a backup. Zigbee stopped working. I also had individual add-on backups, and things started working again after I downgraded zigbee2mqtt from 1.36.1-1 to 1.36.0-1.

It started working again after downgrading to 1.36.0-1 and then I upgraded back to 1.36.1.-1. Hope this helps!

Aposei, 2024-04-20:
Seems like the same problem to me. 

zigbee2mqtt 1.36.1-1 (but I can't change to 1.36.0-1)

`Error 2024-04-20 14:19:20Publish 'set' 'state' to 'Plancha' failed: 'Error: ZCL command 0xa4c138e4e6ae7bd2/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43659 - 1 - 76 - 6 - 11 after 10000ms)'
Debug 2024-04-20 14:19:20Error: ZCL command 0xa4c138e4e6ae7bd2/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 43659 - 1 - 76 - 6 - 11 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:492:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:591:24) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)`

bezdekf, 2024-04-20:
The same problem:

Zigbee2MQTT:error 2024-04-20 16:43:04: Publish 'set' 'state' to 'IKEA 3' failed: 'Error: ZCL command 0xcc86ecfffe8d422e/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:error 2024-04-20 16:43:06: Publish 'set' 'state' to 'WOOX  1' failed: 'Error: ZCL command 0x588e81fffede3426/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

etc

paboman, 2024-04-20:
same here:

`Failed to configure '0xa4c1387bc939184d', attempt 2 (Error: ZCL command 0xa4c1387bc939184d/1 haElectricalMeasurement.configReport([{"attribute":"rmsCurrent","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":50}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:561:9) at Object.rmsCurrent (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:216:5) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3773:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:162:21) at Configure.configure (/app/lib/extension/configure.ts:122:13))`

bezdekf, 2024-04-20:
I didn't change anything and suddenly after 2 days everything works. Grrrr...

> The same problem:
>
> Zigbee2MQTT:error 2024-04-20 16:43:04: Publish 'set' 'state' to 'IKEA 3' failed: 'Error: ZCL command 0xcc86ecfffe8d422e/1 genOnOff.off({}, [...]

I didn't change anything and suddenly after 2 days everything works. But just to be sure, I'm switching from the not recommended CC 2531 to the SONOFF ZB Dongle-P Zigbee 3.0

bezdekf, 2024-04-20:
> The same problem:
>
> Zigbee2MQTT:error 2024-04-20 16:43:04: Publish 'set' 'state' to 'IKEA 3' failed: 'Error: ZCL command 0xcc86ecfffe8d422e/1 genOnOff.off({}, [...]

I didn't change anything and suddenly after 2 days everything works. But just to be sure, I'm switching from the not recommended CC 2531 to the SONOFF ZB Dongle-P Zigbee 3.0

Pepeuch, 2024-04-20:
Est il possible d’avoir une réponse à cette erreur qui est récurrente et surtout qui bloque tout fonctionnement? Par avance merci

mhzawadi, 2024-04-21:
I now have this issue too, was working fine till a stop and start of the z2m.

```
2024-04-21 10:31:22Publish 'set' 'state' to '01 - switch - mbed' failed: 'Error: ZCL command 0x00124b00246cb0b6/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (70))'
```

Conbee2 in a Pi 4

```
DEVLINKS: >-
  /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436161-if00
  /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.2:1.0
DEVNAME: /dev/ttyACM0
DEVPATH: >-
  /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.0/tty/ttyACM0
ID_BUS: usb
ID_MODEL: ConBee_II
ID_MODEL_ENC: ConBee\x20II
ID_MODEL_ID: '0030'
ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.2:1.0
ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0-usb-0_1_2_1_0
ID_REVISION: '0100'
ID_SERIAL: dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436161
ID_SERIAL_SHORT: DE2436161
ID_TYPE: generic
ID_USB_DRIVER: cdc_acm
ID_USB_INTERFACES: ':020201:0a0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: ConBee_II
ID_USB_MODEL_ENC: ConBee\x20II
ID_USB_MODEL_ID: '0030'
ID_USB_REVISION: '0100'
ID_USB_SERIAL: dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436161
ID_USB_SERIAL_SHORT: DE2436161
ID_USB_TYPE: generic
ID_USB_VENDOR: dresden_elektronik_ingenieurtechnik_GmbH
ID_USB_VENDOR_ENC: dresden\x20elektronik\x20ingenieurtechnik\x20GmbH
ID_USB_VENDOR_ID: 1cf1
ID_VENDOR: dresden_elektronik_ingenieurtechnik_GmbH
ID_VENDOR_ENC: dresden\x20elektronik\x20ingenieurtechnik\x20GmbH
ID_VENDOR_ID: 1cf1
MAJOR: '166'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '1797972056257'
```

gregorth, 2024-04-22:
The same issuse started on 1.36.1 
I've moved back to 1.36 but the issue exists:
`zigbee2mqtt  | Zigbee2MQTT:debug 2024-04-22 16:10:54: Error: Command 0x84fd27fffee27a55/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 55120 - 1 - 13 - 6 - 11 after 10000ms)
zigbee2mqtt  |     at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
zigbee2mqtt  |     at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47)
zigbee2mqtt  |     at processTicksAndRejections (node:internal/process/task_queues:95:5)
zigbee2mqtt  |     at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
zigbee2mqtt  |     at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
zigbee2mqtt  |     at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:746:28)
zigbee2mqtt  |     at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:46:17)
zigbee2mqtt  |     at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:3242:28)
zigbee2mqtt  |     at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36)
zigbee2mqtt  |     at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)`

device

Zigbee 3.0 smart light switch module 2 gang (BW-SS7_2gang)
producent: BlitzWolf 

logs attached
[mqtt-e9c00b799d38e6cea56d6ca67013c4a7-0x84fd27fffee27a55-cfb76b68aa064a467d78c69ab1342f1f.json](https://github.com/Koenkk/zigbee2mqtt/files/15064291/mqtt-e9c00b799d38e6cea56d6ca67013c4a7-0x84fd27fffee27a55-cfb76b68aa064a467d78c69ab1342f1f.json)

DiegoJp, 2024-04-22:
Same behaviour with a ZB-GW04 v1.2 (latest fw: [v7.4.1.0](https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zb-gw04-1v2/ncp-uart-hw-v7.4.1.0-zb-gw04-1v2-115200.gbl=)) and some Tuya lightbulbs: TS0505B_1_1 and TS0505B_2_1

galant-ho, 2024-04-22:
Does anyone here have this error with SONOFF Zigbee dongle-P?

I am trying to understand if this is an isolated issue with ConBee2/RaspBee2.  If so then it's something else I need to resolve with my SONOFF Zigbee dongle-P.

paveltsytovich, 2024-04-23:
Have the same issue with TS0002 (model _TZ3000_fisb3ajo) 2 gang switch module

`Publish 'get' 'power_on_behavior' to 'Bathroom_Switch' failed: 'Error: ZCL command 0xa4c13873a4df1d0c/1 genOnOff.read(["moesStartUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 391 - 1 - 5 - 6 - 1 after 10000ms)'`

Actual version 1.36. Coordinator Sonoff dongle E

shanec42, 2024-04-23:
> Does anyone here have this error with SONOFF Zigbee dongle-P?
> 
> I am trying to understand if this is an isolated issue with ConBee2/RaspBee2. If so then it's something else I need to resolve with my SONOFF Zigbee dongle-P.

Yes.  I ran into the errors while running the "Sonoff Zigbee 3.0 USB Dongle Plus" (idVendor=10c4, idProduct=ea60) which I believe is identical.

remcotjeerdsma, 2024-04-23:
Migrated my Zigbee network on a Conbee 2 from Deconz to Zigbee2Mqtt this weekend. Now none of my lamps work anymore because of this. Seems to have started after an unintended shutdown of HassOS and with it Zigbee2Mqtt.

majkers, 2024-04-25:
I've just migrated from Conbee II to Sonoff dongle P and after abour 18 hours just got an error:

`error 2024-04-25 12:02:45: Publish 'set' 'state' to 'gniazdko_xiaomi' failed: 'Error: ZCL command 0x00158d0002347282/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 18793 - 1 - 158 - 6 - 11 after 10000ms)'`

You can see in my previous post https://github.com/Koenkk/zigbee2mqtt/issues/22189#issuecomment-2053533255 with Conbee that the error was a bit different but more or less the same. So it is not a matter of dongle I guess

vxav, 2024-04-26:
Same issue after downgrading to `1.36.0` and `1.35.3`. It seems to be somewhat transient as it works again after a moment.

```
Zigbee2MQTT:error 2024-04-26 15:15:14: Publish 'set' 'state' to 'Light Ceiling Kitchen' failed: 'Error: Command 0xa4c1387e52c34552/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 17076 - 1 - 26 - 6 - 11 after 10000ms)'

Zigbee2MQTT:info  2024-04-26 15:15:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Light Ceiling Kitchen' failed: 'Error: Command 0xa4c1387e52c34552/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 17076 - 1 - 26 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"Light Ceiling Kitchen"},"type":"zigbee_publish_error"}'
```

Matrix-aas, 2024-04-26:
Same problem.
There are 3 Aqara switches, and the “Sonoff zigbee 3.0 usb dongle plus-e” coordinator, after switching to the “MultiPAN RCP” firmware, Zigbee2MQTT began to behave incorrectly with the switches. After restarting the application, it works fine for a couple of minutes, after which errors begin and it becomes impossible to control the switches.
I tried rolling back the coordinator firmware back to ZESP, but nothing changed.
Zigbee2MQTT latest stable version.

_I don’t understand deeply, but I suspect that maybe the switches go to sleep and there is some possibility of waking them up at certain intervals? (re-query as at startup?)
Previously, this problem appeared occasionally, but was solved on its own; there were simply on/off delays of about 10 seconds. This didn't happen often._

https://www.zigbee2mqtt.io/devices/WS-EUK01.html#aqara-ws-euk01
https://www.zigbee2mqtt.io/devices/WS-EUK02.html#aqara-ws-euk02

```
[Error] [2024-04-26 19:45:16] Publish 'set' 'state' to 'Office Switcher' failed: 'Error: ZCL command 0x54ef44100022553f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47174 - 1 - 18 - 6 - 11 after 10000ms)'
[Error] [2024-04-26 19:45:23] Received undefined command from '0'
[Error] [2024-04-26 19:45:36] Publish 'set' 'state' to 'Office Switcher' failed: 'Error: ZCL command 0x54ef44100022553f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47174 - 1 - 19 - 6 - 11 after 10000ms)'
[Error] [2024-04-26 19:45:56] Publish 'get' 'state' to 'Office Switcher' failed: 'Error: ZCL command 0x54ef44100022553f/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47174 - 1 - 20 - 6 - 1 after 10000ms)'
```

delphijb, 2024-04-30:
**[solved for me]**
I had the same problem, with a Conbee II, HAOS (thinkclient) and Z2M 1.36.1
I solved my problem by upgrading the firmware of my Conbee II key to the latest version (0x26780700)

If this solution helps anyone :-)

galant-ho, 2024-05-01:
> Having similar issue with my Aqara E1 Roller Shade Driver starting last week. Thought it was my setting issue so repair it several times. The E1 paired successfully, able to control the blind for [...]

To give an update here... my issue seemed to be resolved by adding a good router near to my Aqara E1 blind.  Aqara appears to be non Zigbee 3.0 standard in which not so compatible with other brand router.
I've got a Aqara smart switch with neutral wire, installed it a few feet away from the E1 solved the problem.
Running pretty well for 3 days straight.

mhzawadi, 2024-05-01:
Thanks for your report, I have just got the update installed (https://community.home-assistant.io/t/how-to-update-conbee-ii-on-hassio-with-deconz-phoscon-integration/402826/12) helped a lot.

I did get an error on boot up

```
Error 2024-05-01 20:10:15Publish 'set' 'state' to '00 - light - hall' failed: 'Error: ZCL command 0x540f57fffe20798e/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (5))'
```

trozman, 2024-05-04:
Similar: suddenly my zigbee network is not working anymore (after 2 years of solid performance).
Errors: ... MAC channel access failure' (225)
... Error: ZCL command ...
(using Sonoff USB dongle plus)

XerxesFB, 2024-05-07:
Similar problem since update to ha 2024.5 and 5.2

Using donlge SONOFF USB Dongle plus E

Publish 'set' 'state' to 'Plug Levoit Luftreiniger' failed: 'Error: ZCL command 0xa4c138167ce8ca96/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13021 - 1 - 21 - 6 - 11 after 10000ms)'
 seems that only the few last added plugs have the problem

BurningTheSky, 2024-05-07:
Same issue for me. 
Sonoff P dongle
Zigbee2MQTT version: 1.37.0
HA Core 2024.5.2
HA Supervisor 2024.04.4 

```
Publish 'set' 'state' to 'Toilet and Under Stairs Switch' failed: 'Error: ZCL command 0x54ef44100076d47e/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37191 - 1 - 255 - 6 - 11 after 10000ms)'
```

NoAdO, 2024-05-08:
Same problem with LLKZMK11LM for about 4 weeks for now, sometimes work but sometimes i get "Timeout - 12367 - 1 - 198 - 6 - 11 after 10000ms" "Timeout - 12367 - 1 - 183 - 6 - 11 after 10000ms" etc. Already tried to restart, recreate container, re-pair device. I have 20 devices in the network and this one worked perfectly before, but have lags now. Other devices do not seem to be affected. 
Zigbee2MQTT (docker): 1.37.0
HA Core (docker) 2024.5.2
SONOFF ZB Dongle-P
Zigbee device model lumi.relay.c2acn01
```
[2024-05-08 00:20:46] error:  z2m: Publish 'set' 'state' to 'ZB Relay Bath' failed: 'Error: ZCL command 0x00158d000428ad57/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 12367 - 1 - 198 - 6 - 11 after 10000ms)'
```

majkers, 2024-05-08:
It would be nice if someone from developers write something about possible cause or fix...

trozman, 2024-05-08:
> Same issue for me. Sonoff P dongle Zigbee2MQTT version: 1.37.0 HA Core 2024.5.2 HA Supervisor 2024.04.4
>
> `Publish 'set' 'state' to 'Toilet and Under Stairs Switch' failed: 'Error: ZCL command [...]

Have you tried to update the Sonoff dongle firmware to 20230507? It solved the issue for me.

BurningTheSky, 2024-05-08:
> Have you tried to update the Sonoff dongle firmware to 20230507? It solved the issue for me.

I haven't, do I have to repair all devices after a firmware upgrade?

BurningTheSky, 2024-05-08:
> > > > Same issue for me. Sonoff P dongle Zigbee2MQTT version: 1.37.0 HA Core 2024.5.2 HA Supervisor 2024.04.4
> > > > `Publish 'set' 'state' to 'Toilet and Under Stairs Switch' failed: 'Error: ZCL command 0x54ef44100076d47e/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37191 - 1 - 255 - 6 - 11 after 10000ms)'`
> > > 
> > > 
> > > Have you tried to update the Sonoff dongle firmware to 20230507? It solved the issue for me.
> > 
> > 
> > I haven't, do I have to repair all devices after a firmware upgrade?
> 
> No, I didn't repair. I just switched routers (smart sockets) off - and then on (pulled them out of the wall sockets) after the FW update. This was necessary to get rid of errors.

I had to repair some light switches but not remove them from z2m so not too bad. All sorted and working for me now though, thanks!

kovu94, 2024-05-08:
Same issue here, started today. I have around 100 zigbee devices and the Home Assistant Skyconnect Stick.

Publish 'set' 'state' to 'EG Schlafzimmer Schalter Wand' failed: 'Error: ZCL command 0xa4c13899e3d176ef/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 11242 - 1 - 67 - 6 - 11 after 10000ms)'

It's a Moes Wall-Switch:

Zigbee-Modell
    TS0011
Zigbee-Hersteller
    _TZ3000_hhiodade
Beschreibung
    Wall light switch (1 gang)

Tried to remove and reset the switch. It's discovered/connected but I can't turn it on or off.
Checked for a firmware update for the Skyconnect stick, but on the homepage they say that no update is available.
Beside this - not important - switch, at the moment everything else works fine.

My versions (HA running on a Intel NUC):
Core 2024.5.2
Supervisor 2024.05.1
Operating System 12.3
Frontend 20240501.1

I just installed the Operating System Update to 12.3 with a clean reboot. Switch is still not working.

openshwprojects, 2024-05-09:
Same issue on SNZB-04
```
Info 2024-05-09 18:15:08Configuring '0x842e14fffe00c3bd'
Error 2024-05-09 18:15:30Failed to configure '0x842e14fffe00c3bd', attempt 1 (Error: ZCL command 0x842e14fffe00c3bd/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:412:23) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:766:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:401:29) at Object.configureMagicPacket (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:380:9) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:2096:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:160:21) at Configure.configure (/app/lib/extension/configure.ts:121:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:74:17))
```

asd2003ru, 2024-05-09:
Same error. Sometimes commands slip through with a delay.

j-paulus, 2024-05-10:
Started having this problem with one specific device: "TRADFRI control outlet. E1603/E1702/E1708" after the recent major update in z2m.
Coordinator is Sonoff Dongle-e with 7.4.2 [GA] FW, ezsp driver, problem happening with both z2m 1.37.0 and now 1.37.1.

Error message in log is:
```
[2024-05-10 08:27:16] error: 	z2m: Publish 'set' 'state' to 'Ikea Plug' failed: 'Error: ZCL command 0xb43a31fffe54afc0/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":100})'
```

Prior to the update the dongle was using the original 6-series FW and ember driver, and the network was solid.

kovu94, 2024-05-10:
Weird, today I received an update for Z2M

Zigbee2MQTT version
[1.37.1]
Coordinator type
EmberZNet
Coordinator revision
7.4.1 [GA]

After a deletion of the switch and a new pair, not it works fine again. 
Fingers crossed 😊

galant-ho, 2024-05-10:
Those who has been having this issue tried to upgrade Z2M to 1.37.1confimed issue resolved?

j-paulus, 2024-05-11:
@galant-ho As mentioned in my earlier message, I am having this problem also with 1.37.1. 

I tried re-pairing the problematic device, but the error keeps on appearing.

rw-django-fan-2020, 2024-05-11:
Finally I solved my problem. I had to change the config "channel" from 11 (default) to 15.

closed 2024-05-11 by rw-django-fan-2020

Tonio16, 2024-05-15:
> Those who has been having this issue tried to upgrade Z2M to 1.37.1confimed issue resolved?

I have the sonoff P dongle with zignee2mqtt 1.37.1 and I’have the issue. Zigbee2mqtt is running on a LXC. 
I also notice that ota are not working.

Antoine

AxxlFoley, 2024-05-17:
> Finally I solved my problem. I had to change the config "channel" from 11 (default) to 15.

I tried the same .. did not solve the issue for me

rw-django-fan-2020, 2024-05-17:
> > Finally I solved my problem. I had to change the config "channel" from 11 (default) to 15.
> 
> Good this worked for you ... But is this a stable fix for a longer period ? I tried the same .. did not solve the issue for me
> 
> I tried a lot :
> 
>     * update to 1.37.1
> 
>     * channel switch to 15
> 
>     * move to ember
> 
>     * update firmware of my Sonoff-dongle-e to latest 7.4.30
> 
> 
> Issue keeps apperaing .. so i do not think it is solved.

Did you repair your zigbee devices? Channel changing means repairing.
https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#permit-join

If you did repair and it does not work try the channels 20 or 25.

I guess I got these issue because I had to change my device and location of the zigbee adapter. It is now closer to my wlan router. And zigbee and wifi signals can cause interferences.

trozman, 2024-05-17:
> Good this worked for you ... But is this a stable fix for a longer period ? I tried the same .. did not solve the issue for me
>
> I tried a lot :
>
> * update to 1.37.1
> * channel switch to 15
> * move to [...]

After updating fw on the dongle, did you reset the routers?
This was a crucial step for me. After (only)updating fw (Sonoff P), it was worse. Then I unplugged and plugged back all the mains-powered Zigbee devices and suddenly all of the errors were gone.

shanec42, 2024-05-17:
I've now updated to 1.37.1, and everything appears (knock on wood) to have returned to normal.

galant-ho, 2024-05-18:
Few things I would like to bring this up hopefully could help some of you to resolve this issue.
I had such problem solely to my Aqara E1 blind motor around second week April.

Then I added a Aqara smart switch (with neutral) near to where my E1 installed, it resolved the problem for a while.  Yea just for a while and it gave out issue again.  This time I noticed E1 doesn't pair with the smart switch as router.  Completely no idea why.

I decided to move a IKEA smart outlet right near to where the smart switch was (same room), pair it to Z2M.  E1 connected to it with very strong LQI.  I also wondered battery of E1 dropped to 50% would also trigger this problem.  Hoped not the case.

NoAdO, 2024-05-20:
For me issue was solved when i disconnected NAS for a long period because of replacing UPS battery and forgot to connect 2 routers to power. No problem for about 2 weeks for now. I can guess the problem is related to self-calibration routes throung devices, so if we break the network (switch channel, stick update, no power for long time etc.) routes may create another way and problem will be "magically" solved. 

If you still have such troubles, try to power off single devices for a period of time and look for errors. If you found traitor, think what you can do with it. I have an old cc2531 stick so if the problem returns, i'm going to open another Z2M instance with traitor device and cc2531 and look will this fix my problem.

GWsmart, 2024-05-24:
Please HELP!!!
Similaire issue with conbee 2 on rpi4 8gb
flash new firmware on conbee2 without a result
Repair most of divices was good! Everything works, 
Than i made a new HA update and i have the same: Error: ZCL command! 
radio channel z2m: 11 wlan: 11
Before 05/2024 it works over one year!
What can i do to solve the problem

ArbiterGR, 2024-05-25:
I've been experiencing the same thing. Using ember driver and latest non-edge version of z2m.

GWsmart, 2024-05-25:
@ArbiterGR
witch logs do you mean?
`[2024-05-24 22:29:34] info: 	z2m: Zigbee: disabling joining new devices.
[2024-05-24 22:29:35] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-05-24 22:29:36] info: 	z2m: Connected to MQTT server
[2024-05-24 22:29:36] info: 	z2m: Started frontend on port 8099
[2024-05-24 22:29:36] info: 	z2m: Zigbee2MQTT started!
[2024-05-24 22:30:02] error: 	z2m: Publish 'set' 'state' to 'licht/eg/hwr' failed: 'Error: ZCL command 0x001788010d43dfd6/11 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":30}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (3))'
`
This is an example from z2m addon...
Or do you need other logs?

ArbiterGR, 2024-05-25:
> @ArbiterGR witch logs do you mean? `[2024-05-24 22:29:34] info: z2m: Zigbee: disabling joining new devices. [2024-05-24 22:29:35] info: z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883 [...]

I think you misunderstood me, all i'm saying is that i want to help the developers track the issue down and asking if they need any logs from me. Btw i'm getting the same ZCL command timeouts as you.

GCRVL, 2024-06-10:
This sadly is still an issue on my Home Assistant Yellow.

Patafoin, 2024-06-10:
Same here. I managed to make my device work by removing it with force deletion option on and adding it again. Worked some days and then the error appeared again.

[43 more replies]