Koenkk/zigbee2mqtt #22192: Upgraded buttons are unable to read the status correctly, model: J.XUAN WSZ01.

lhdbx2023, 2024-04-12:
### What happened?

After upgrading from version 1.36.0 to 1.36.1, the button functionality of J.XUAN WSZ01 is not working. Prior to the upgrade, the converter was functioning correctly. The upgraded version includes native support, but it is still not working. We suspect there might be an issue with the supported file characters.

### What did you expect to happen?

To resolve this issue, we can either fix the problem or discontinue support for this model. In the meantime, we can continue using the converter for implementation.

### How to reproduce it (minimal and precise)

After upgrading to version 1.36.1, I am unable to use it.

### Zigbee2MQTT version

1.36.1 commit: unknown

### Adapter firmware version

0.6.161

### Adapter

EZSP v8

### Setup

x86-64

### Debug log

Received Zigbee message from '0x00158d000129eff8', type 'attributeReport', cluster '65029', data '{"1":1}' from endpoint 1 with groupID 0

No converter available for 'WSZ01' with cluster '65029' and type 'attributeReport' and data '{"1":1}'

MQTT publish: topic 'zigbee2mqtt/bridge/response/device/generate_external_definition', payload '{"data":{"id":"0x00158d000129eff8","source":"const {identify, lock} = require('zigbee-herdsman-converters/lib/modernExtend');\n\nconst definition = {\n zigbeeModel: ['JD-SWITCH\u000002'],\n model: 'JD-SWITCH\u000002',\n vendor: 'Beesmart',\n description: 'Automatically generated definition',\n extend: [identify(), lock({\"pinCodeCount\":50})],\n meta: {},\n};\n\nmodule.exports = definition;"},"status":"ok","transaction":"jtszu-1"}'

[image]

Koenkk, 2024-04-13:
Could you provide the data/database.db entry of this device?

sxrcm, 2024-04-15:
> Could you provide the data/database.db entry of this device?

{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d000129bc8d","nwkAddr":62104,"manufId":4151,"manufName":"Beesmart","powerSource":"Battery","modelId":"JD-SWITCH\u000002","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":112,"inClusterList":[0,3,257,65024,65025],"outClusterList":[4,3,5,257,65024,65025,65535],"clusters":{"65024":{"attributes":{"3":43200,"4":"0x00158d000129bc8d","7":"B74EE1\u0000\u0000"}},"65025":{"attributes":{"8193":-51}},"65029":{"attributes":{"1":1}},"closuresDoorLock":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20181027","swBuildId":"02.00.01\u0000","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1713095063540}

lhdbx2023, 2024-04-15:
Okay, I'll give it to you right away

{"id":1,"type":"Coordinator","ieeeAddr":"0x187a3efffe393a98","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1712910466952}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d000129c088","nwkAddr":21283,"manufId":4611,"manufName":"Beesmart","powerSource":"Battery","modelId":"door sensor","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,65024,65025,4111],"outClusterList":[4,3,1,1280,65024,65025,4111],"clusters":{"65025":{"attributes":{"8193":-52}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":78}},"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0x187a3efffe393a98","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"hwVersion":20,"dateCode":"20181020","swBuildId":"02.01.00","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1713183774505}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d000129eff8","nwkAddr":52356,"manufId":4151,"manufName":"Beesmart","powerSource":"Battery","modelId":"JD-SWITCH\u000002","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":112,"inClusterList":[0,3,257,65024,65025],"outClusterList":[4,3,5,257,65024,65025,65535],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20181027","swBuildId":"02.00.01\u0000","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1713185728744}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d000129d00e","nwkAddr":27237,"manufId":4611,"manufName":"Beesmart","powerSource":"Battery","modelId":"wall pir","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1280,"inClusterList":[0,3,1,1280,65024,65025,4111],"outClusterList":[4,3,1,1280,65024,65025,4111],"clusters":{"65024":{"attributes":{"3":43205,"4":"0x00158d000129d00e","5":"02.03.01","7":"8307F6\u0000\u0000"}},"65025":{"attributes":{"8193":-16}},"genBasic":{"attributes":{"manufacturerName":"Beesmart"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x187a3efffe393a98","zoneState":1}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":190}}},"binds":[],"configuredReportings":[],"meta":{}}},"hwVersion":20,"dateCode":"20190325","swBuildId":"02.03.01","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1713184343635}
{"id":5,"type":"Router","ieeeAddr":"0x00158d0001273efe","nwkAddr":24763,"manufId":4151,"manufName":"BEESMART","powerSource":"Mains (single phase)","modelId":"00090bdc","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":83,"inClusterList":[0,4,3,6,8,5,65024,65025,4111],"outClusterList":[25,65024,65025,4111],"clusters":{"65024":{"attributes":{"3":300,"4":"0x00158d0001273efe","7":"49784C\u0000�\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}},"genBasic":{"attributes":{"32768":0,"swBuildId":"","zclVersion":1}},"genOnOff":{"attributes":{"8192":0,"onOff":1}},"haElectricalMeasurement":{"attributes":{"activePower":1}},"seMetering":{"attributes":{"currentSummDelivered":[0,357],"unitOfMeasure":0,"multiplier":1,"divisor":100,"summaFormatting":242}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"2017-04-28","swBuildId":"","zclVersion":1,"interviewCompleted":true,"meta":{"configured":-2009206100},"lastSeen":1713185434115}
{"id":6,"type":"Router","ieeeAddr":"0x70b3d52b6002a371","nwkAddr":43914,"manufId":4660,"manufName":"_TZ3000_hyarhbyx","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,6,3,4,5,57345],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TZ3000_hyarhbyx","powerSource":1,"zclVersion":2,"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZTDQ31YX105"}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"moesStartUpOnOff":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":4,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":57345,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion": [... cut ...] ntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":768,"attrId":7,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":71,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1713185636949}

lhdbx2023, 2024-04-15:
{"id":1,"type":"Coordinator","ieeeAddr":"0x187a3efffe393a98","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1712910466952}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d000129c088","nwkAddr":21283,"manufId":4611,"manufName":"Beesmart","powerSource":"Battery","modelId":"door sensor","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,65024,65025,4111],"outClusterList":[4,3,1,1280,65024,65025,4111],"clusters":{"65025":{"attributes":{"8193":-52}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":78}},"genBasic":{"attributes":{}},"ssIasZone":{"attributes":{"iasCieAddr":"0x187a3efffe393a98","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"hwVersion":20,"dateCode":"20181020","swBuildId":"02.01.00","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1713183774505}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d000129eff8","nwkAddr":52356,"manufId":4151,"manufName":"Beesmart","powerSource":"Battery","modelId":"JD-SWITCH\u000002","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":112,"inClusterList":[0,3,257,65024,65025],"outClusterList":[4,3,5,257,65024,65025,65535],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20181027","swBuildId":"02.00.01\u0000","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1713185728744}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d000129d00e","nwkAddr":27237,"manufId":4611,"manufName":"Beesmart","powerSource":"Battery","modelId":"wall pir","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1280,"inClusterList":[0,3,1,1280,65024,65025,4111],"outClusterList":[4,3,1,1280,65024,65025,4111],"clusters":{"65024":{"attributes":{"3":43205,"4":"0x00158d000129d00e","5":"02.03.01","7":"8307F6\u0000\u0000"}},"65025":{"attributes":{"8193":-16}},"genBasic":{"attributes":{"manufacturerName":"Beesmart"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x187a3efffe393a98","zoneState":1}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":190}}},"binds":[],"configuredReportings":[],"meta":{}}},"hwVersion":20,"dateCode":"20190325","swBuildId":"02.03.01","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1713184343635}
{"id":5,"type":"Router","ieeeAddr":"0x00158d0001273efe","nwkAddr":24763,"manufId":4151,"manufName":"BEESMART","powerSource":"Mains (single phase)","modelId":"00090bdc","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":83,"inClusterList":[0,4,3,6,8,5,65024,65025,4111],"outClusterList":[25,65024,65025,4111],"clusters":{"65024":{"attributes":{"3":300,"4":"0x00158d0001273efe","7":"49784C\u0000�\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}},"genBasic":{"attributes":{"32768":0,"swBuildId":"","zclVersion":1}},"genOnOff":{"attributes":{"8192":0,"onOff":1}},"haElectricalMeasurement":{"attributes":{"activePower":1}},"seMetering":{"attributes":{"currentSummDelivered":[0,357],"unitOfMeasure":0,"multiplier":1,"divisor":100,"summaFormatting":242}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"2017-04-28","swBuildId":"","zclVersion":1,"interviewCompleted":true,"meta":{"configured":-2009206100},"lastSeen":1713185434115}
{"id":6,"type":"Router","ieeeAddr":"0x70b3d52b6002a371","nwkAddr":43914,"manufId":4660,"manufName":"_TZ3000_hyarhbyx","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,6,3,4,5,57345],"outClusterList":[10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TZ3000_hyarhbyx","powerSource":1,"zclVersion":2,"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZTDQ31YX105"}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"moesStartUpOnOff":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":4,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":57345,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x187a3efffe393a98","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"da [... cut ...] ntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":768,"attrId":7,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":71,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1713185636949}

Koenkk, 2024-04-15:
The issue is with the generated external converter, the correct converter is this one: https://github.com/Koenkk/zigbee-herdsman-converters/blob/a5e7cdfec08f7d2bc595d6dfd83dab9b1e54b5f0/src/devices/jxuan.ts#L33

closed 2024-04-16 by lhdbx2023