Koenkk/zigbee2mqtt #22193: Aqara Light Sensor T1 Detection Period Fails to Set

chris-github1975, 2024-04-12:
### What happened?

I am using the web interface for Zigbee2MQTT within HomeAssistant.
I choose my device and go to the "Exposes Tab"
There is a slider bar for changing the detection period, the default is 5, I change the value to anything other than 5 and I receive an error.

### What did you expect to happen?

I expected the value to change to one of my choosing.

### How to reproduce it (minimal and precise)

I am using the web interface for Zigbee2MQTT within HomeAssistant.
I choose my device and go to the "Exposes Tab"
There is a slider bar for changing the detection period, the default is 5, I change the value to anything other than 5 and I receive an error.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff EZSP v8 6.10.3.0 build 297 Firmware 1.36.1

### Setup

Home Assistant Add-On

### Debug log

Error:
Publish 'set' 'detection_period' to 'T1-01Z' failed: 'Error: ZCL command 0x54ef441000a5fee9/1 manuSpecificLumi.write({"0":{"value":[6],"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44426 - 1 - 128 - 64704 - 4 after 10000ms)'

Debug:
2024-04-12 09:33:27Error: ZCL command 0x54ef441000a5fee9/1 manuSpecificLumi.write({"0":{"value":[8],"type":33}}, 
{"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44426 - 1 - 21 - 64704 - 4 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at EZSPAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:492:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:760:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:362:9) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/lumi.ts:3791:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:259:36) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

mrskycriper, 2024-04-12:
Have you tried waking it up? It is as sleepy device, so it will not listen to commands 99% of the time.

chris-github1975, 2024-04-12:
> Have you tried waking it up? It is as sleepy device, so it will not listen to commands 99% of the time.

Hi, Yes it is definitely awake, the light sensor is reading lux values as the light levels change,

mrskycriper, 2024-04-12:
But have you tried waking it up manually, using the pairing button. Sometimes they will not respond to commands even while they are sending data.

chris-github1975, 2024-04-12:
> But have you tried waking it up manually, using the pairing button? Sometimes they will not respond to commands even while they are sending data.

Well... I am amazed, everyday is a school day !
I never even knew about pressing the pairing button to wake it. I did need to press a few times but it eventually worked,
Thank you so much for you help !

closed 2024-04-12 by chris-github1975