Koenkk/zigbee2mqtt #22195: ModernExtend levelConfig when using multiple Endpoints

Nailik, 2024-04-12:
### What happened?

Using modernExtend with a setup like:

```
deviceEndpoints({endpoints: {'left': 4, 'right': 3}}),
light({endpointNames: ['left', 'right'], configureReporting: true, levelConfig: {}}),
```

When changing the levelConfig (onLevel) in the UI i see 
`Received MQTT message on 'zigbee2mqtt/licht_wohnzimmer/set' with data '{"level_config_left":{"on_level_left":116}}'`
After requesting update from device:
`'{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":50,"ballast_maximum_level_right":254,"ballast_minimum_level":1,"ballast_minimum_level_right":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":254,"brightness_left":150,"brightness_right":100,"dimmer_mode":"auto","indicator_mode_left_btn":"always_on","indicator_mode_right_btn":"always_off","level_config":{"current_level_startup":"minimum","execute_if_off":false,"on_level":150},"linkquality":188,"power_on_behavior":null,"state":"ON","state_left":"ON","state_right":"ON","switch_actions_left_btn":"dimmer_opposite","switch_actions_right_btn":"dimmer_opposite"}'`

You can see that the `on_level` did not change correctly but it works by using dev console (endpoint 3 or 4, .LevelCtrl cluster and onLevel attribute) there also read and write works. 

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

7.2.2.0 build 190

### Adapter

 EZSP v11

### Setup

HomeAssistant

mrskycriper, 2024-04-12:
This should be fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7374

Nailik, 2024-04-12:
I tried on this commit https://github.com/Koenkk/zigbee2mqtt/commit/e10ab5c

And it doesn't work. What also looks weird in the logs is that there is only one level_config field but there should be 2 (left and right dimmer)

```
[2024-04-12 17:52:03] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/licht_wohnzimmer', payload '{"ballast_lamp_alarm_lamp_burn_hours":false,"ballast_maximum_level":50,"ballast_maximum_level_right":254,"ballast_minimum_level":1,"ballast_minimum_level_right":1,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":254,"brightness_left":254,"brightness_right":254,"dimmer_mode":"auto","indicator_mode_left_btn":"always_on","indicator_mode_right_btn":"always_off","level_config":{"current_level_startup":"minimum","execute_if_off":false,"on_level":4},"linkquality":216,"power_on_behavior":null,"state":"ON","state_left":"OFF","state_right":"ON","switch_actions_left_btn":"dimmer_opposite","switch_actions_right_btn":"dimmer_opposite"}'
[2024-04-12 17:52:03] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/licht_wohnzimmer/left', payload '{"brightness":254,"indicator_mode":"always_on","state":"OFF","switch_actions":"dimmer_opposite"}'
[2024-04-12 17:52:03] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/licht_wohnzimmer/right', payload '{"ballast_maximum_level":254,"ballast_minimum_level":1,"brightness":254,"indicator_mode":"always_off","state":"ON","switch_actions":"dimmer_opposite"}'
```

Nailik, 2024-04-12:
Btw device definition:

```
zigbeeModel: ['2GANG/DIMMER/2'],
    model: 'MEG5126-0300/MEG5172-0000',
    vendor: 'Schneider Electric',
    description: 'Merten MEG5172 PlusLink Dimmer insert with Merten Wiser System M Push Button (2fold)',
    fromZigbee: [fz.wiser_lighting_ballast_configuration],
    toZigbee: [tz.ballast_config, tz.wiser_dimmer_mode],
    exposes: [
        e.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        e.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        e.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)'),
    ],
    extend: [
        deviceEndpoints({endpoints: {'left': 4, 'right': 3, 'left_btn': 22, 'right_btn': 21}}),
        light({endpointNames: ['left', 'right'], configureReporting: true, levelConfig: {}}),
        switchActions('left_btn'), switchActions('right_btn'), indicatorMode('left_btn'),
    ],
```

mrskycriper, 2024-04-12:
> And it doesn't work

What part isn't working? Only one level config? Both? Dimming too? I'm a bit confused here

Nailik, 2024-04-12:
Only the levelConfig i want to change "onLevel" attribute withing "LevelControl" 
It works in Dev console
[image]
but it doesn't have an effect in ui
[image]

Nailik, 2024-04-12:
I did just notice that when i use old extend there is a save button next to the LevelConfig and then it works, could this be the issue?

[image]

Nailik, 2024-04-14:
@Koenkk could you maybe have a look why the save changes button is missing?

Nailik, 2024-04-14:
I think the issue is not ModernExtend but the Save button when using mutliple endpoints.

One endpoint with legacy extend, button is visible:

```
const definition = {
    zigbeeModel: ['2GANG/DIMMER/2'],
    model: 'MEG5126-0300/MEG5172-0000',
    vendor: 'Schneider Electric',
    description: 'Merten MEG5172 PlusLink Dimmer insert with Merten Wiser System M Push Button (2fold)',
    fromZigbee: [fz.wiser_lighting_ballast_configuration, fz.level_config],
    toZigbee: [tz.ballast_config, tz.wiser_dimmer_mode, tz.level_config],
    exposes: [
        e.light_brightness().withLevelConfig().withEndpoint(),
        e.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        e.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        e.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)'),
    ],
    extend: [
        deviceEndpoints({endpoints: {'left': 4, 'right': 3, 'left_btn': 22, 'right_btn': 21}}),
        light({endpointNames: ['left', 'right'], configureReporting: true}),
        switchActions('left_btn'), switchActions('right_btn'), indicatorMode('left_btn'),
    ],
};
```

[image]

Multiple endpoint (with legacy extend using .withEndpoint), save button is missing

```
const definition = {
    zigbeeModel: ['2GANG/DIMMER/2'],
    model: 'MEG5126-0300/MEG5172-0000',
    vendor: 'Schneider Electric',
    description: 'Merten MEG5172 PlusLink Dimmer insert with Merten Wiser System M Push Button (2fold)',
    fromZigbee: [fz.wiser_lighting_ballast_configuration, fz.level_config],
    toZigbee: [tz.ballast_config, tz.wiser_dimmer_mode, tz.level_config],
    exposes: [
        e.light_brightness().withLevelConfig().withEndpoint('left'),
        e.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        e.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        e.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)'),
    ],
    extend: [
        deviceEndpoints({endpoints: {'left': 4, 'right': 3, 'left_btn': 22, 'right_btn': 21}}),
        light({endpointNames: ['left', 'right'], configureReporting: true}),
        switchActions('left_btn'), switchActions('right_btn'), indicatorMode('left_btn'),
    ],
};
```

[image]

Multiple endpoint with modern extend (save button is missing)

```
const definition = {
    zigbeeModel: ['2GANG/DIMMER/2'],
    model: 'MEG5126-0300/MEG5172-0000',
    vendor: 'Schneider Electric',
    description: 'Merten MEG5172 PlusLink Dimmer insert with Merten Wiser System M Push Button (2fold)',
    fromZigbee: [fz.wiser_lighting_ballast_configuration, fz.level_config],
    toZigbee: [tz.ballast_config, tz.wiser_dimmer_mode, tz.level_config],
    exposes: [
        e.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        e.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        e.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)'),
    ],
    extend: [
        deviceEndpoints({endpoints: {'left': 4, 'right': 3, 'left_btn': 22, 'right_btn': 21}}),
        light({endpointNames: ['left'], configureReporting: true, levelConfig: {}}),
        switchActions('left_btn'), switchActions('right_btn'), indicatorMode('left_btn'),
    ],
};
```

[image]

Koenkk, 2024-04-15:
> One endpoint with legacy extend, button is visible:

Is one endpoint + modern extend working?

Nailik, 2024-04-16:
no it seems like the indentation of level config might be the issue

one endpoint (remove other stuff)

```
const definition = {
    zigbeeModel: ['2GANG/DIMMER/2'],
    model: 'MEG5126-0300/MEG5172-0000',
    vendor: 'Schneider Electric',
    description: 'Merten MEG5172 PlusLink Dimmer insert with Merten Wiser System M Push Button (2fold)',
    fromZigbee: [fz.wiser_lighting_ballast_configuration, fz.level_config],
    toZigbee: [tz.ballast_config, tz.wiser_dimmer_mode, tz.level_config],
    exposes: [
        e.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        e.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        e.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)'),
    ],
    extend: [
        deviceEndpoints({endpoints: {'left': 4}}),
        light({endpointNames: ['left'], configureReporting: true, levelConfig: {}}),
    ],
};
```

[image]

closed 2024-11-13 by github-actions[bot]