Koenkk/zigbee2mqtt #22197: TS0502B Light controller turns off after turned on : bad brightness value

Clem-, 2024-04-12:
### What happened?

My led strip is very often turning it off itself just after I turn it on.
I suspected the power supply with poor quality but a new one show same behavior.

Looking at the logs, i noticed that the TS0502B light controller sometimes report a **brightness of 1000**.
Then, after the light is turned on again, the out of range brightness raise an error (see logs attached).

### What did you expect to happen?

The light remains turned on.
The brightness value remains under 255.

### How to reproduce it (minimal and precise)

Publish a MQTT message to turn on TS0502B led controller.
If it reports a brightness of 1000, the next time it is turned on might raise an error, making it turned on immediatly.

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20221226

### Adapter

Slaesh's

### Setup

Docker container

### Debug log

Light is turned on, `brightness:254` published, device reports `genLevelCtrl` 1000, then `brightness:1000` published
```
debug 2024-04-12 17:04:46Received MQTT message on 'zigbee2mqtt/0xa4c138d42e947426/set' with data '{"state":"ON"}'
debug 2024-04-12 17:04:46Publishing 'set' 'state' to '0xa4c138d42e947426'
info 2024-04-12 17:04:46MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":254,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":51,"state":"ON"}'
debug 2024-04-12 17:04:46Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2024-04-12 17:04:46Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2024-04-12 17:04:46MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":254,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":76,"state":"ON"}'
info 2024-04-12 17:04:46MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":254,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":76,"state":"ON"}'
debug 2024-04-12 17:04:46Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
info 2024-04-12 17:04:46MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":254,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":72,"state":"ON"}'
debug 2024-04-12 17:04:46Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"61440":0}' from endpoint 1 with groupID 0
debug 2024-04-12 17:04:47Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'lightingColorCtrl', data '{"tuyaRgbMode":0}' from endpoint 1 with groupID 0
info 2024-04-12 17:04:47MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":254,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":69,"state":"ON"}'
debug 2024-04-12 17:04:47Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":1000}' from endpoint 1 with groupID 0
info 2024-04-12 17:04:47MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":1000,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":72,"state":"ON"}'
debug 2024-04-12 17:04:47Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'lightingColorCtrl', data '{"57600":[1000,0]}' from endpoint 1 with groupID 0
info 2024-04-12 17:04:47MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":1000,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":72,"state":"ON"}'
```

Light is turned off, brightness is set to 10, then brightness is set to 1.
```
debug 2024-04-12 17:07:31Received MQTT message on 'zigbee2mqtt/0xa4c138d42e947426/set' with data '{"state":"OFF"}'
debug 2024-04-12 17:07:31Publishing 'set' 'state' to '0xa4c138d42e947426'
info 2024-04-12 17:07:31MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":1000,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":72,"state":"OFF"}'
debug 2024-04-12 17:07:31Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'lightingColorCtrl', data '{"tuyaRgbMode":0}' from endpoint 1 with groupID 0
info 2024-04-12 17:07:31MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":1000,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":69,"state":"OFF"}'
debug 2024-04-12 17:07:31Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":10}' from endpoint 1 with groupID 0
```

Light is turned on, and turns off immediatly (because brightness out of range ?)
```
debug 2024-04-12 17:11:57Received MQTT message on 'zigbee2mqtt/0xa4c138d42e947426/set' with data '{"state":"ON"}'
debug 2024-04-12 17:11:57Publishing 'set' 'state' to '0xa4c138d42e947426'
error 2024-04-12 17:11:57Publish 'set' 'state' to '0xa4c138d42e947426' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":1000,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 1000)'
debug 2024-04-12 17:11:57RangeError: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":1000,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received 1000) at new NodeError (node:internal/errors:405:5) at writeU_Int8 (node:internal/buffer:740:11) at Buffer.writeUInt8 (node:internal/buffer:750:10) at BuffaloZcl.writeUInt8 (/app/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:41:21) at BuffaloZcl.write (/app/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:249:18) at BuffaloZcl.write (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:645:26) at ZclFrame.writePayloadCluster (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:153:21) at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:72:18) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:389:22) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:357:25
info 2024-04-12 17:11:57MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xa4c138d42e947426' failed: 'RangeError [ERR_OUT_OF_RANGE]: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":1000,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (The value of \"value\" is out of range. It must be >= 0 and <= 255. Received 1000)'","meta":{"friendly_name":"0xa4c138d42e947426"},"type":"zigbee_publish_error"}'
```

Koenkk, 2024-04-14:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/22078

closed 2024-04-14 by Koenkk

Clem-, 2024-04-16:
Hi,
Unfortunately, my main issue (light switching off just after being turned on), isn't solved by the latest fixt mentioned in #22078.
I can't explain what is causing this.

Logs after light being turned on:
``` 
zigbee2mqtt  | [2024-04-16 12:52:28] debug:     z2m: Received MQTT message on 'zigbee2mqtt/0xa4c138d42e947426/set' with data '{"state":"ON"}'
zigbee2mqtt  | [2024-04-16 12:52:28] debug:     z2m: Publishing 'set' 'state' to '0xa4c138d42e947426'
zigbee2mqtt  | [2024-04-16 12:52:28] debug:     zh:controller:endpoint: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":26,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt  | [2024-04-16 12:52:28] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,1)
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":26,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":65,"state":"ON"}'
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":26,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":65,"state":"ON"}'
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":26,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":83,"state":"ON"}'
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":26,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":83,"state":"OFF"}'
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'lightingColorCtrl', data '{"tuyaRgbMode":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":26,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":83,"state":"OFF"}'
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":102}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":260,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":87,"state":"OFF"}'
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":26}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":26,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":87,"state":"OFF"}'
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'lightingColorCtrl', data '{"57600":[102,0]}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":26,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":87,"state":"OFF"}'
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     zh:controller:endpoint: ZCL command 0xa4c138d42e947426/1 genOnOff.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":122,"writeUndiv":false})
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,2)
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":26,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":303,"do_not_disturb":true,"linkquality":83,"state":"OFF"}'
[... cut ...] leRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":123,"writeUndiv":false})
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,1)
zigbee2mqtt  | [2024-04-16 12:52:29] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"61440":0}' from endpoint 1 with groupID 0
```

Koenkk, 2024-04-16:
What's the last working z2m version?

Clem-, 2024-04-16:
The turning off issue was present since the beginning. 
it is only recently that I tried to investigate what was going wrong, after replacing my power supply that I first suspected.

Clem-, 2024-04-19:
After updating to `latest-dev` with the fix for ERR_OUT_OF_RANGE, the light keeps turning off after trying to turned on (only sometimes, like always). No more brightness value out of range though. 

Here's an example after turning the light on with Z2M frontend :
```
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: Received MQTT message on 'zigbee2mqtt/0xa4c138d42e947426/set' with data '{"state":"ON"}'
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: Publishing 'set' 'state' to '0xa4c138d42e947426'
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     zh:controller:endpoint: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":255,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,1)
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":255,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"linkquality":76,"state":"ON"}'
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":255,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"linkquality":80,"state":"ON"}'
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":255,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"linkquality":80,"state":"ON"}'
```
I am not sure why there is that much duplicate publish messages, but at a certain time, the device reports `"state":"OFF"`:
```
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"61440":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'lightingColorCtrl', data '{"tuyaRgbMode":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":255,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"linkquality":80,"state":"ON"}'
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":255,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"linkquality":83,"state":"OFF"}'
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genLevelCtrl', data '{"61440":1000}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-19 17:39:56] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":255,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"linkquality":58,"state":"OFF"}'
zigbee2mqtt  | [2024-04-19 17:39:57] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":255}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-19 17:39:57] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":255,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"linkquality":80,"state":"OFF"}'
zigbee2mqtt  | [2024-04-19 17:39:57] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'lightingColorCtrl', data '{"57600":[1000,0]}' from endpoint 1 with groupID 0
zigbee2mqtt  | [2024-04-19 17:39:57] debug:     z2m: MQTT publish: topic 'zigbee2mqtt/0xa4c138d42e947426', payload '{"brightness":255,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"linkquality":80,"state":"OFF"}'
zigbee2mqtt  | [2024-04-19 17:39:57] debug:     zh:controller:endpoint: ZCL command 0xa4c138d42e947426/1 genOnOff.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":123,"writeUndiv":false})
zigbee2mqtt  | [2024-04-19 17:39:57] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,1)
[... cut ...] onSequenceNumber":127,"writeUndiv":false})
zigbee2mqtt  | [2024-04-19 17:40:26] debug:     zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,1)
zigbee2mqtt  | [2024-04-19 17:40:26] debug:     z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genBasic', data '{"65506":53,"65508":0,"appVersion":101}' from endpoint 1 with groupID 0
```

@Koenkk if there is anything I can do to help investigating this issue, please let me know :) Thanks !

Koenkk, 2024-04-21:
Do you have a `transition` set for this device in your `configuration.yaml`?

Clem-, 2024-04-21:
No transition set for this light in configuration.yaml nor light_profiles.csv

Koenkk, 2024-04-22:
Does it also happen when turning on via the dev console (click on the device in the frontend) and sending this?

[image]

Clem-, 2024-04-22:
It is working good with this command. No sudden turning off. 
Even by sending multiple time Command 2 (for toggle I guess), it behaves fine. 

Could it be due to the transition value?
Is there a way to fully disable transition for this device?

Clem-, 2024-04-23:
It is quite challenging to limit the number of lines due to a large network of 60 devices.
Even after disabling HA integration to limit logging I still have literally thousands of lines just after startup.
Not to mention that the bug is intermittent and not easy catch.

So here's the part where the bug happens with full logging level:
```
[2024-04-23 23:59:19] debug: 	z2m: Received MQTT message on 'zigbee2mqtt_debug/0xa4c138d42e947426/set' with data '{"state":"ON","brightness":251}'
[2024-04-23 23:59:19] debug: 	z2m: Publishing 'set' 'brightness' to '0xa4c138d42e947426'
[2024-04-23 23:59:19] debug: 	zh:controller:endpoint: ZCL command 0xa4c138d42e947426/1 genLevelCtrl.moveToLevelWithOnOff({"level":251,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-04-23 23:59:19] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c138d42e947426:1461/1 (0,0,3)
[2024-04-23 23:59:19] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":1461,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":139,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,58,4,251,0,0]}}
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:writer: --> frame [254,16,36,1,181,5,1,1,8,0,139,0,30,6,1,58,4,251,0,0,218]
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-04-23 23:59:19] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,139,77]
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,139,77]
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,139] - 77
[2024-04-23 23:59:19] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":139}
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,8,0,181,5,1,1,0,61,0,43,167,81,0,0,5,24,58,11,4,0,181,5,29,1]
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,8,0,181,5,1,1,0,61,0,43,167,81,0,0,5,24,58,11,4,0,181,5,29,1]
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,8,0,181,5,1,1,0,61,0,43,167,81,0,0,5,24,58,11,4,0,181,5,29] - 1
[2024-04-23 23:59:19] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":8,"srcaddr":1461,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":61,"securityuse":0,"timestamp":5351211,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,58,11,4,0]}}
[2024-04-23 23:59:19] debug: 	zh:controller: Received payload: clusterID=8, address=1461, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=61, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":58,"commandIdentifier":11},"payload":{"cmdId":4,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-04-23 23:59:19] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt_debug/0xa4c138d42e947426', payload '{"brightness":251,"color":{"h":0,"hue":0},"color_mode":"color_temp","color_temp":250,"do_not_disturb":true,"linkquality":61,"state":"ON"}'
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,6,0,181,5,1,1,0,58,0,214,170,81,0,0,7,24,26,10,0,0,16,1,181,5,29,204]
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,6,0,181,5,1,1,0,58,0,214,170,81,0,0,7,24,26,10,0,0,16,1,181,5,29,204]
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,181,5,1,1,0,58,0,214,170,81,0,0,7,24,26,10,0,0,16,1,181,5,29] - 204
[2024-04-23 23:59:19] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":1461,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":58,"securityuse":0,"timestamp":5352150,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,26,10,0,0,16,1]}}
[2024-04-23 23:59:19] debug: 	zh:controller: Received payload: clusterID=6, address=1461, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=58, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":26,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":16,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-04-23 23:59:19] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-04-23 23:59:19] debug: 	z2m: Received Zigbee message from '0xa4c138d42e947426', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
[... cut ...]
[2024-04-23 23:59:21] debug: 	zh:zstack:unpi:parser: <-- [254,33,68,129,0,0,0,239,58,229,1,1,0,47,0,209,58,83,0,0,13,9,33,2,0,247,18,2,0,4,0,0,7,58,226,188,28,248]
[2024-04-23 23:59:21] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,58,229,1,1,0,47,0,209,58,83,0,0,13,9,33,2,0,247,18,2,0,4,0,0,7,58,226,188,28,248]
[2024-04-23 23:59:21] debug: 	zh:zstack:unpi:parser: --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,58,229,1,1,0,47,0,209,58,83,0,0,13,9,33,2,0,247,18,2,0,4,0,0,7,58,226,188,28] - 248
```

Of course if you still need more info, like the full I will be pleased to provide it.
Thank you again Koen for your kind support!

Clem-, 2024-05-03:
Hi @Koenkk,
Sorry for the delay. I finally managed to get you a short log file after having the idea to build an automation to reproduce the bug after z2m startup.

In this scenario, I turned on the light to brightness 2, turned it off (remember that it remains actually on), and turned it on brightness 255: [z2m_on2_off_on255.txt](https://github.com/Koenkk/zigbee2mqtt/files/15200863/z2m_on2_off_on255.txt)

For an even shorter log, I set the light in "not really turned off" state (first 2 steps), and only after restart z2m and turn the light on brightness 1: [z2m_on1.txt](https://github.com/Koenkk/zigbee2mqtt/files/15200920/z2m_on1.txt)

No that I can easily produce those logs, feel free to ask for additional testing.
Thank you :)

Koenkk, 2024-05-05:
Thanks, could you also provide your complete `configuration.yaml`?

Clem-, 2024-05-05:
@Koenkk here you go: 
[configuration.yaml](https://github.com/Koenkk/zigbee2mqtt/files/15214439/configuration.txt)

Koenkk, 2024-05-05:
Can you remove `transition: 0` from `0xa4c138d42e947426` and see if it fixed?

Clem-, 2024-05-05:
Well.. it looks like that was it!
My script doesn’t reproduce the bug anymore. Need to test it further but looking pretty good. 

I don’t understand though, I set `transition: 0` in order to disable transition, but that is not doing what I expected right? A value of 0 is not preventing transition?

Koenkk, 2024-05-06:
No, then it will send transition 0 instead of the regular on/off commands.

Clem-, 2024-05-07:
I mean how can one bypass the default transition and just turn the light on/off instantly?

Koenkk, 2024-05-07:
Maybe I don't understand your question, but by not specifying a `transition`.

Clem-, 2024-05-07:
Well, even if I don't set a transition in the service call, there's always a smooth transition when I turn on and off a light.

Koenkk, 2024-05-08:
If that's the case even with a on/off command, I don't think you can change this (firmware behaviour)

Clem-, 2024-05-08:
The behavior changes if I set the Transition to 0 in the specific parameters in the frontend (screenshot above). 
That is why I expected the `transition: 0 ` in the payload to have a similar effect. 

But thanks to your explanations I now understand that they are not the same parameters (one is written in the device memory unlike the other is just an argument of the payload). 

Thank you very much for your time!