Koenkk/zigbee2mqtt #22201: high trafic between zigbee2mqtt and MQTT Server (fhem)

matze20211, 2024-04-13:
### What happened?

i have a vpn network with a remote zigbee2mqtt server and a local mqtt server. i guess before version 1.35 there was about 500 MB traffic a day between those servers. Now with version 1.36.1 there is 1 GB traffic with the same amount of devices (31).  

### What did you expect to happen?

I expected the same traffic

### How to reproduce it (minimal and precise)

i would downgrade, if i knew how

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20230507

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Pi

Koenkk, 2024-05-05:
This is because of the logging changes with [1.37.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.37.0) 😄

closed 2024-05-05 by Koenkk

matze20211, 2024-05-16:
with 1.37.1 traffic high again?