Koenkk/zigbee2mqtt #22204: ZNCZ04LM how to downgrade firmware to OTA version? Older OTA firmware no problems, newer firmware problems

hspjanssen, 2024-04-13:
### What happened?

I have several ZNCZ04LM units, most of them have firmware build date 09-06-2019, they work very good. Some of them have firmware build date 01-23-2022, those units do not show measurements like Voltage, Power etc. I do want to downgrade those with the newer build date to the older build date. 

Newer firmware then 01-23-2022 is not available via OTA, when I click on "check for new update" on an item with the newer firmware update (01-23-2022), I receive the message "OTA: Firmware on '0x04cf8cdf3c759cac' (lumi.plug.mmeu01) is newer than latest firmware online". When I click on "check for new update" on an item with the older (09-06-2019) firmware update I receive the message "No update available for '0x0x04cf8cdf3c759bad". So I assume that the version in the OTA database is 09-06-2019, how can I downgrade the units with 01-23-2022 to 09-06-2019?

### What did you expect to happen?

See above

### How to reproduce it (minimal and precise)

See above

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20210120

### Adapter

CC2652RB 

### Setup

Virtual Debian installation on Proxmox

HanspeterSaeure, 2024-07-07:
I got the same issue with mine. I tried the downgrade guides with the edited FW on my HA server ... doesn´t work. Kepp getting the message "OTA: Firmware on '0x04cf8cdf3c759cac' (lumi.plug.mmeu01) is newer than latest firmware online".

hspjanssen, 2025-01-22:
still an issue.....

closed 2025-04-01 by github-actions[bot]