Koenkk/zigbee2mqtt #22207: [New device support]: Zigbee garage door module Forta SBG-Z2

ValKos, 2024-04-14:
### Link

https://allegro.pl/oferta/sterownik-bramy-garazowej-tuya-zigbee-z-czujnikiem-15344759426#

### Database entry

{"id":1,"type":"Coordinator","ieeeAddr":"0xe0798dfffee2498c","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1711417170985}

### Comments

For this device need to map sensor (TUYA dp 3).
Please find the converter below:
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

// Add a new converter for handling the specific Tuya data point
fz.tuya_garage_door_sensor = {
    cluster: 'manuSpecificTuya',
    type: 'commandDataResponse',
    convert: (model, msg, publish, options, meta) => {
        const dp = msg.data.dpValues[0].dp;
        const value = msg.data.dpValues[0].data[0];
        if (dp === 3) { // Check if the data point is for the garage door sensor
            return { contact: value === 0 }; // Convert to a boolean
        }
    },
};

const definition = {
    fingerprint: [{modelID: 'TS0603', manufacturerName: '_TZE608_fmemczv1'}],
    model: 'SBG-Z2',
    vendor: 'Forta',
    description: 'Garage door opener with the wired sensor',
    fromZigbee: [fz.tuya_garage_door_sensor, legacy.fromZigbee.matsee_garage_door_opener],
    toZigbee: [legacy.toZigbee.matsee_garage_door_opener, legacy.toZigbee.tuya_data_point_test],
    whiteLabel: [{vendor: 'Forta', model: 'SBG-Z2'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.binary('trigger', ea.STATE_SET, true, false).withDescription('Trigger the door movement'),
        e.contact('Sensor State', ea.STATE_GET, true, false).withDescription('Garage door sensor state')
    ],
    meta: {
        tuyaDatapoints: [
            [3, 'Sensor State', tuya.valueConverter.trueFalse1],
             // <- mapped the datapoint
        ],
    },
};

module.exports = definition;

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0603'],
    model: 'TS0603',
    vendor: '_TZE608_fmemczv1',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

ValKos, 2024-04-14:
'extend' definition needs to remove.

GSzabados, 2024-04-15:
@ValKos, how does this device work? Where do you need to connect the "signal" cable? It looks quite interesting.

ValKos, 2024-04-15:
> @ValKos, how does this device work? Where do you need to connect the "signal" cable? It looks quite interesting.

It's quite typical module, where you should connect wired sensor to two connectors. The problem with it that existed on Z2M converters can't handle this opening signal by default, and constantly show "N/A" and it needs to map third TUYA dp to do so.

GSzabados, 2024-04-15:
I understand the wired sensor, but not the signal one. Do you need to connect to the garage door controller module like the switch and it shorts the wires when triggered?

ValKos, 2024-04-15:
> I understand the wired sensor, but not the signal one. Do you need to connect to the garage door controller module like the switch and it shorts the wires when triggered?

Correct. As it usually implements on the similar modules, this device shorts the garage motor dry contact for a second to open or close the gate.

GSzabados, 2024-04-15:
Thanks! That is really cool! Like a Gogogate module, but with Zigbee. The extra would be if it would have an input for the a button as well. Nothing can be perfect...

ValKos, 2024-04-16:
It has connectors for the monostable button and is quite similar to the supported garage door opener (https://www.zigbee2mqtt.io/devices/PJ-ZGD01.html). However, it cannot be used with the existing PJ-ZGD01 converter due to a gate sensor issue.

kristoficko, 2024-04-30:
Hi @ValKos,
I just connected my new [zigbee garage door cover switch with wired door senzor](https://www.aliexpress.com/item/1005006776475188.html) and was hoping I could re-use your definitions to make it supported, as it seems very similar.

I'm just not sure I understand what to do - do I need to add multiple external converters? You have one definition in comments and then another, separate one, much simpler in the OP.

Thanks for clarification!

ValKos, 2024-04-30:
You only need one converter, and you can try using my converter from the original message in full. If it's not compatible, try adapting it to your device. Adaptation works well if you explore all TUYA datapoints. In my case, I simply reused the Matsee converter code and just added an extra subconverter for handling the sensor, as it wasn't properly working.

RuncoR, 2024-07-10:
Hi @ValKos

I bought a module from OXT (link below) which appears in z2m as "_TZE608_fmemczv1", so I tried to use your "coordinator" but unfortunately z2m doesn't replace the information in the given module at all. I'm not sure if I'm doing everything correctly, but when I created a coordinator with a different code (for adding Tuya devices), the data was substituted, but unfortunately neither the contact sensor nor the switch works. Does your code have any chance of working at all?

To use your coordinator, I simply created a file where I copied your code and added it to the list of external coordinators.

https://allegro.pl/oferta/oxt-sterownik-z-sensorem-do-silnika-bramy-garazu-zigbee-tuya-15524830102

Sorry for my English, I had to use a translator :)

michalkrzempek, 2024-08-07:
This device is becoming very popular in Poland due to its relatively low price. Could someone be kind enough to create a user-friendly guide on how to connect it to Home Assistant?

wjansenw, 2024-08-15:
The zigbee2mqtt-dev release has support for this device now.
Still trying to figure which automations I can use now... The switch seems to behave randomly: if the door is closed, and I use the switch it goes back to the same state shortly after. If the door is open(ing), and I use the switch it changes state.
I assumed that the switch would act like a button in homeassistant, and not as a switch.

gs-danielnikonczuk, 2024-08-28:
@ValKos I used [THIS](https://github.com/zigpy/zha-device-handlers/issues/3263) ZHA quirk for HA ZHA and it 
worked nicely, but only for the switch to short the garage engine to send an open/stop/close signal. Unfortunately, it didn't work for the sensor. I'm not sure if I have a broken device or just need additional adaptation. Still need to learn a lot to make my own mods for converters.

closed 2025-01-06 by github-actions[bot]