Koenkk/zigbee2mqtt #22209: TS0601 devices shows as unsupported  (simmilar to #22037)

LykkeGit, 2024-04-14:
### What happened?

When I update Z2M from 1.36.0-1 to 1.36.1-1 in HA my custom converter stops working. The device is named LM-150-24-G1Z2 from the supplier LTECH

++++

I seem to have the same issue as in #22037 with my TS0601 converter - used to work but now it ishows as unsupported when I update to 1.16.1. Removing "const ext..." didn't help

Rolled back to 1.36.0-1 and it is all working again

++++++
[20:51:18] INFO: Preparing to start...
[20:51:19] INFO: Socat not enabled
[20:51:21] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-04-03 20:51:31: Failed to load external converter file 'TS0601.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
Require stack:

/app/dist/util/utils.js
/app/dist/util/settings.js
/app/index.js)
Zigbee2MQTT:error 2024-04-03 20:51:31: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
Zigbee2MQTT:error 2024-04-03 20:51:31: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
++++++

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
// Since a lot of TuYa devices use the same modelID, but use different datapoints
// it's necessary to provide a fingerprint instead of a zigbeeModel
fingerprint: [
{
// The model ID from: Device with modelID 'TS0601' is not supported
// You may need to add \u0000 at the end of the name in some cases
modelID: 'TS0601',
// The manufacturer name from: Device with modelID 'TS0601' is not supported.
manufacturerName: '_TZE200_qyss8gjy',
},
],
model: '_TZE200_qyss8gjy',
vendor: 'TuYa',
description: 'LTECH 24V LED Driver',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
configure: tuya.configureMagicPacket,
extend: tuya.extend.switch({powerOnBehavior: true}),
exposes: [tuya.exposes.lightBrightness(), e.power_on_behavior(),
tuya.exposes.countdown()],
meta: {
// All datapoints go in here
tuyaDatapoints: [
[1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
[2, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
[3, 'brightness', tuya.valueConverter.scale0_254to0_1000],
[7, 'countdown', tuya.valueConverter.countdown],
],
},
};

module.exports = definition;

### How to reproduce it (minimal and precise)

That the converter continued to work

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

Can't find

### Adapter

usb-Nabu_Casa_SkyConnect_v1.0

### Setup

Add on to stock HA (204.4.3) running in VM on Synology NAS

### Debug log

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

Koenkk, 2024-04-15:
Check the first point of the release notes: https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1

closed 2024-04-15 by Koenkk

LykkeGit, 2024-04-15:
> Check the first point of the release notes: https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1

Thank you, but if I remove that line I get another error. 

"[22:33:42] INFO: Preparing to start...
[22:33:42] INFO: Socat not enabled
[22:33:45] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:error 2024-04-15 22:33:54: Failed to load external converter file 'TS0601.js' (Cannot read properties of undefined (reading 'switch'))
Zigbee2MQTT:error 2024-04-15 22:33:54: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
Zigbee2MQTT:error 2024-04-15 22:33:54: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device"

Is it this line:
"    extend: tuya.extend.switch({powerOnBehavior: true}),"

Probably some syntax problem, sorry about my ignorance

Koenkk, 2024-04-16:
Not sure what's wrong but I've added out-of-the-box support now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

LykkeGit, 2024-04-17:
Thank you!, much appreciated :-)