Koenkk/zigbee2mqtt #22211: Hive  KEYPAD001 works only partially

burner-, 2024-04-14:
### What happened?

Arming/Disarming from the server wont work at all. To log it will cause MAC timeout. 
Also when wake up device by pressing key pad will show communication error led (red icon at right top corner)
From log I see that after key press device will send several times status message. Most probably device will wait some kind ansver from server to it but because server wont reply anything it will give error.  Also there is no transaction id in status messages so there is no way to reply those at mqtt side. 
This is original dev thread for that device https://github.com/Koenkk/zigbee-herdsman-converters/pull/2967

zigbee_herdsman_converters_version
18.42.0
zigbee_herdsman_version
0.35.1

### Zigbee2MQTT version

1.36.0 commit: 86ed71c

### Adapter firmware version

 20211217

### Adapter

SMLIGHT SLZB-05

### Setup

docker at x86-64

### Debug log

Debug 2024-04-14 17:34:40Received Zigbee message from 'tt31_codepad', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Info 2024-04-14 17:34:40MQTT publish: topic 'zigbee2mqtt/tt31_codepad', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":116,"occupancy":true,"tamper":false,"voltage":3000}'
Debug 2024-04-14 17:34:43Received Zigbee message from 'tt31_codepad', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0
Info 2024-04-14 17:34:43MQTT publish: topic 'zigbee2mqtt/tt31_codepad', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":160,"occupancy":true,"tamper":false,"voltage":3000}'
Debug 2024-04-14 17:34:46Received Zigbee message from 'tt31_codepad', type 'commandGetPanelStatus', cluster 'ssIasAce', data '{}' from endpoint 1 with groupID 0

PepYk, 2024-09-07:
My Hive Keypad001 ends up with an error when I press a button:
EventBus error 'OnEvent/deviceMessage': CommandResponse 0x000d6ffffec6ef0a/1 ssIasAce.getPanelStatusRsp({"secondsremain":0,"audiblenotif":0,"alarmstatus":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false}) failed (Parameter 'panelstatus' is missing)

burner-, 2024-10-28:
looks that this bug is still exist

closed 2025-01-06 by github-actions[bot]

thebigspin, 2025-06-02:
I've found a fix for this in herdsman-convertors here (KEYPAD001):
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/hive.ts

`globalStore.getValue(data.endpoint, "panelStatus")` is returning null, causing the `panelstatus` to be missing in the payload. 

fix seems to be to set it to 0 if null:
```
const panelstatusobj = globalStore.hasValue(data.endpoint, "panelStatus")
    ? globalStore.getValue(data.endpoint, "panelStatus")
    : 0x00;

const payload = {
    panelstatus: panelstatusobj,
    secondsremain: 0x00,
    audiblenotif: 0x00,
    alarmstatus: 0x00,
};         
```
Error seems to be gone when running that as an external converter, but haven't had time to check it's working fully. 

Can submit a PR, but would be good to understand what has caused the getValue to start returning null.

Thrasher2020, 2025-07-18:
I also have the same issue. How can I add an external converter so my keypad can work again?