Koenkk/zigbee2mqtt #22214: Problems changing mqtt latest version of HA

javiges, 2024-04-14:
### What happened?

I have problems since the mqtt update in this month’s update and several devices, lights and LED strips that work through z2mqtt give me an error and I don’t know how to fix it, I have tried everything, spoken in telegram forums and no one tells me How can it be fixed or is it z2mqtt who has to do it, I leave you my log, I hope you can help me, thank you very much in advance.

[homeassistant.components.mqtt.light.schema_json] Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Lámpara Salón/availability'}], 'availability_mode': 'all', 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Lámpara Salón/set', 'device': {'identifiers': ['zigbee2mqtt_0x00158d0004f910fb'], 'manufacturer': 'Innr', 'model': 'E27 bulb RGBW (RB 285 C)', 'name': 'Lámpara Salón', 'sw_version': '2.1', 'via_device': 'zigbee2mqtt_bridge_0x00124b0029dc0fb9'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'json_attributes_topic': 'zigbee2mqtt/Lámpara Salón', 'max_mireds': 555, 'min_mireds': 153, 'name': None, 'object_id': 'lámpara_salón', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.36.1', 'support_url': 'https://www.zigbee2mqtt.io'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Lámpara Salón', 'supported_color_modes': ['xy', 'color_temp'], 'unique_id': '0x00158d0004f910fb_light_zigbee2mqtt', 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3
[homeassistant.components.mqtt.light.schema_json] Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Luz Estanterias/availability'}], 'availability_mode': 'all', 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Luz Estanterias/set', 'device': {'identifiers': ['zigbee2mqtt_0x680ae2fffe11125e'], 'manufacturer': 'IKEA', 'model': 'TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white (LED1624G9)', 'name': 'Luz Estanterias', 'sw_version': '2.3.093', 'via_device': 'zigbee2mqtt_bridge_0x00124b0029dc0fb9'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'json_attributes_topic': 'zigbee2mqtt/Luz Estanterias', 'max_mireds': 500, 'min_mireds': 153, 'name': None, 'object_id': 'luz_estanterias', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.36.1', 'support_url': 'https://www.zigbee2mqtt.io'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Luz Estanterias', 'supported_color_modes': ['xy', 'color_temp'], 'unique_id': '0x680ae2fffe11125e_light_zigbee2mqtt', 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3
[homeassistant.components.mqtt.light.schema_json] Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Luz Lavadora/availability'}], 'availability_mode': 'all', 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Luz Lavadora/set', 'device': {'identifiers': ['zigbee2mqtt_0x00158d0004f91eb6'], 'manufacturer': 'Innr', 'model': 'E27 bulb RGBW (RB 285 C)', 'name': 'Luz Lavadora', 'sw_version': '2.1', 'via_device': 'zigbee2mqtt_bridge_0x00124b0029dc0fb9'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'json_attributes_topic': 'zigbee2mqtt/Luz Lavadora', 'max_mireds': 555, 'min_mireds': 153, 'name': None, 'object_id': 'luz_lavadora', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.36.1', 'support_url': 'https://www.zigbee2mqtt.io'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Luz Lavadora', 'supported_color_modes': ['xy', 'color_temp'], 'unique_id': '0x00158d0004f91eb6_light_zigbee2mqtt', 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3
[homeassistant.components.mqtt.light.schema_json] Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Led Tv/availability'}], 'availability_mode': 'all', 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Led Tv/set', 'device': {'identifiers': ['zigbee2mqtt_0xcc86ecfffe56aee7'], 'manufacturer': 'TuYa', 'model': 'Zigbee RGB light (TS0503B)', 'name': 'Led Tv', 'via_device': 'zigbee2mqtt_bridge_0x00124b0029dc0fb9'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'json_attributes_topic': 'zigbee2mqtt/Led Tv', 'name': None, 'object_id': 'led_tv', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.36.1', 'support_url': 'https://www.zigbee2mqtt.io'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Led Tv', 'supported_color_modes': ['xy'], 'unique_id': '0xcc86ecfffe56aee7_light_zigbee2mqtt', 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3
[homeassistant.components.mqtt.light.schema_json] Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Escritorio Sergio/availability'}], 'availability_mode': 'all', 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Escritorio Sergio/set', 'device': {'identifiers': ['zigbee2mqtt_0x842e14fffe15fa26'], 'manufacturer': 'Lidl', 'model': 'Livarno Lux smart LED light strip 2.5m (HG06104A)', 'name': 'Escritorio Sergio', 'via_device': 'zigbee2mqtt_bridge_0x00124b0029dc0fb9'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'json_attributes_topic': 'zigbee2mqtt/Escritorio Sergio', 'max_mireds': 500, 'min_mireds': 150, 'name': None, 'object_id': 'escritorio_sergio', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.36.1', 'support_url': 'https://www.zigbee2mqtt.io'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Escritorio Sergio', 'supported_color_modes': ['xy', 'color_temp'], 'unique_id': '0x842e14fffe15fa26_light_zigbee2mqtt', 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3
[homeassistant.components.mqtt.light.schema_json] Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Led Cama Sergio/availability'}], 'availability_mode': 'all', 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Led Cama Sergio/set', 'device': {'identifiers': ['zigbee2mqtt_0xcc86ecfffe56af49'], 'manufacturer': 'TuYa', 'model': 'Zigbee RGB light (TS0503B)', 'name': 'Led Cama Sergio', 'via_device': 'zigbee2mqtt_bridge_0x00124b0029dc0fb9'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect'], 'json_attributes_topic': 'zigbee2mqtt/Led Cama Sergio', 'name': None, 'object_id': 'led_cama_sergio', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '1.36.1', 'support_url': 'https://www.zigbee2mqtt.io'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Led Cama Sergio', 'supported_color_modes': ['xy'], 'unique_id': '0xcc86ecfffe56af49_light_zigbee2mqtt', 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3
[homeassistant.components.mqtt.light.schema_json] Deprecated flag `color_mode` used in MQTT JSON light config , the `color_mode` flag is not used anymore and should be removed. Got: {'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Led Escritorio Martina/availability'}], 'availability_mode': 'all', 'brightness': True, 'brightness_scale': 254, 'color_mode': True, 'command_topic': 'zigbee2mqtt/Led Escritorio Martina/set', 'device': {'identifiers': ['zigbee2mqtt_0x588e81fffed27b19'], 'manufacturer': 'Lidl', 'model': 'Livarno Lux smart LED light strip 2.5m (HG06104A)', 'name': 'Led Escritorio Martina', 'via_device': 'zigbee2mqtt_bridge_0x00124b0029dc0fb9'}, [... cut ...] 'sw_version': '1.36.1', 'support_url': 'https://www.zigbee2mqtt.io'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Led Cama Martina', 'supported_color_modes': ['xy'], 'unique_id': '0x9035eafffe0533ef_light_zigbee2mqtt', 'platform': 'mqtt'}. This will stop working in Home Assistant Core 2025.3

### What did you expect to happen?

no problem

### Zigbee2MQTT version

1.36.1

### Adapter firmware version

20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

Docker z2mqtt

francisp2, 2024-04-16:
> This will stop working in Home Assistant Core 2025.3

It is a warning, not an error. You can switch to the dev version if you don't want to see it.

Koenkk, 2024-04-16:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/pull/22142

closed 2024-04-16 by Koenkk