Koenkk/zigbee2mqtt #22217: [New device support]: Frient Smart Button

Ulrar, 2024-04-15:
### Link

https://frient.com/products/smart-button/

### Database entry

{"id":94,"type":"EndDevice","ieeeAddr":"0x0015bc0039002efd","nwkAddr":50325,"manufId":4117,"epList":[1,32],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[5,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"32":{"epId":32,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1713176942636}

### Comments

Hi,

I tried following the page to support the device, but it looks like that expects the pairing to succeed which is not the case here.
In the dev console all the fields are just empty, and I can fill them in myself but it doesn't look like that convert is then matched to the device, not sure what the process should be here. The logs just say the pairing failed, and the button itself keeps blinking as it it hadn't joined a network.

### External definition

```shell
const {deviceEndpoints, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: [''],
    model: '',
    vendor: '',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"32":32}}), onOff({"powerOnBehavior":false})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

Ulrar, 2024-04-15:
Following the model number on the back of the device itself, I found this : https://www.zigbee2mqtt.io/devices/SBTZB-110.html
Looks like the same device, not sure why the frient version, if there is a difference, fails

Ulrar, 2024-06-11:
Could it be a mismatch in converter, because it's for Develco but this one advertises Frient ?

Koenkk, 2024-06-11:
The interview failed, starting from 1.38.0 you can retrigger it from the device about page in the frontend

Ulrar, 2024-06-11:
I've retried the interview numerous times, the issue is it fails every time :

```
error 2024-06-11 18:00:30Interview failed for '0x0015bc0039002efd with error 'Error: Interview failed because can not get node descriptor ('0x0015bc0039002efd')'
error 2024-06-11 18:00:30Failed to interview '0x0015bc0039002efd', device has not successfully been paired
```

As I was saying the button's LED keeps blinking for 15 minutes regardless, even though the doc says it should stop once it joined. Pressing the button always starts that 15 minutes of slow blinking, even if Z2M has the device listed (as unsupported)

Koenkk, 2024-06-11:
Can you try pairing it closer to the coordinator? The device not responding is not something that can be fixed from z2m directly.

Ulrar, 2024-06-11:
I've been doing it from pretty close already, but even right next to it same result. Could it be the battery maybe ? I don't think I have anything for those screws, I'll need to order something to find out

Ulrar, 2024-06-11:
Actually, the doc I found online says `The device will scan each ZigBee channel starting from 11 to 24`. I'm on channel 25, is it credible that it'd show up in Z2M but wouldn't be able to respond ? I'm surprised it even shows up, but I don't know how the joining process work, maybe that's coordinator side only

Koenkk, 2024-06-11:
I would expect it wouldn't join at all then, but it's worth trying another channel.

closed 2024-12-29 by github-actions[bot]