Koenkk/zigbee2mqtt #22219: Aqara ceiling light T1M (CL-L02D) - Power outgage memory not working

brainstormi, 2024-04-15:
### What happened?

Trying to enable or disable 'Power outgage memory' doesn't work, reporting an error about Attribute being unsuported.

### Zigbee2MQTT version

1.36.1-1

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Setup

Addon, Home Assistant 2024.4.2 running virtualized in Proxmox 8.1 x86-64

### Debug log

Publish 'set' 'power_outage_memory' to 'Luz despacho' failed: 'Error: ZCL command 0x54ef441000b7c15f/1 genBasic.write({"65305":{"value":1,"type":16}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

brainstormi, 2024-04-15:
#https://github.com/Koenkk/zigbee-herdsman-converters/pull/7203

Koenkk, 2024-05-02:
Device doesn't seem to support it, disabled it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-02 by Koenkk

brainstormi, 2024-05-05:
> Device doesn't seem to support it, disabled it!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Far from being an expert around zigbee, but seems to be implented in ZHA although the attribute for this device would be different:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/7203#issuecomment-2019058679